home-assistant/operating-system #910: bluetoothctl & Integration: mitemp_bt not working after upgrade to HassOS 4.14

maximcoder, 2020-10-17:
**HassOS release with the issue:**
HassOS 4.14

**Supervisor logs:**
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[14:22:14] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[14:22:15] INFO: Start local supervisor watchdog...
20-10-17 14:22:19 INFO (MainThread) [__main__] Initialize Supervisor setup
20-10-17 14:22:19 INFO (MainThread) [supervisor.bootstrap] Initialize Supervisor Sentry
20-10-17 14:22:19 INFO (MainThread) [supervisor.bootstrap] Setup coresys for machine: raspberrypi3
20-10-17 14:22:19 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attach to Supervisor homeassistant/armv7-hassio-supervisor with version 247
20-10-17 14:22:19 INFO (MainThread) [__main__] Setup Supervisor
20-10-17 14:22:20 INFO (MainThread) [supervisor.host.info] Update local host information
20-10-17 14:22:20 INFO (MainThread) [supervisor.host.services] Update service information
20-10-17 14:22:20 INFO (MainThread) [supervisor.host.network] Update local network information
20-10-17 14:22:21 INFO (MainThread) [supervisor.host.sound] Update PulseAudio information
20-10-17 14:22:21 INFO (MainThread) [supervisor.host.apparmor] Load AppArmor Profiles: {'hassio-supervisor'}
20-10-17 14:22:21 INFO (MainThread) [supervisor.host.services] Reload local service hassos-apparmor.service
20-10-17 14:22:21 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-dns with version 9
20-10-17 14:22:21 INFO (MainThread) [supervisor.plugins.dns] Start CoreDNS plugin
20-10-17 15:23:10 INFO (SyncWorker_0) [supervisor.docker.dns] Start DNS homeassistant/armv7-hassio-dns with version 9 - 172.30.32.3
20-10-17 15:23:10 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
20-10-17 15:23:10 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-audio with version 17
20-10-17 15:23:10 INFO (MainThread) [supervisor.plugins.audio] Start Audio plugin
20-10-17 15:23:12 INFO (SyncWorker_0) [supervisor.docker.audio] Start Audio homeassistant/armv7-hassio-audio with version 17 - 172.30.32.4
20-10-17 15:23:12 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-cli with version 26
20-10-17 15:23:12 INFO (MainThread) [supervisor.plugins.cli] Start cli plugin
20-10-17 15:23:13 INFO (SyncWorker_0) [supervisor.docker.cli] Start CLI homeassistant/armv7-hassio-cli with version 26 - 172.30.32.5
20-10-17 15:23:13 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-observer with version 2020.10.1
20-10-17 15:23:14 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-hassio-multicast with version 3
20-10-17 15:23:14 INFO (MainThread) [supervisor.plugins.multicast] Start Multicast plugin
20-10-17 15:23:15 INFO (SyncWorker_0) [supervisor.docker.multicast] Start Multicast homeassistant/armv7-hassio-multicast with version 3 - Host
20-10-17 15:23:15 INFO (MainThread) [supervisor.updater] Fetch update data from https://version.home-assistant.io/stable.json
20-10-17 15:23:15 INFO (SyncWorker_1) [supervisor.docker.interface] Attach to homeassistant/raspberrypi3-homeassistant with version 0.116.4
20-10-17 15:23:15 INFO (MainThread) [supervisor.homeassistant.secrets] Load Home Assistant secrets: 3
20-10-17 15:23:15 INFO (MainThread) [supervisor.hassos] Detect HassOS 4.14 / BootSlot B
20-10-17 15:23:16 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/a0d7b954 repository
20-10-17 15:23:16 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/git/15ef4d2f repository
20-10-17 15:23:16 INFO (MainThread) [supervisor.store.git] Load add-on /data/addons/core repository
20-10-17 15:23:17 INFO (MainThread) [supervisor.store] Load add-ons from store: 69 all - 69 new - 0 remove
20-10-17 15:23:17 INFO (MainThread) [supervisor.addons] Found 6 installed add-ons
20-10-17 15:23:17 INFO (SyncWorker_1) [supervisor.docker.interface] Attach to hassioaddons/portainer-armv7 with version 1.2.2
20-10-17 15:23:17 INFO (SyncWorker_0) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-configurator with version 5.1.0
20-10-17 15:23:17 INFO (SyncWorker_2) [supervisor.docker.interface] Attach to hassioaddons/ssh-armv7 with version 7.6.0
20-10-17 15:23:17 INFO (SyncWorker_5) [supervisor.docker.interface] Attach to esphome/esphome-hassio-armv7 with version 1.15.2
20-10-17 15:23:17 INFO (SyncWorker_3) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-duckdns with version 1.12.1
20-10-17 15:23:17 INFO (SyncWorker_4) [supervisor.docker.interface] Attach to homeassistant/armv7-addon-samba with version 9.3.0
20-10-17 15:23:17 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
20-10-17 15:23:17 INFO (MainThread) [supervisor.discovery] Load 0 messages
20-10-17 15:23:17 INFO (MainThread) [supervisor.ingress] Load 0 ingress session
20-10-17 15:23:17 INFO (MainThread) [__main__] Run Supervisor
20-10-17 15:23:17 INFO (MainThread) [supervisor.api] Start API on 172.30.32.2
20-10-17 15:23:17 INFO (MainThread) [supervisor.hassos] Rauc: B - marked slot kernel.1 as good
20-10-17 15:23:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' start 0 add-ons
20-10-17 15:23:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SYSTEM' start 0 add-ons
20-10-17 15:23:17 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.SERVICES' start 2 add-ons
20-10-17 15:23:17 WARNING (MainThread) [supervisor.addons.validate] Unknown options web
20-10-17 15:23:17 WARNING (SyncWorker_5) [supervisor.docker.addon] SSH & Web Terminal run with disabled protected mode!
20-10-17 15:23:19 INFO (SyncWorker_5) [supervisor.docker.addon] Start Docker add-on hassioaddons/ssh-armv7 with version 7.6.0
20-10-17 15:23:20 INFO (SyncWorker_0) [supervisor.docker.addon] Start Docker add-on homeassistant/armv7-addon-duckdns with version 1.12.1
20-10-17 15:23:23 INFO (MainThread) [supervisor.api.security] /host/info access from a0d7b954_ssh
20-10-17 15:23:26 INFO (SyncWorker_5) [supervisor.docker.interface] Start homeassistant
[... 561 lines cut ...]
[  951.740338] udevd[190]: starting version 3.2.9
[  951.759035] udevd[191]: starting eudev-3.2.9
[  986.593044] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  986.596048] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 1302.602358] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1302.606361] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 1618.575116] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1618.578178] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 1934.618806] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1934.622001] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2250.612757] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2250.616557] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 2566.605119] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2566.608878] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

**Description of problem:**
The  bluethootctl is not working and   mitemp_bt.sensor not working affter upgrade to HassOS 4.14 from HassOS 4.13
Logger: homeassistant.components.mitemp_bt.sensor
Source: components/mitemp_bt/sensor.py:163
Integration: mitemp_bt (documentation, issues)
First occurred: 11:39:17 AM (192 occurrences)
Last logged: 12:10:48 PM

Polling error
Polling error Could not read data from Mi Temp sensor 58:2D:34:3A:85:A9

mohannadalsharief, 2020-10-17:
Same issue here

mohannadalsharief, 2020-10-17:
Anu idea how to downgrade?

maximcoder, 2020-10-18:
> Anu idea how to downgrade?

nope, I will probably reinstall the whole HA

maximcoder, 2020-10-18:
> Anu idea how to downgrade?

just figure it out  for downgrade do this in cli

ha os update --version "4.13"

mohannadalsharief, 2020-10-18:
Great

ShonP40, 2020-10-18:
Had the same issue with the Bluetooth tracker.
Downgrading to 4.13 solved it.

markom80, 2020-10-18:
Had to downgrade because ruuvitag didn't work

Ernst79, 2020-10-18:
The custom component of mitemp_bt has a similar [issue](https://github.com/custom-components/sensor.mitemp_bt/issues/123). Bluetooth does not seem to work anymore after upgrading to HassOs 4.14

Cevipower, 2020-10-18:
Same issue here. Stopped comms with MiFlora devices. "hcitool dev" dos not return any device

brianx87, 2020-10-18:
HAving bluetooth issues;

```
2020-10-18 21:32:06 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for inkbird which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-10-18 21:32:40 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up inkbird platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 193, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/inkbird/sensor.py", line 76, in setup_platform
    inkbird_devices.append( InkbirdUpdater(hass, inkbird_devices) )
  File "/config/custom_components/inkbird/sensor.py", line 93, in __init__
    self.scanner.start()
  File "/usr/local/lib/python3.8/site-packages/bluepy/btle.py", line 790, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python3.8/site-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'le on' (code: 17, error: Invalid Index)
```

agners, 2020-10-19:
@mohannadalsharief @ShonP40 @markom80 @Ernst79 @Cevipower @brianx87 are you all using RPi 3?

markom80, 2020-10-19:
> @mohannadalsharief @ShonP40 @markom80 @Ernst79 @Cevipower @brianx87 are you all using RPi 3?

Yes, RPi 3B+

mohannadalsharief, 2020-10-19:
That's a yes for me

Cevipower, 2020-10-19:
> @mohannadalsharief @ShonP40 @markom80 @Ernst79 @Cevipower @brianx87 are you all using RPi 3?

Yes, this is my case too. RPi3

Ernst79, 2020-10-19:
I'm not having the issue myself (I don't run Home Assistant OS), only made a link to the issue of @mohannadalsharief which he raised on the mitemp_bt github page. But he is using a Rpi3, as he mentioned two posts above.

farkeld, 2020-10-19:
I have the same problem wirh my Rpi 3B+ but also the ethernet adapter is not working

ShonP40, 2020-10-19:
> @mohannadalsharief @ShonP40 @markom80 @Ernst79 @Cevipower @brianx87 are you all using RPi 3?

Yea
RPI 3B+

brianx87, 2020-10-19:
Yes I'm on a Pi3B+, reverting to 14.3 fixes the issue temporarily

elRadix, 2020-10-19:
Seems to be only related to rpi, works wel on my NUC with an external USB BT dongle

TheBisk, 2020-10-19:
> @mohannadalsharief @ShonP40 @markom80 @Ernst79 @Cevipower @brianx87 are you all using RPi 3?

me too, same problem with bluetooth device tracker and mitemp_bt
I'm on Rpi 3B+

alb147, 2020-10-19:
Same problem rpi3b+

Logger: homeassistant
Source: components/bluetooth_tracker/device_tracker.py:57
First occurred: 16:42:02 (117 occurrences)
Last logged: 16:51:36

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 177, in update_bluetooth
    await perform_bluetooth_update()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 140, in perform_bluetooth_update
    devices = await hass.async_add_executor_job(discover_devices, device_id)
  File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 57, in discover_devices
    result = bluetooth.discover_devices(
  File "/usr/local/lib/python3.8/site-packages/bluetooth/bluez.py", line 26, in discover_devices
    device_id = _bt.hci_get_route()
OSError: [Errno 19] No such device

eiros, 2020-10-19:
Same problem& Rpi 3b
Downgrade to 4.13 helped.
Terminal command: **ha os update --version=4.13**
You can use Web UI of add-on "Terminal & SSH"

Soni93200, 2020-10-19:
rpi 3B+ here same for me

iras24sx, 2020-10-19:
Yes, same for me, again rpi3b+.

2020-10-19 19:37:25 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 177, in update_bluetooth
    await perform_bluetooth_update()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 140, in perform_bluetooth_update
    devices = await hass.async_add_executor_job(discover_devices, device_id)
  File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/bluetooth_tracker/device_tracker.py", line 57, in discover_devices
    result = bluetooth.discover_devices(
  File "/usr/local/lib/python3.8/site-packages/bluetooth/bluez.py", line 26, in discover_devices
    device_id = _bt.hci_get_route()
OSError: [Errno 19] No such device

Elitaz, 2020-10-19:
I found other fix working for me.  **sudo apt-get remove bluez** and  **sudo apt-get install bluez**

pvizeli, 2020-10-19:
A short github lession:
[image]

Means it will be fix with next release

PR #915 (2020-10-20): RaspberryPI: Fix issue with correct SYMLINK with ttyAMA
Fix: #910 

Partial different approach as https://github.com/home-assistant/operating-system/pull/862 - however, we use this only for the Bluetooth scripts.
closed 2020-10-20 by agners