home-assistant/operating-system #912: VMDK Live Snapshot Changing Z-Wave USB TTY mapping

MwC-Trexx, 2020-10-18:
When creating a live snapshot of running system with USB Z-Wave dongle pass-through, the serial device within HASSOS changes post snapshot.

Scenario:
(Pre-Snapshot) Z-wave dongle mapped as follows:

/dev/ttyACM0
/dev/serial/by-id/usb-0658_0200-if00

(Post-Snapshot) Z-wave dongles mapped as follows:

/dev/ttyACM1
/dev/serial/by-id/usb-0658_0200-if00

This of course breaks the OZW 1.6 integration due to the device port change.  OZW also doesn't work if trying to map via /dev/serial config.

Any options for persisting the /dev/tty value based on /dev/serial not changing?

agners, 2020-10-20:
Just use `/dev/serial/by-id/usb-0658_0200-if00`, it is meant as a stable symlink to the real device. See also https://www.home-assistant.io/docs/z-wave/installation/#changing-device-paths.

closed 2020-10-20 by agners

MwC-Trexx, 2020-10-20:
That doesn't appear to be compatible at this time with OZW 1.6 beta.

```
[20201020 15:09:12.930 CDT] [ozw.library] [info]: Info - Node: 0   Opening controller /dev/serial/by-id/usb-0658_0200-if00/dev/ttyACM0 
[20201020 15:09:12.930 CDT] [ozw.library] [info]: Info - Node: 0 Trying to open serial port /dev/serial/by-id/usb-0658_0200-if00/dev/ttyACM0 (attempt 1) 
[20201020 15:09:12.931 CDT] [ozw.library] [critical]: Error - Node: 0 ERROR: Cannot open serial port /dev/serial/by-id/usb-0658_0200-if00/dev/ttyACM0. Error code 20 
[20201020 15:09:12.931 CDT] [ozw.library] [critical]: Error - Node: 0 ERROR: Failed to open serial port /dev/serial/by-id/usb-0658_0200-if00/dev/ttyACM0 
[20201020 15:09:12.931 CDT] [ozw.library] [warning]: Warning - Node: 0 WARNING: Failed to init the controller (attempt 1) 
[20201020 15:09:17.931 CDT] [ozw.library] [info]: Info - Node: 0   Opening controller /dev/serial/by-id/usb-0658_0200-if00/dev/ttyACM0 
[20201020 15:09:17.931 CDT] [ozw.library] [info]: Info - Node: 0 Trying to open serial port /dev/serial/by-id/usb-0658_0200-if00/dev/ttyACM0 (attempt 1) 
```