home-assistant/operating-system #913: When booting from USB, "Import config from USB" does not work

ocalvo, 2020-10-18:
**HassOS release with the issue:**

5.3

**Supervisor logs:**

**Journal logs:**

Cannot activate ssh on 22222

**Kernel logs:**

Don't have access to the base os.

**Description of problem:**
1. Boot HassOS from USB directly on a RPI4 64bit.
2. Create a config usb stick
3. Goto to supervisor: Import from USB config
4. Reboot system

Expected:
Configuration from the USB stick is applied
Actual:
No config is applied to the host system

cogneato, 2020-10-18:
You actually don't need to reboot after importing the authorized_keys file. If it is done correctly, ssh access over 22222 works immediately.

ocalvo, 2020-10-18:
Thanks, but in any case I cannot change the system boot config.

agners, 2020-10-18:
Hm, I am almost sure I used this once, but I usually just use the bootup mechanism (on boot-up the confiugration is imported as well). I'll have to retry here.

ocalvo, 2020-10-18:
Where are the authorized_keys file stored in the host system?

drthanwho, 2021-02-27:
I can also reproduce this on my side with an SSD only install. Key isn't imported from Supervisor. Gives weird error in Core and nothing on Supervisor. Rebooting host imports the key successfully however. 

```log
2021-02-27 02:01:03 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on os/config/sync request Cannot connect to host 172.30.32.2os:80 ssl:default [Name does not resolve]
2021-02-27 02:01:03 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call os/config/sync - 
```

maxgyver87, 2021-03-13:
I am having the same issue here with a home assistant blue (Odroid N2+)
Am trying to import a authorized_keys file like the [documentation](https://developers.home-assistant.io/docs/operating-system/debugging/#generating-ssh-keys) explained.

I tried multiple ways of formatting the usb drive.
The content of the file is in one line ala `ssh-rsa some_long/string@b6x927-ssh`.

Import gives a error message:

Failed to import from USB
Unknown error, see supervisor logs

Supervisor log: 
`INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE usb hardware /dev/bus/usb/001/005`

Core Log:
`
2021-03-13 17:36:09 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on os/config/sync request Cannot connect to host 172.30.32.2os:80 ssl:default [Name does not resolve]2021-03-13 17:36:09 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call os/config/sync - 
2021-03-13 17:36:10 ERROR (MainThread) [custom_components.remote_homeassistant] websocket connection had an error`

maxgyver87, 2021-03-13:
I am having the same issue here with a home assistant blue (Odroid N2+)

**Core Version**: core-2021.3.4
**Supervisor Version**: supervisor-2021.03.4
**Home Assistant OS**: 5.12

Am trying to import a authorized_keys file like the [documentation](https://developers.home-assistant.io/docs/operating-system/debugging/#generating-ssh-keys) explained.

I tried multiple ways of formatting the usb drive.
The content of the file is in one line ala `ssh-rsa some_long/string@b6x927-ssh`.
Applied chmod 600 authorized_keys

Import from USB gives this error message:

**Failed to import from USB
Unknown error, see supervisor logs**

Supervisor log: 
`21-03-13 17:50:26 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD usb hardware /dev/bus/usb/001/006`

Core Log:
```
2021-03-13 17:50:57 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on os/config/sync request Cannot connect to host 172.30.32.2os:80 ssl:default [Name does not resolve]
2021-03-13 17:50:57 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call os/config/sync - 
```

drthanwho, 2021-03-13:
@maxgyver87 did rebooting not import the key?

maxgyver87, 2021-03-13:
@drthanwho  
to be sure I rebooted the system after every attempt.
I get a 
`root@192.xxx.xxx.xxx: Permission Denied (publickey).`

drthanwho, 2021-03-13:
@maxgyver87 assuming you were SSHing on port 22222 and not 22, and that you picked the equivalent private key (god knows how many times I've accidentally selected the wrong private key) then your issue seems more likely to be with your actual public key. 

And especially given this issue is on systems booting from USB. HA Blue doesn't boot from SSD. I would try asking for help over at Discord

maxgyver87, 2021-03-13:
@drthanwho
ill look at it again in a few days. Maybe start fresh.
this issue is bothering me for three weeks now.

Jpsy, 2021-03-16:
Exactly the same problem here.
Including the strange error msg with broken IP:  `annot connect to host 172.30.32.2os:80`

Booting with USB stick attached imported the pub key successfully. So this is a workaround.

rpdmiranda, 2021-03-24:
I'm having the same issue here, but booting from a SD card with a SSD plugged in an USB port.

## System Health

version | core-2021.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.83-v8
arch | aarch64
timezone | America/Sao_Paulo

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4919
Installed Version | 1.11.3
Stage | running
Available Repositories | 775
Installed Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 12 de abril de 2021 21:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.03.6
docker_version | 19.03.13
disk_total | 57.8 GB
disk_used | 4.9 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.6.0), SSH & Web Terminal (8.0.4), File editor (5.2.0), Node-RED (8.2.0), Grocy (0.11.1), Home Assistant Google Drive Backup (0.103.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 0
views | 4
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

rattatosk, 2021-03-26:
Having the same issue:
```
2021-03-26 10:22:30 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on os/config/sync request Cannot connect to host 172.30.32.2os:80 ssl:default [Name does not resolve]
2021-03-26 10:22:30 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call os/config/sync - 

```
Tried different formats on usb, but no solution. 

Home Assistant OS
core-2021.3.4
supervisor-2021.03.6
Home Assistant OS 5.12

Hi @ocalvo 
> I validated that by using potainer to have access to the host and place the file /root/.ssh/authorized_keys I now have access to the host system.

How/where do you get access through portainter to the host  ? Would love to do use your approach.

Roagert, 2021-04-01:
Im also having this same issue, i am not able to reboot my Home assistant setup for it to load the configuration file.
I dont know if it has anything to the matter if im using a Raspberry Pi 4 8GB with HA installed on a USB ?

closed 2022-01-11 by github-actions[bot]