home-assistant/operating-system #914: When booting from USB udev rules loaded in /etc/udev/rules.d are not executed

ocalvo, 2020-10-18:
**HassOS release with the issue:**

5.3

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
1. Create a cutom udev rule like described (here)[https://www.home-assistant.io/integrations/sms/]
2. Insert a USB modem stick.
Expected:
UDev rule is executed and modem is changed to modem mode.
Actual:
Modem is left in storage mode.

Note: running `/sbin/usb_modeswitch -X -v 12d1 -p 14fe` manually does change the modem mode.

agners, 2020-11-02:
Does a reboot helps? E.g. is the udev rule installed but not activated?

ocalvo, 2020-11-02:
No, reboot does not help.

ocalvo, 2021-03-11:
Issue was that file was encoded with CRLF and not LF.
You can close it.