home-assistant/operating-system #916: Problems when trying to update from 4.13 to 4.14

thijsdejong, 2020-10-19:
**HassOS release with the issue:** 4.13

**Supervisor logs:**

```
20-10-19 10:30:15 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.14/hassos_rpi3-4.14.raucb
20-10-19 10:30:50 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.14.raucb
20-10-19 10:30:51 INFO (MainThread) [supervisor.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-10-19 10:31:15 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.14/hassos_rpi3-4.14.raucb
20-10-19 10:31:16 INFO (MainThread) [supervisor.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-10-19 10:31:16 ERROR (MainThread) [supervisor.hassos] HassOS update failed with: Installation error: Failed verifying checksum: Read from /run/rauc/bundle/rootfs.img failed: Input/output error
20-10-19 10:31:16 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-391586' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 159, in update
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError
20-10-19 10:31:50 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.14.raucb
20-10-19 10:31:50 INFO (MainThread) [supervisor.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-10-19 10:32:15 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.14/hassos_rpi3-4.14.raucb
20-10-19 10:32:30 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-10-19 10:32:50 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.14.raucb
20-10-19 10:32:50 INFO (MainThread) [supervisor.utils.gdbus] Start dbus monitor on de.pengutronix.rauc
20-10-19 10:33:11 INFO (MainThread) [supervisor.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-10-19 10:33:11 INFO (MainThread) [supervisor.utils.gdbus] Stop dbus monitor on de.pengutronix.rauc
20-10-19 10:33:11 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-391649' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:125> exception=FileNotFoundError(2, 'No such file or directory')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 148, in update
    int_ota.unlink()
  File "/usr/local/lib/python3.8/pathlib.py", line 1321, in unlink
    self._accessor.unlink(self)
FileNotFoundError: [Errno 2] No such file or directory: '/data/tmp/hassos-4.14.raucb'
20-10-19 10:33:11 ERROR (MainThread) [supervisor.hassos] HassOS update failed with: Installation error: Failed updating slot rootfs.0: Opening output device /dev/disk/by-partlabel/hassos-system0 failed: Device or resource busy
```

**Journal logs:**

**Kernel logs:**

[image]

**Description of problem:**
When I click the update button, the following logs appear. This is only after pressing the button once. The operating system does not update.

pvizeli, 2020-10-19:
@agners that is a bug with loop device and fix on Kernel 5.9 (mostly)

agners, 2020-10-20:
@pvizeli I guess you are referring to #544? It looks a bit different though, as it is saying `Installation error: Failed verifying checksum: Read from /run/rauc/bundle/rootfs.img failed: Input/output error` (the other issue is about `Failed mounting bundle: failed to run mount: Child process exited with code 32`.

@thijsdejong is this reproducibly lead to the same error? Which version did you originally install?

thijsdejong, 2020-10-21:
@agners I started off on this device on 4.12. It's a raspberry pi 3 B+.

I'm not sure what you mean, but after restarting numerous times I still get the exact same errors. Is there anything I could do to test more?

agners, 2020-10-21:
I tried to reproduce this using the same HW/SW combination (Raspberry Pi 3 B+, installed 4.12, updated to 4.13 and then to 4.14). I was not able to reproduce it, the update went through smoothly:
```
20-10-21 19:08:43 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.14/hassos_rpi3-4.14.raucb
20-10-21 19:09:53 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.14.raucb
20-10-21 19:09:53 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
20-10-21 19:10:10 INFO (MainThread) [supervisor.utils.gdbus] Stopping dbus monitor on de.pengutronix.rauc
20-10-21 19:10:10 INFO (MainThread) [supervisor.hassos] Install of Home Assistant Operating System 4.14 success
20-10-21 19:10:10 INFO (MainThread) [supervisor.host.control] Initialize host reboot over systemd
```

Can you access Home Assistant Operating System via SSH (port 22222) and run the following command after an update try:

```
ha > login
# journalctl -u rauc
...
```

In case you don't have access to Home Assistant Operating System, see this guide how to set it up: https://developers.home-assistant.io/docs/operating-system/debugging/

closed 2020-12-30 by agners