home-assistant/operating-system #917: "TypeError: Failed to fetch" when restoring snapshot on fresh install

TheGroundZero, 2020-10-19:
**HassOS release with the issue:**

hassos_ova-4.14

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Reinstalled HassOS in Proxmox using hassos_ova-4.14.qcow2 as new disk after latest install caused HA to no longer boot ([EUFI issue](https://pve.proxmox.com/wiki/Roadmap#_6.2_known_issues_ovmf)) and the [fix](https://pve.proxmox.com/wiki/OVMF/UEFI_Boot_Entries) didn't work (Proxmox no longer detected drive).

Followed Juan's [method of installing HA on Proxmox](https://www.juanmtech.com/install-proxmox-and-virtualize-home-assistant/).  
Downloaded qcow2.tar, unzipped, transfered qcow2 file, ran command to add as disk to HA instance, changed disks and booted from new disk.

[image]

Once Home Assistant has started, I select the option to restore from a previous snapshot.  
I then select a fairly recent full snapshot to restore from.  
As soon as it starts the restoration process, I get the error `TypeError: Failed to fetch`
[image]

Rebooting the HA instance at this stage causes it to never come back online again.

At this moment, I'm stuck without a working copy of HA

TheGroundZero, 2020-10-19:
Restored a VM back-up. Got it working again.

Won't be installing any updates to HA or rebooting any time soon.

ludeeus, 2020-10-20:
This has been fixed in the frontend

closed 2020-10-20 by ludeeus

djoeney, 2020-10-21:
I’ve got the same message as titled, what worked for me is doing a “ha core restart”. After a few minutes I was happy to see my backed up image again.

ManlyJack, 2020-10-23:
Why was this closed? It's still an issue.. I'm having it right now...

Pr0gr4mm3r, 2020-10-23:
Same issue here! The second I load a recent snapshot (0.115) into my fresh install (0.116), I get the message: "TypeError: Failed to fetch". Nothing else seems to happen anymore, so the restore fails. Rebooting messes up the install which forces me to start all over: using balenaEtcher to burn the Home Assistant image onto my SD card. Sigh...

bobvmierlo, 2020-10-24:
Same issue here, bricked my VM due to a full virtual disk. Trying to restore my snapshot from the onboarding page, but instantly get this error. After a while HA starts up but it's full of errors. The most important error seems to be that HA doesn't have internet connection so updating addon's and integrations that require internet access dont work

danlun, 2020-10-27:
I have the same exactly. Clicking the link to restore from the startup screen after a fresh install and after choosing my tar-file and a little while I get "Failed to fetch". The process stops....

bobvmierlo, 2020-10-27:
> Same issue here, bricked my VM due to a full virtual disk. Trying to restore my snapshot from the onboarding page, but instantly get this error. After a while HA starts up but it's full of errors. [...]

Addition to my last comment:
I've got it to work, the issue with HA not having internet connection was due to my router still having the old MAC address linked to the static IP of the hassio VM.
So although it shows the error the restore proces seems to work if you just wait long enough. But the error is strange so it needs fixing anyway

danlun, 2020-10-27:
Thanks bobvmierlo, that did help. I moved my tar-file from my network NAS to my PC (still in network) and for some reason it now worked and I got the question what to restore, chosed wipe and restore and after a while Im back up running. But I do agree, with the rror message delivered its hard to trust and just wait. Need a fix!

ManlyJack, 2020-10-27:
Never got it to work, tried about 4 times but in the end just had to make a new user and upload the snapshot file in the snapshot section and restore it from inside home assistant.

RichardvanGestel, 2020-11-09:
Same problem here.
HassOS 4.15 HomeAssistant 0.117.5
Supervisor 2020.11.0
"Upload snapshot" esults in an error message:  failed to fetch

Madseason41, 2020-11-17:
Same here, tried on firefox and chrome, wired and wireless.

alexbelgium, 2020-11-19:
same here with latest hassio image

chrissfoot, 2020-11-24:
This is happening to me with the latest 5.5 image. In fact, as long as I've been using HA it's never worked for me. I try on each new install and end up having to create the user and samba the backup in every time

alexbelgium, 2020-11-24:
Well I found a way to make it work. Here is what I do :
- clean install and do initial config
- install Google drive backup addon and download snapshot
- restore all addons, wait until its restored
- restore folders for ssh, share and local addons, wait until its restored, then reboot
- restore configuration and wait 10 min for reboot

mikewesner, 2022-08-20:
@ludeeus this issue should be reopened.  I just tried this use case today and the same issue exists.

apple047, 2023-01-05:
it appears the fault persists  identical symptoms experiencing now

TETZUO, 2023-02-23:
Issue is back again tried today
[image]

kitty99hub, 2023-04-06:
Same error for me today.... :(
Cannot restore backup on new HASSOS installation.

TooTallOne, 2023-08-22:
Another same error here post!  Claimed that upload failed and 15 minutes later system restored.  This has been an error 2020?