home-assistant/operating-system #920: RPi 3B+ wont boot / blank screen

FaserF, 2020-10-21:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I cant provide logs as the os wont even boot. Tried it with 2 different micro SD (one 32GB, one 16GB). With other OS (raspberry pi os for example) it works without any issues. Homeassistant OS wont even boot (first boot), no screen result with an HDMI cable plugged in and Ethernet was connected.
I have now tried the following versions:
- 5.3 (32 & 64bit)
- 4.14 (32 & 64 bit)
- 4.13 (32 bit)

I have upgraded the firmware to the latest and I tried formatting the micro SD multiple times.

agners, 2020-10-21:
Hm, I did test 4.14 on a Raspberry Pi 3 Model B+ just two days ago to reproduce the Bluetooth issue, so I am pretty sure that this image works.

Can you describe how you created the SD card?

The image is compressed, so it needs to be uncompressed and dd'ed onto the bare SD card (`/dev/sdx` without a number). Otherwise balenaEtcher can do this in one step.

FaserF, 2020-10-21:
I tried the following:
- Download the release from GitHub
- Install Balena Etcher (tried it on 3 different PC's)
- Choose the file from GitHub without uncompressing it as Etcher does that automatically
- flashes the OS onto the microSD

(I have also tried manually formatting it before using Etcher)

agners, 2020-10-21:
Do the SD cards show partitions on the PC? E.g. do you see the hassos-boot partition with `config.txt` etc on it?

Does the system seem to boot (e.g. is the green activity LED blinking after powering on?). If so, maybe the HDMI signal does not work properly with your Monitor, can you try a different one?

Otherwise a serial console might help to get more debug information. This requires a TTL UART adapter and a serial terminal emulator such as PuTTY. Then you can enable the serial console to see if it shows more information:
https://github.com/home-assistant/operating-system/tree/dev/Documentation/boards/raspberrypi#serial-console

FaserF, 2020-10-22:
Yes I can see the partitions on PC. And I can access the hassos-boot partition with config.txt on it.

The system does not seem to boot, no green activity LED after powering on and only a black screen -> so HDMI detects something.
And with other OSs it works (green activity LED + HDMI output + boot).

I will try and see if a friend of mine has such an adapter and then have a look.
Thanks in advance for your help.

FaserF, 2020-10-26:
Tried it now with the same microSD on a raspberrypi 4 -> Everything works fine with no issues.

agners, 2020-10-27:
The only reason I can think of is that U-Boot (used by HAOS but not used in Raspberry Pi OS) in combination with your micro SD and RPi Model 3+ causes issues.

I just tried `hassos_rpi3-4.15.img.gz` flashed on a SanDisk Extreme 32GB micro SD (UHS-I, V30/A1), running on Raspberry Pi 3 Model B+, and it is works fine.

FaserF, 2020-10-27:
Thanks for the Info. We have now Sold the raspberry pi 3 and bought a raspberry pi 4.
So the issue isn’t solved but it works fine with the raspberry pi 4

martinscholz83, 2020-11-28:
I got the same problem with my Pi3 B+. It seems to be a generell problem, because i tried one year ago. See here https://community.home-assistant.io/t/problems-with-hass-io-image-files-for-raspberry-pi/159005/15. I can any other raspberry OS except hassio.

flochtililoch, 2020-11-28:
Also got the issue with 4.18 and 4.17 - 4.16 booted just fine.

Nemesis24, 2020-11-29:
The same for me but on a raspberry pi 4B, i try 4.18, 4,17, 5.4, 5.5, 5.6... nothing work... 4.16, is ok, and after 
installation i can update to 4.17, I've got a raspBee on Gpio, on 4,16 the raspBee LEDs are off, zha don't found the raspBee....with 4.17 the raspBee LEDs are ok, zha don't found the Raspbee

rgreuel, 2020-12-31:
I was seeing the same issue on a pi 3B with 5.9 and 5.8. I finally tried 4.16 and I'm booting up fine.

elisaado, 2021-02-02:
4.16 doesn't seem to work for me.

rjdza, 2021-03-05:
> Thanks for the Info. We have now Sold the raspberry pi 3 and bought a raspberry pi 4.
> So the issue isn’t solved but it works fine with the raspberry pi 4

Closing the issue was a mistake, since it isn't resolved and is still affecting people.

ezankel, 2021-07-02:
This should not be closed. it is affecting me with any build after 4.15 (haven't tried 4.16). Can confirm that any 6.x build I have attempted to install also gives me a blank screen. 

This thread outlines the troubleshooting that has been done, with no resolution other than staying on 4.15 permanently for my 3B+
https://community.home-assistant.io/t/cli-not-starting-on-new-install/317855/35

elisaado, 2021-07-02:
I've given up on HassOS and am running it inside a docker on ArchARM on my Pi.

raidnet-ms, 2021-07-05:
I'm having the same issue on rap pi4 8gb

agners, 2021-07-29:
Actually, the original poster closed this issue, so I guess that means the problem has resolved for him.

Please test HAOS 6.2, and if you experience the problem create a new issue with as much details as possible (hardware setup, make and type of SD card etc.).

closed 2021-07-29 by agners

rjdza, 2021-09-10:
> 
> Actually, the original poster closed this issue, so I guess that means the problem has resolved for him.
> 

Actually, no.  He gave up and sold his RPi 3B and got an RPi 4.  So not resolved.

https://github.com/home-assistant/operating-system/issues/920#issuecomment-717502921

You lose the bonus points you would have gotten had you actually read the comments.

FWIW, I no longer care about this issue as I have given up on the project.  It seems nice and all, but issues and responses like this make it too unreliable.

alexd73, 2021-12-05:
I got the same issue with v6&v7. I decided to use a Supervised solution, but I think you should fix this problem. If you need more details I'm here for you

alexd73, 2021-12-06:
Perhaps this will be useful?

[image]

agners, 2021-12-06:
@alexd73 the `Cannot read property 'message' of null` error seems to be a problem the current version of balenaEtcher (see https://github.com/balena-io/etcher/issues/3648).

So far I was unable to reproduce the problem. I test images regularly on Raspberry Pi 3 Model B and Raspberry Pi 3 Modle B+. 

Just to make sure, I did exactly that with the latest balenaEtcher just now:

1. Downloaded the image from https://github.com/home-assistant/operating-system/releases/download/7.0/haos_rpi3-64-7.0.img.xz
2. Open balenaEtecher and select the `haos_rpi3-64-7.0.img.xz` image and the target SD card
3. Flashed the image
4. Put the SD card into Raspberry Pi 3 Model B+, plugged in Ethernet and Power
5. Wait for few minutes, access frontend on http://homeassistant:8123
6. Wait for another few minutes until preparation finished

So the 7.0 rpi3-64 image seems to work fine here.

Can you describe what exactly happens when you insert the SD card and power up the system? Is the green LED blinking at any time? Do you have any HDMI output? Can you try with just Ethernet and Power connected (no USB devices)?

MartinsTilaks, 2021-12-15:
I'am using Rpi 3 B+ and flashing os with Raspberry pi imager. After pluging in rpi with Home Assistant Os it boots for few seconds ( Green light blinking and on screen scrolling text), but soon screen turns black and no Green light. Tried different sd cards and 2 different rpi's models 3b+. Had it worked with rpi debian image and docker, but now trying with Home Assistant OS.

colohan, 2022-01-08:
FYI, I just encountered what I thought was the same issue.  Eventually I figured it out:  on the install instruction page is a not-immediately-obvious UI element:  a selector you have to use to choose the appropriate OS image for your hardware.  If you mistakenly flash the Raspberry Pi 4 OS onto an SDCard and try to boot it on a Raspberry Pi 3, you get the described symptoms.  If you flash the Raspberry Pi 3 OS, it seems to work.

ezankel, 2022-01-08:
> FYI, I just encountered what I thought was the same issue. Eventually I figured it out: on the install instruction page is a not-immediately-obvious UI element: a selector you have to use to choose [...]

Yes this is true, but it's not what many of us encountered. I know for me it happened with a fresh install of the correct image (I downloaded multiple times). It also happened when attempting to upgrade we l from with a previously functioning instance of HASS. Unless the HAOS itself was pulling the wrong install file, which is indeed a bug.

agners, 2022-01-10:
@colohan that of course explains why the screen stays black, thanks for sharing!

@ezankel I checked the community link you shared above, and there is a lot of issues in the same thread, some DNS related etc. If you (or anyone else) are still experience issues that a newly flashed `rpi3` or `rpi3-64` just leads to a empty black screen, please open a new issue so we can get to the bottom of this. I am locking this as the original poster moved on to new hardware.