home-assistant/operating-system #922: 5.3 64b development build boot from USB at Pi4 8G

kdelios, 2020-10-22:
**HassOS release with the issue:**
5.3 64b development build

5.x 64b development build is capable booting from USB3 using an SSD (Pi4). It is confirmed by using a few SSD’s and a SATA to USB3 adapter cable.

However, when using a NVMe SSD and a NVMe SSD Enclosure, it fails to boot from USB3 ports. It boots successfully from USB2 ports.

I am using the following:

https://www.amazon.com/Protection-Enclosure-Innovative-Connecting-Thunderbolt/dp/B07Q7H5QLX

https://www.amazon.co.uk/Patriot-P300-128GB-Low-Power-Consumption/dp/B0822Y6N1C

It looks like UASP is not supported by the development build so far.
I have tried to set the quirks in the CONFIG folder by editing the cmdline.txt file, but it doesn’t work.

The same NVMe set boots normally from USB3 at Raspbian.
Confirmed at Pi4 2G, and Pi4 8G.

Hopefully it will be fixed at the next releases.

agners, 2020-10-26:
I happen to have the very same ICY BOX with the JMS583 chipset. The problem seems power related: When I connect it via a powered USB 3.0 Hub, everything seems to work fine! See also: https://github.com/home-assistant/operating-system/issues/796#issuecomment-695828529

Can you try using a powered USB 3.0 hub? If you like to use an enclosure with a different chipset, e.g. RTL9210.

kdelios, 2020-10-27:
I am not sure that is "power related". Unfortunately I don't have a USB 3.0 hub available to test it. But it boots without issues with raspbian without a hub, so I am not convinced that power shortage is the problem here. In addition, I am using a Pi4 8G which has different hw and allowing more power to USB3 devices.

Is your USB 3.0 hub externally powered?

My setup consumes aprox. 1W@230VAC more (6W vs 5W) in comparison to a normal SSD, so NMVe is consuming more power (as expected). But this 0.2A@5VDC increase when using a 3A power supply is not big enough in my opinion. Even 6W@230VAC is aprox. 1.3A@5VDC.

I have no alternative to use HA on raspbian supervised for the time being.

agners, 2020-10-27:
I debugged quite a bit into it a month ago, and I realized that the JMS583 is violating a USB mass storage spec when connected directly to the RPi4. Read the details here: https://www.mail-archive.com/u-boot@lists.denx.de/msg384126.html

If it's power, its likely a [inrush problem](https://en.wikipedia.org/wiki/Inrush_current). My theory is that the SSD/JMS583. We are using U-Boot, it could be that it starts to access the SSD earlier than Raspbian does when using Linux, or that Linux is just more tolerant.

In any case, once I had it up, I noticed other storage related problems after pushing the system a bit. I then decided to just run this thing behind a USB Hub, because all problems disappeared :-)

It would be interesting to see if Raspbian remains stable with that SSD directly attached.

kdelios, 2020-10-27:
I run my production HA install supervised on Raspbian with the combo directly attached to USB3 for 40 days without issues so far. I seems stable enough.

ShurikenGitHub, 2020-11-16:
I have a simular issue on 5.5 64b with the Orico ORI-2139U3-CR SSD case and the Samsung 850 evo 250gb, USB2.0 boots fine but USB3.0 gets stuck at:
[image]
I'm using an Raspberry Pi 4B 4gb.
The weirdest thing of all is that the system booted succesfully once but when I reflashed it, it didn't work again.
The SSD with case works fine with RasPiOS so it's weird that it fails with HassOS. 
So if the issue was powerrelated it would be weird that it boots with RasPiOS.

BlackTurtle123, 2020-11-25:
This is one from my issues. Nvme 970 Samsung evo. Boots on usb2, USB 3 black screen.
I do use an external power source

agners, 2020-11-26:
@ShurikenGitHub this is due to the UAS incompatibility with Raspberry Pi. You have to add a `usb-storage.quirks` for your device. See: https://www.raspberrypi.org/forums/viewtopic.php?t=245931

BlackTurtle123, 2020-11-26:
> @ShurikenGitHub this is due to the UAS incompatibility with Raspberry Pi. You have to add a `usb-storage.quirks` for your device. See: https://www.raspberrypi.org/forums/viewtopic.php?t=245931

I did do this already for my external ssd. Followed that 2 or 3 days ago.

kdelios, 2020-12-20:
> @ShurikenGitHub this is due to the UAS incompatibility with Raspberry Pi. You have to add a `usb-storage.quirks` for your device. See: https://www.raspberrypi.org/forums/viewtopic.php?t=245931

Not the case in my setup. Did the quirks stuff at the beginning. No joy.
Today I did some additional tests:
Updated my Pi4 to the latest stable bootloader EEPROM (2020-12-11.bin).
My combo did not boot HassOS 5.9, not even from USB2. In addition, it refused to boot with Raspbian OS. Both OS did not boot with & without a powered USB3 hub.
Rolled back to the 2020-09-03.bin bootloader EEPROM.
My combo refused to boot HassOS 5.9 from USB3 (tested with and without a powered USB3 hub). However, boots successfully from USB2.  In addition, boots successfully Raspbian OS from USB3 (with and without a powered USB3 hub).
Also, the trick to plug the USB cable a few seconds after powering up the Pi4, doesn't work either.

kdelios, 2020-12-21:
Additional information regarding the  the latest stable bootloader EEPROM (2020-12-11.bin):

If I edit the e rpi-eeprom-config and add at the end of the list: USB_MSD_PWR_OFF_TIME=0, then the behavior is exactly the same with 2020-09-03.bin bootloader EEPROM:
Fails to boot HassOS 5.9 from USB3 (tested with and without a powered USB3 hub). However, boots successfully from USB2.
In addition, boots successfully Raspbian OS (and Debian OS) from USB3 (with and without a powered USB3 hub).

kdelios, 2021-01-11:
Tested again with the latest beta Pi4 bootloader (2021-01-05.bin) with the same same results:

Fails to boot HassOS 5.10 (64b) from USB3 (tested with and without a powered USB3 hub). However, boots successfully from USB2.
In addition, boots successfully Raspbian OS (and Debian OS) from USB3 (with and without a powered USB3 hub).

It is quite clear that the issue is within HassOS. No doubt.

kdelios, 2021-01-31:
Not possible to boot HassOS from NVMe/USB3 (JMS583 enclosure).
The only way to use the above mentioned combo, is leaving the operating system (HassOS) and the boot on the MicroSD card in the read only partition (or almost), and moving the data partition containing Home Assistant itself, the yaml files, the database, the logs and the add -on, on external disk, i.e. where all frequent writes take place, using the # datactl move /dev/sda command.
In addition, I was able to enable TRIM.

closed 2021-01-31 by kdelios

ur7x, 2021-03-02:
I can add that I have the exact same problem and experience, on both powered and unpowered USB3 -> NVMe adapters... Raspbian will boot  HASSio will not.  Would love to see this fixed in future releases.   Ironic that M.2 SATA drives will boot, but not the latest generation of SSDs.

ur7x, 2021-03-06:
Update! There is a solution for this. 
For the JMS583 based devices HASSio needs them to be on Firmware version 2.00.09... You can google that and find the tools and bin files needed to patch the adapters and then HASSio boots just fine.

kdelios, 2021-03-06:
> Update! There is a solution for this.
> For the JMS583 based devices HASSio needs them to be on Firmware version 2.00.09... You can google that and find the tools and bin files needed to patch the adapters and then HASSio boots just fine.

Good job!
I confirm that fw update is the solution. Updated to 02.00.09 and my combo boots fine HassOS.