home-assistant/operating-system #925: No soundcard VDMK 5.3

SilverWx, 2020-10-22:
**HassOS release with the issue:**

arch | x86_64
-- | --
chassis | vm
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 5.3
installation_type | Home Assistant OS
os_name | Linux
os_version | 5.4.63
python_version | 3.8.5
supervisor | 249
timezone | Europe/Paris
version | 0.116.4
virtualenv | false

**Supervisor logs:**

**Audio logs:**
```
I: [pulseaudio] main.c: Using state directory /var/lib/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: yes
W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please make sure that you actually do want to do that.
W: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
I: [pulseaudio] main.c: System supports high resolution timers
D: [pulseaudio] memblock.c: Using private memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
I: [pulseaudio] cpu-x86.c: CPU flags: CMOV MMX SSE SSE2 SSE3 SSSE3 SSE4_1 SSE4_2 MMXEXT 
I: [pulseaudio] svolume_mmx.c: Initialising MMX optimized volume functions.
I: [pulseaudio] remap_mmx.c: Initialising MMX optimized remappers.
I: [pulseaudio] svolume_sse.c: Initialising SSE2 optimized volume functions.
I: [pulseaudio] remap_sse.c: Initialising SSE2 optimized remappers.
I: [pulseaudio] sconv_sse.c: Initialising SSE2 optimized conversions.
D: [pulseaudio] database-tdb.c: Opened TDB database '/var/lib/pulse/08f8a9e86771403da24760009fc7636f-device-volumes.tdb'
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/var/lib/pulse/08f8a9e86771403da24760009fc7636f-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
D: [pulseaudio] database-tdb.c: Opened TDB database '/var/lib/pulse/08f8a9e86771403da24760009fc7636f-stream-volumes.tdb'
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/var/lib/pulse/08f8a9e86771403da24760009fc7636f-stream-volumes'.
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1 added for object /org/pulseaudio/stream_restore1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry0
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
D: [pulseaudio] database-tdb.c: Opened TDB database '/var/lib/pulse/08f8a9e86771403da24760009fc7636f-card-database.tdb'
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/var/lib/pulse/08f8a9e86771403da24760009fc7636f-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-connect" (index: #4; argument: "").
I: [pulseaudio] module-udev-detect.c: Found 0 cards.
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-13.0/modules/module-bluez5-discover.so': success
D: [pulseaudio] dbus-util.c: Successfully connected to D-Bus system bus 4427751def0f7504cc96717c5f920120 as :1.45
I: [pulseaudio] module.c: Loaded "module-bluez5-discover" (index: #7; argument: "").
I: [pulseaudio] module.c: Loaded "module-bluetooth-discover" (index: #6; argument: "").
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #8; argument: "auth-anonymous=1 auth-cookie-enabled=0 socket=/data/external/pulse.sock").
I: [pulseaudio] module-default-device-restore.c: No previous default sink setting, ignoring.
I: [pulseaudio] module-default-device-restore.c: No previous default source setting, ignoring.
I: [pulseaudio] module.c: Loaded "module-default-device-restore" (index: #9; argument: "").
I: [pulseaudio] module.c: Loaded "module-rescue-streams" (index: #10; argument: "").
D: [pulseaudio] module-always-sink.c: Autoloading null-sink as no other sinks detected.
I: [pulseaudio] sink.c: Created sink 0 "auto_null" with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] sink.c:     device.description = "Dummy Output"
I: [pulseaudio] sink.c:     device.class = "abstract"
I: [pulseaudio] sink.c:     device.icon_name = "audio-card"
I: [pulseaudio] source.c: Created source 0 "auto_null.monitor" with sample spec s16le 2ch 44100Hz and channel map front-left,front-right
I: [pulseaudio] source.c:     device.description = "Monitor of Dummy Output"
I: [pulseaudio] source.c:     device.class = "monitor"
I: [pulseaudio] source.c:     device.icon_name = "audio-input-microphone"
D: [null-sink] module-null-sink.c: Thread starting up
[... 34 lines cut ...]
I: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
D: [pulseaudio] bluez5-util.c: oFono is running: no
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPHSProfile 00001108-0000-1000-8000-00805f9b34fb
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci1
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci1
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci1
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci1
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0
I: [pulseaudio] client.c: Created 0 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 33, local 33
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 0 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.

```
**Kernel logs:**

**Description of problem:**
I use integrated realtek soundcard which I shared with VMwhare to the virtual machine. If I understand well, support for virtualsoundcard is ok since 5.0, or it is only pci soundcard ?

pvizeli, 2020-10-22:
You need to find out what is needed for kernel modules. We can show you after that where you can add it to make an PR

SilverWx, 2020-10-23:
ok, I will try to do that ^^
However, I have no idea if VMWare workstation expose directly the realtek soundcard or a virtual soundcard interface.

SilverWx, 2020-10-23:
[alsa-driver-RTv5.18.zip](https://github.com/home-assistant/operating-system/files/5428305/alsa-driver-RTv5.18.zip)
Here is the Linux driver for realtel card

agners, 2020-10-27:
Yeah the driver `CONFIG_SND_HDA_CODEC_REALTEK` should be enabled, and hence support this hardware. Can you check `dmesg` from the console?

SilverWx, 2020-10-28:
I will check that as soon I can :)

SilverWx, 2020-11-08:
sorry for the delay, here is my dmesg :
`[    0.000000] Linux version 5.4.63 (builder@93b0fc19d33c) (gcc version 9.3.0 (Buildroot -gc152fdeb)) #1 SMP Thu Oct 22 15:57:16 UTC 2020
[    0.000000] Command line: apparmor=1 security=apparmor console=tty1 bootchooser.active=A  root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait systemd.machine_id=08f8a9e86771403da24760009fc7636f zram.enabled=1 zram.num_devices=3
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000ef29fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ef2a000-0x000000000ef32fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ef33000-0x000000000ef41fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ef42000-0x000000000ef47fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ef48000-0x000000000ef60fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000ef61000-0x000000000ef65fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ef66000-0x000000000fee6fff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fee7000-0x000000000ff56fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000ff57000-0x000000000ff72fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000ff73000-0x000000000ff76fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000ff77000-0x000000005e7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x0c26b018-0x0c273057] usable ==> usable
[    0.000000] e820: update [mem 0x0c26b018-0x0c273057] usable ==> usable
[    0.000000] e820: update [mem 0x0c269018-0x0c26a857] usable ==> usable
[    0.000000] e820: update [mem 0x0c269018-0x0c26a857] usable ==> usable
[    0.000000] e820: update [mem 0x0c266018-0x0c268057] usable ==> usable
[    0.000000] e820: update [mem 0x0c266018-0x0c268057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000001000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000000c266017] usable
[    0.000000] reserve setup_data: [mem 0x000000000c266018-0x000000000c268057] usable
[    0.000000] reserve setup_data: [mem 0x000000000c268058-0x000000000c269017] usable
[    0.000000] reserve setup_data: [mem 0x000000000c269018-0x000000000c26a857] usable
[    0.000000] reserve setup_data: [mem 0x000000000c26a858-0x000000000c26b017] usable
[    0.000000] reserve setup_data: [mem 0x000000000c26b018-0x000000000c273057] usable
[    0.000000] reserve setup_data: [mem 0x000000000c273058-0x000000000ef29fff] usable
[    0.000000] reserve setup_data: [mem 0x000000000ef2a000-0x000000000ef32fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000ef33000-0x000000000ef41fff] usable
[    0.000000] reserve setup_data: [mem 0x000000000ef42000-0x000000000ef47fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000ef48000-0x000000000ef60fff] usable
[    0.000000] reserve setup_data: [mem 0x000000000ef61000-0x000000000ef65fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000ef66000-0x000000000fee6fff] usable
[    0.000000] reserve setup_data: [mem 0x000000000fee7000-0x000000000ff56fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000ff57000-0x000000000ff72fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000000ff73000-0x000000000ff76fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000ff77000-0x000000005e7fffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] efi: EFI v2.31 by VMware, Inc.
[    0.000000] efi:  SMBIOS=0xff74000  ACPI 2.0=0xff68000  MEMATTR=0xfd91018 
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.16221537.B64.2005150253 05/15/2020
[    0.000000] vmware: hypercall mode: 0x00
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 3493.420 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000005] vmware: using sched offset of 6494281395 ns
[    0.000009] tsc: Detected 3493.420 MHz processor
[    0.010364] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.010382] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.010390] last_pfn = 0x5e800 max_arch_pfn = 0x400000000
[    0.010410] MTRR default type: uncachable
[    0.010411] MTRR fixed ranges enabled:
[    0.010413]   00000-9FFFF write-back
[    0.010414]   A0000-FFFFF uncachable
[... 1450 lines cut ...]
[... cut ...]
[78425.267170] audit: type=1325 audit(1604489613.639:211): table=nat family=2 entries=11
[78425.269224] audit: type=1325 audit(1604489613.642:212): table=nat family=2 entries=12
[78425.283177] eth0: renamed from veth326ed16
[78425.287277] IPv6: ADDRCONF(NETDEV_CHANGE): veth9af648b: link becomes ready
[262071.173870] udevd[189]: starting version 3.2.9
[262071.185653] udevd[190]: starting eudev-3.2.9
[296426.396838] udevd[192]: starting version 3.2.9
[296426.405110] udevd[193]: starting eudev-3.2.9
`

agners, 2020-11-24:
Hm, given that the  Realtek driver really is there, its more likely that VMware actually emulates a different device rather than passing/emulating the one from the host. From that post, it seems this is configurable: https://oliveirapaulo.com/2016/08/03/fix-no-sound-in-xubuntu-virtual-machine/

Can you maybe try editing the vmx file and make sure `sound.virtualDev` is set to `"hdaudio"`

SilverWx, 2020-11-27:
> Hm, given that the Realtek driver really is there, its more likely that VMware actually emulates a different device rather than passing/emulating the one from the host. From that post, it seems this [...]

Thank you for finding that ! It make sense, I will try asap

SilverWx, 2020-11-27:
It seems to work, thank you again ! I feel stupid I didn't t think about that :p
[image]

closed 2020-11-27 by SilverWx

agners, 2020-11-27:
@SilverWx thanks for the update, glad this fixes it!