home-assistant/operating-system #928: With HassOS 5.4 Raspberry 3 does not boot anymore

TheZoker, 2020-10-23:
**HassOS release with the issue:** 5.4

**Supervisor logs:** None

**Journal logs:** None

**Kernel logs:** None

**Description of problem:**
When I upgraded from HassOS 5.3 to 5.4 today, my Pi 3B+ does not boot anymore. When I connect a HDMI cable, the only output I get is the colored image. I reflashed the SD card with 5.4 again to see if it was a upgrade error, but still same issue. I then flashed version 5.3 and the Pi booted normal again. Unfortunately I can't attach any logs, since there where just no logs anywhere. So I assume that something in 5.4 broke something for the Pi3 (maybe other Pi versions, but I only have version 3 to test here)

archerne, 2020-10-23:
Same issue with the 64bit Rasp Pi 4 8GB

archerne, 2020-10-23:
I did a fresh install of 5.4 and it started (still testing to make sure everything works) but looks like the upgrade from 5.3 to 5.4 breaks something

SmarterHomeLife, 2020-10-23:
Seems to break something with the SD card, I upgraded my system that runs on an SSD with no issues but upgrading or a fresh install on an SD card fails to boot for me.

SeanPM5, 2020-10-24:
Can confirm, my secondary instance on an RPi4 (64bit / 4GB) was bricked after updating from 5.3 to 5.4.  It can no longer boot off the SD card.

fabiocastagnino, 2020-10-24:
Same issue with the 64bit Rasp Pi 4 8GB, upgrading from 5.2 to 5.4 or clean install. It can no longer boot off the SD card.

Gadgit83, 2020-10-24:
Same for me - RPI 4 8GB. 64bit image. Rolled back to 4.15 and worked fine. 5.3 appears to be booting OK.

Thought it was me - first time I'm trying the Pi with HA. Tried everything, different USB cables, different power supply, reflashed card, used balenEtcher and the Pi Imager, no luck. Should have known better than to rely on a dev version :).

cogneato, 2020-10-25:
I wasn't sure what "colored image" was referring to, but now I see.

TheZoker, 2020-10-25:
Thanks @cogneato for the hint, I placed an example image in the issue description 🙂

selmaohneh, 2020-10-25:
Same here. But my screen is not colored, it just stays black. 🙄

cogneato, 2020-10-25:
@TheZoker That was more me confirming what you were describing. I have a picture I was going to upload too but yours is straighter than mine. :+1: 

I've since booted a fresh install of HA OS 5.3 64 bit without issues on this pi4 4GB. We probably won't hear back from anyone about 5.4 for another 12 hours at least.

agners, 2020-10-25:
Hm, it seems that the firmware bump in #899 is responsible for this. Using `start4.elf` and `fixup4.dat` from the firmware we used before (https://github.com/raspberrypi/firmware/commit/7caead9416f64b2d33361c703fb243b8e157eba4) makes it work for me.

I guess we have to revert the firmware until we bump to Linux 5.4.

Jpsy, 2020-10-26:
Can confirm that update from 5.3 to 5.4 works flawlessly on an SSD based system without SD card.

Beduir, 2020-10-26:
I have successfully upgraded from 5.3 to 5.4 (SSD without SD card). But the Razberry board stopped working. I went back to 5.3 and everything works again.

Jpsy, 2020-10-26:
@Beduir:  Did your system stall immediately after upgrade or did it work for some time before it stopped?
I am on 5.4 for about 20 hours now without any problems.

fabiocastagnino, 2020-10-26:
in my case the system die immediately after upgrade (RPi4 8gb)

Beduir, 2020-10-26:
@Jpsy 
Everything worked well except Razberry (Z-wave).

Jpsy, 2020-10-26:
@Beduir: Aaaah, I got you wrong. Did not check that Razberry is a Z-wave Pi hat!

archerne, 2020-10-26:
@Beduir I had an issue with my Raspbee 2 hat as well, the serial changed from /dev/ttyAMA0 to /dev/ttyS0, once I made that change the hat worked again

Beduir, 2020-10-27:
@archerne 
It does not help.

Yeelight lamps also stop working. The reason is not yet clear.

agners, 2020-10-27:
Reverted to the firmware used in 5.3, should be working again with the next release.

closed 2020-10-27 by agners

tmm1, 2020-10-27:
Sorry for all the trouble here. I'm not quite sure what happened, and why only some RPI are affected. I will try to investigate more.

tmm1, 2020-10-30:
I think this issue is related to https://github.com/raspberrypi/firmware/issues/1457

The Sep 2020 stable release does not support booting from hybrid GPT from SD card. This was fixed in subsequent beta firmware released after the Sep 2020 stable build. The last known working firmware for hybrid GPT before Sep 2020 is Jun 2020 (1.20200601), which is what was being used before.

deftdawg, 2020-11-02:
Booting 5.4 from didn't work for me either on Pi4 4GB.   @agners unless the devs actually did something that fixes the issue, the issue should be reopened; otherwise a lot of users gonna suffer when it goes GA.