home-assistant/operating-system #931: HA OS 5.x reports only updates for 4.x releases

Jpsy, 2020-10-25:
**HassOS release with the issue:**

5.2, 5.3

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
_Frontend -> Supervisor_ on an OS 5.x based system reports the available updates for the 4.x release series, but not the updates for the 5.x series.
The following screenshot was taken more than 3 days after OS 5.4 was released. So this is not a timing problem but a real HA problem. 
[image]
I am not sure though, whether _HA OS -> issues_ is the right place to report this issue as it could be a mere frontend problem.

ludeeus, 2020-10-25:
This happens because 5.x is on the dev channel only, and you are on the beta channel.
<https://version.home-assistant.io/stable.json>
<https://version.home-assistant.io/beta.json>
<https://version.home-assistant.io/dev.json>

4.15 is the latest version of the beta channel.

However, this is "fixed" in the supervisor by hiding those "updates" and will be a part of the next supervisor release.

closed 2020-10-25 by ludeeus

Jpsy, 2020-10-25:
Thanks for the details @ludeeus. Is there a way in Home Assistant (Hass.io install) to switch to dev channel?

ludeeus, 2020-10-25:
You can use the CLI `ha su options --help`, but only do that if it's a secondary unit.

Jpsy, 2020-10-26:
With 5.x I am on a dev OS anyway. So is there any additional risk to set the Supervisor to dev channel?
Would this change more than just the releases that are displayed? Wouldn't I still have to manually trigger the update to a new dev release?