home-assistant/operating-system #932: RPI3b Update from HassOS 4.14 to 4.15 ERROR Device or resource busy

Villa-Buenavista, 2020-10-25:
Update fails again and again with error message in system logs:

20-10-25 11:09:48 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-10-25 11:13:41 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-10-25 11:17:01 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.15.raucb
20-10-25 11:17:01 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
20-10-25 11:17:20 INFO (MainThread) [supervisor.utils.gdbus] Stopping dbus monitor on de.pengutronix.rauc
20-10-25 11:17:20 ERROR (MainThread) [supervisor.hassos] Home Assistant Operating System update failed with: Installation error: Failed updating slot rootfs.1: Opening output device /dev/disk/by-partlabel/hassos-system1 failed: Device or resource busy

System Info / Frontend:
Home Assistant 0.116.4
Frontend version: 20201001.2 - latest 

System Health:
  | armv7l
-- | --
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.14
installation_type | Home Assistant OS
os_name | Linux
os_version | 4.19.127-v7
python_version | 3.8.5
supervisor | 249
timezone | Europe/Berlin
version | 0.116.4
virtualenv | false

Any help would be highly appreciated. Thanks!

agners, 2020-10-27:
Hmm, this looks very similar to #916.

Can you try using `ha os update` from the console?

Villa-Buenavista, 2020-10-27:
Hi, unfortunately no, tried that before.

[image]

warrendt, 2020-10-27:
> Hi, unfortunately no, tried that before.
> 
> [image]

I have this same error, using a Proxmox virtual machine, no raspberry,

agners, 2020-10-27:
@Villa-Buenavista  can you also run `ha supervisor logs` after that command? Is it showing the same issue?

Villa-Buenavista, 2020-10-27:
> 
> 
> @Villa-Buenavista can you also run `ha supervisor logs` after that command? Is it showing the same issue?

Unfortunately, yes:
[image]

agners, 2020-10-27:
Can you SSH to the host OS (see https://developers.home-assistant.io/docs/operating-system/debugging) and run the following after an update attempt:

```
ha > login
# journalctl -u rauc
```

warrendt, 2020-10-28:
> Can you SSH to the host OS (see https://developers.home-assistant.io/docs/operating-system/debugging) and run the following after an update attempt:
> 
> ```
> ha > login
> # journalctl -u rauc
> ```

[image]

Nothing out of the ordinary ?

Villa-Buenavista, 2020-10-28:
> 
> 
> Can you SSH to the host OS (see https://developers.home-assistant.io/docs/operating-system/debugging) and run the following after an update attempt:
> 
> ```
> ha > login
> # journalctl -u rauc
> ```

I am trying but am clearly lost. I am in terminal using:
ssh root@hassio -p 22222

but get: Connection refused.

Is  this still relevant? Then i would need more hints as to what to do or is warrendt´s answer sufficient?

Villa-Buenavista, 2020-10-28:
I just tried again after the new supervisor version came in (# 2020.10.0). But unfortunately with the same result. 
20-10-28 15:03:13 INFO (SyncWorker_5) [supervisor.docker.interface] Restarting homeassistant/raspberrypi3-homeassistant
20-10-28 15:03:33 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
20-10-28 15:04:08 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-10-28 15:05:17 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
20-10-28 15:07:15 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-10-28 15:07:47 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.15.raucb
20-10-28 15:07:48 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
20-10-28 15:10:53 INFO (MainThread) [supervisor.utils.gdbus] Stopping dbus monitor on de.pengutronix.rauc
20-10-28 15:10:53 ERROR (MainThread) [supervisor.hassos] Home Assistant Operating System update failed with: Installation error: Failed updating slot rootfs.1: Opening output device /dev/disk/by-partlabel/hassos-system1 failed: Device or resource busy

agners, 2020-10-28:
@warrendt this looks like a successful update indeed. Is the system log showing anything concerning now?

@Villa-Buenavista no, @warrendt's log looks actually good. If you still can reproduce the issue, it would be interesting to know what rauc is doing underneath (RAUC is the update system). Connection refused indicates that you don't have the SSH key setup. Please check https://developers.home-assistant.io/docs/operating-system/debugging on how to do that.

Villa-Buenavista, 2020-10-29:
OK. Will try again tonight with some more time.

warrendt, 2020-10-30:
> @warrendt this looks like a successful update indeed. Is the system log showing anything concerning now?
>
> @Villa-Buenavista no, @warrendt's log looks actually good. If you still can reproduce the [...]

Is there perhaps a way to put the file manually ? 
https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_ova-4.15.raucb
Where do I put it? 
I have a feeling it may be taking too long to download the file? I have been having some issues with GitHub being super slow from my fibre. I am troubleshooting with my ISP

Villa-Buenavista, 2020-10-30:
Have not got around to checking that rauc log. Will do it during the weekend. 

But have in the meantime updated to 0.117 & 0.117.1 w/o problems. Probably unrelated but I now got a different supervisor error after trying to update OS:

20-10-30 11:25:04 INFO (MainThread) [supervisor.homeassistant.core] Successful started Home Assistant 0.117.1
20-10-30 11:25:04 INFO (SyncWorker_1) [supervisor.docker.interface] Cleanup images: ['homeassistant/raspberrypi3-homeassistant:0.117.0']
20-10-30 11:56:18 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-10-30 11:56:18 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-10-30 11:56:25 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 0 new - 0 remove
20-10-30 12:56:15 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-10-30 13:00:27 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-10-30 13:01:10 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.15.raucb
20-10-30 13:01:11 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
20-10-30 13:01:27 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-10-30 13:01:28 INFO (MainThread) [supervisor.utils.gdbus] Stopping dbus monitor on de.pengutronix.rauc
20-10-30 13:01:28 ERROR (MainThread) [supervisor.hassos] Home Assistant Operating System update failed with: Installation error: Failed verifying checksum: Read from /run/rauc/bundle/rootfs.img failed: Input/output error
20-10-30 13:01:28 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-52873' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:131> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 170, in update
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError
20-10-30 13:01:56 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.15.raucb
20-10-30 13:01:56 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
20-10-30 13:22:42 INFO (MainThread) [supervisor.host.info] Updating local host information
20-10-30 13:22:43 INFO (MainThread) [supervisor.host.services] Updating service information
20-10-30 13:22:43 INFO (MainThread) [supervisor.host.network] Updating local network information
20-10-30 13:22:44 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
Failed to load cookie file from cookie: No such file or directory
20-10-30 14:52:00 INFO (MainThread) [supervisor.snapshots] Found 9 snapshot files
20-10-30 14:56:15 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-10-30 14:56:25 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-10-30 14:56:25 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-10-30 14:56:28 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 0 new - 0 remove
20-10-30 15:29:24 INFO (MainThread) [supervisor.host.info] Updating local host information
20-10-30 15:29:25 INFO (MainThread) [supervisor.host.services] Updating service information
20-10-30 15:29:25 INFO (MainThread) [supervisor.host.network] Updating local network information
20-10-30 15:29:26 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information

warrendt, 2020-11-02:
> @warrendt this looks like a successful update indeed. Is the system log showing anything concerning now?
>
> @Villa-Buenavista no, @warrendt's log looks actually good. If you still can reproduce the [...]

I have seemed to come out ok here, think it was a timeout on the git download. Thanks :)

Villa-Buenavista, 2020-11-03:
I updated the supervisor which failed at first. I received an IP error (172.30.32.2:80 unavailable or block) similar to this:
https://github.com/home-assistant/supervisor/issues/1576

As described there pulling the power supply & booting up fixed it (for now). The new supervisor came up and I was able to update the OS as well. I have no idea if there is a logical connection but wanted to document it here. 

As I am now running OS 4.15 I cannot provide further logs. Thanks for the support!

[image]

closed 2021-05-12 by stale[bot]