home-assistant/operating-system #940: Raspberry Pi Bluetooth Crashing

Veldkornet, 2020-10-27:
**HassOS release with the issue:**

4.15

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
It seems that after a period of time of usage, the Bluetooth becomes unstable and crashes. This is noticable when using the room-assistant add-on. 
```
10/25/2020, 11:23:39 AM - error - BluetoothClassicService: Command failed: hcitool -i hci0 cc "64:a2:f9:f1:b8:7b" && hcitool -i hci0 rssi "64:a2:f9:f1:b8:7b"
Invalid device: Network is down
10/25/2020, 11:23:45 AM - error - BluetoothClassicService: Command failed: hcitool -i hci0 cc "88:63:DF:8A:1B:DC" && hcitool -i hci0 rssi "88:63:DF:8A:1B:DC"
Invalid device: Network is down
```

See also https://github.com/mKeRix/room-assistant/issues/314

agners, 2020-10-27:
What environment is that exactly? can you fill out the sections please?

Veldkornet, 2020-10-27:
Updated version info

Veldkornet, 2020-10-27:
I initially thought that it was just the Bluetooth Crashing, but it seems that all my USB dongles also stop working. 

See below from the OoenZWave add-on

```
[20201027 12:42:10.788 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/844425682796561/" 
[20201027 12:42:10.788 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1125900659507220/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1125900659507220/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1407375636217878/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1407375636217878/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1688850612928536/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1688850612928536/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1970325589639192/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/1970325589639192/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/2251800566349844/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/2251800566349844/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/2533275543060502/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/115/value/2533275543060502/" 
[20201027 12:42:10.789 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/128/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/128/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/128/value/744488977/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/128/value/744488977/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/132/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/132/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/281475729653779/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/281475729653779/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/562950706364435/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/562950706364435/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/844425683075091/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/844425683075091/" 
[20201027 12:42:10.790 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/1125900659785747/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/1125900659785747/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/752943123/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/132/value/752943123/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/134/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/134/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/134/value/752975895/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/134/value/752975895/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/134/value/281475729686551/" 
[20201027 12:42:10.791 CET] [ozw.mqtt.publisher] [debug]: Removing Stale Topic/Msg:  "OpenZWave/1/node/44/instance/1/commandclass/134/value/281475729686551/" 
[20201027 12:42:10.792 CET] [ozw.mqtt.publisher] [debug]: Topics:  "OpenZWave/1/node/44/instance/1/commandclass/134/value/562950706397207/" 
[... cut ...]
[20201027 12:42:26.580 CET] [ozw.daemon] [info]: QT-OpenZWave Version:  "1.2.0" 
[20201027 12:42:26.580 CET] [ozw.daemon] [info]: QT Version:  5.12.5 
1603798946: Socket error on client qt-openzwave-1, disconnecting.
In exit
[cont-finish.d] executing container finish scripts...
[cont-finish.d] mqtt.sh: executing... 
1603798946: mosquitto version 1.6.8 terminating
1603798946: Saving in-memory database to /data/mosquitto.db.
[12:42:27] INFO: Ensure upstream MQTT server has the correct OZW status
```

agners, 2020-10-27:
Hm, can you check kernel messages using `dmesg`?

Veldkornet, 2020-10-27:
Hmm, currently all I get is the below
[image]

Veldkornet, 2020-10-27:
FYI, I discovered that I can get it working again by issuing the following command:

`bluetoothctl power on`

Veldkornet, 2020-10-28:
Additionally, I discovered the following in the dmesg when the bluetooth and USB dongles all crashed. Surprisingly I could get the bluetooth up and running again with the command form above, but for the USB dongels I needed to reboot the Pi.

```
[78888.068025] ftdi_sio ttyUSB0: failed to get modem status: -71
[78888.068497] ftdi_sio ttyUSB0: error from flowcontrol urb
[78888.072631] ftdi_sio ttyUSB0: ftdi_set_termios FAILED to set databits/stopbits/parity
[78888.073133] ftdi_sio ttyUSB0: ftdi_set_termios urb failed to set baudrate
[78888.073622] ftdi_sio ttyUSB0: failed to set flow control: -71
[78888.080910] ftdi_sio ttyUSB0: failed to set flow control: -71
[78888.192974] cdc_acm 1-1.4.3:1.0: failed to set dtr/rts
[78897.885365] ftdi_sio ttyUSB0: failed to get modem status: -71
[78897.885945] ftdi_sio ttyUSB0: error from flowcontrol urb
[78921.318248] udevd[190]: starting version 3.2.9
[78921.355004] udevd[191]: starting eudev-3.2.9
[78983.052316] udevd[530]: worker [2675] /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4 is taking a long time
[78983.055293] udevd[191]: worker [196] /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4 is taking a long time
[78983.055528] udevd[207]: worker [5692] /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4 is taking a long time
[79041.540720] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[79041.540756] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[79103.400388] udevd[207]: worker [5692] /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4 timeout; kill it
[79103.400442] udevd[207]: seq 8298 '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4' killed
[79103.403491] udevd[207]: worker [5692] terminated by signal 9 (Killed)
[79103.403530] udevd[207]: worker [5692] failed while handling '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4'
[79103.404977] udevd[530]: worker [2675] /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4 timeout; kill it
[79103.405094] udevd[530]: seq 8298 '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4' killed
[79103.406603] udevd[191]: worker [196] /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4 timeout; kill it
[79103.406665] udevd[191]: seq 8298 '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4' killed
[79103.407185] udevd[530]: worker [2675] terminated by signal 9 (Killed)
[79103.407212] udevd[530]: worker [2675] failed while handling '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4'
[79103.408291] udevd[191]: worker [196] terminated by signal 9 (Killed)
[79103.408328] udevd[191]: worker [196] failed while handling '/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4'
[79104.309143] cdc_acm 1-1.4.3:1.1: urb 3 failed submission with -22
[79125.222482] cdc_acm 1-1.4.3:1.1: urb 3 failed submission with -22
[79167.466752] cdc_acm 1-1.4.3:1.1: urb 3 failed submission with -22
[79250.228134] cdc_acm 1-1.4.3:1.1: urb 3 failed submission with -22
[79333.030947] cdc_acm 1-1.4.3:1.1: urb 3 failed submission with -22
```

agners, 2020-10-28:
Ugh, this looks like quite some USB issues. Hard to tell what is going on, from bad cables, bad USB device to power issues pretty much everything is possible. I suggest to remove one device and see if it makes bluetooth stable again (and dmesg not showing such errors)....

agners, 2020-11-06:
CDC ACM crash is addressed in the upcoming 4.16 release.

closed 2020-11-06 by agners

Veldkornet, 2020-11-11:
@agners was this specific to the RaspberryPi? I switched over to am Intel NUC, but the Bluetooth still dies on me every so often.