home-assistant/operating-system #941: Error updating OS from 4.13 to 4.15: signature verification failed

mrgingles, 2020-10-27:
Hi everyone!
I'm trying to update from OS 4.13 to 4.15, but I'm getting this error:
`20-10-27 07:57:46 ERROR (MainThread) [supervisor.hassos] Home Assistant Operating System update failed with: signature verification failed: error:2E09A09E:CMS routines:CMS_SignerInfo_verify_content:verification failure`

**HassOS release with the issue:**
Home Assistant 0.115.2

**Supervisor**
249

Can not use the terminal because of a `execvp failed: bas message error` and can not install a new terminal from the add-on store due to a `500` error. I neither can connect using ssh due to a `Too many authentication failures. Disconnected from 192.168.1.40 port 22` error (really don't think there are too many auth failures).

It looks like there is another problem somewhere with my environment.

Thank you for your help!

agners, 2020-10-27:
It looks like the update corrupted during transmission or when reading from the storage. What storage are you using?

mrgingles, 2020-10-27:
> It looks like the update corrupted during transmission or when reading from the storage. What storage are you using?

I'm using an SD card (still 48% memory free) on a Raspberry 3 model B.

agners, 2020-10-27:
This is reproducible after a reboot? Maybe a firewall or similar preventing the download from working/completing?

mrgingles, 2020-10-28:
I rebooted through the admin panel but Home Assistant didn't came up again. Then I unplugged the power in order to hard-reboot my Raspberry and tried the update again.

This is the output inside the logs:

```
20-10-28 11:44:41 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-10-28 11:50:13 INFO (MainThread) [supervisor.hassos] OTA update is downloaded on /data/tmp/hassos-4.15.raucb
20-10-28 11:50:13 INFO (MainThread) [supervisor.utils.gdbus] Starting dbus monitor on de.pengutronix.rauc
20-10-28 11:50:13 INFO (MainThread) [supervisor.utils.gdbus] Stopping dbus monitor on de.pengutronix.rauc
20-10-28 11:50:46 ERROR (MainThread) [supervisor.hassos] Home Assistant Operating System update failed with: Failed to check bundle identifier: Invalid identifier. Did you pass a valid RAUC bundle?
20-10-28 11:50:46 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-4204' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:127> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 166, in update
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError
```

Thank you!

dodavies, 2020-11-01:
I have the same problem same error messages, worked on one of my pi's but not on the other, repeated attempts

dodavies, 2020-11-02:
20-11-02 10:55:55 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-11-02 11:05:56 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb: 
20-11-02 11:05:56 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-331' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:131> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 142, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 92, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

ludeeus, 2020-11-02:
That is not the same issue @dodavies, you are reaching the timeout threshold of 10minuttes, join the #homeassistant_os channel on Discord if you need help with troubleshooting that,

dodavies, 2020-11-08:
After about fifty attempts this finally worked

closed 2021-05-12 by stale[bot]