home-assistant/operating-system #944: Cant upgrade to 4.15

terencevs, 2020-10-28:
I thought I would log this as a new issue as I have noticed a few people logging the issue off other peoples issues which are not the same (I am guilty of this myself)

**HassOS release with the issue: 4.13**

**Supervisor logs:**
When doing update from frontend:
`
20-10-26 18:29:19 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-64-4.15.raucb: 20-10-26 18:29:19 ERROR (MainThread) [asyncio] Task exception was never retrieved future: <Task finished name='Task-595' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:127> exception=HassOSUpdateError()> Traceback (most recent call last): File "/usr/src/supervisor/supervisor/hassos.py", line 138, in update int_ota = await self._download_raucb(version) File "/usr/src/supervisor/supervisor/hassos.py", line 88, in _download_raucb raise HassOSUpdateError() supervisor.exceptions.HassOSUpdateError
`

When doing update from terminal:
`
20-10-28 06:30:30 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-64-4.15.raucb
20-10-28 06:40:30 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-64-4.15.raucb: 
`

**Description of problem:**
Above errors happen when trying to update from Frontend and via termainal using hass os update.

I have tried to wget https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-64-4.15.raucb from terminal and this works. 

Also tried rebooting Raspberry Pi.
There is enough space.

agners, 2020-10-28:
This seems to be a supervisor issue. What supervisor version are you using?

terencevs, 2020-10-29:
[image]

XxCabbyoO, 2020-10-29:
i have the same error since a few month for every OS Update, for every superuser version. and there is still no fix 👎 

same Issue: https://github.com/home-assistant/operating-system/issues/617

terencevs, 2020-10-30:
Interesting enough, I tried again late last night and it worked.

This after trying for over a week multiple times a day.

dodavies, 2020-11-02:
20-11-02 10:55:55 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb
20-11-02 11:05:56 WARNING (MainThread) [supervisor.hassos] Can't fetch versions from https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi3-4.15.raucb: 
20-11-02 11:05:56 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-331' coro=<HassOS.update() done, defined at /usr/src/supervisor/supervisor/hassos.py:131> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/hassos.py", line 142, in update
    int_ota = await self._download_raucb(version)
  File "/usr/src/supervisor/supervisor/hassos.py", line 92, in _download_raucb
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

Robertoasd, 2020-11-03:
Exact same problem here on my pi-4. 

Downloading the file times out. time out of the update process is 10 minutes but downloading the file from the shell takes 26 minutes. So at my PI it will always fail. Downloading it on my laptop on the same network takes seconds.

```
~ $ time wget https://github.com/home-assistant/operating-system/releases/download/4.15/hassos_rpi4-4.15.raucb
Connecting to github.com (140.82.118.4:443)
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (52.216.114.27:443)
saving to 'hassos_rpi4-4.15.raucb'
hassos_rpi4-4.15.rau 100% |*************************************************************************************************************************************************************************************************************************************************************************************| 90.8M  0:00:00 ETA
'hassos_rpi4-4.15.raucb' saved

real    26m5.204s
user    0m0.218s
sys     0m2.310s
~ $ 
```

Downloading the huge ubuntu ISO takes a few minutes.

```
 ~ $ time wget https://releases.ubuntu.com/18.04/ubuntu-18.04.5-desktop-amd64.iso
Connecting to releases.ubuntu.com (91.189.88.248:443)
saving to 'ubuntu-18.04.5-desktop-amd64.iso'
ubuntu-18.04.5-deskt 100% |*************************************************************************************************************************************************************************************************************************************************************************************| 2091M  0:00:00 ETA
'ubuntu-18.04.5-desktop-amd64.iso' saved

real    3m5.690s
user    0m0.472s
sys     0m16.516s
```

Putting the file on the root of a USB-stick named it CONFIG and then booting does not work for me either. So I am stuck at 4.12.

Is there anybody who knows how to solve this? Extending the 10 minute timeout would be a great temporary solution to start with.

Robertoasd, 2020-11-05:
I just love mysteries! After giving updating at least 40 tries I am finally at 4.15 from 4.13!

terencevs, 2020-11-05:
> I just love mysteries! After giving updating at least 40 tries I am finally at 4.15 from 4.13!

Yeah same experience... Love a good mystery!

XxCabbyoO, 2020-11-06:
Same case with me, at some point it worked. But I have felt rather 100x on update clicked in the last weeks

I rather think that something has been fixed by the Homeassitant site?

The error will probably reappear with the next OS version Update :(

fabaff, 2020-11-09:
I see this issue on multiple remote installations as well. The owners are telling me the same things that were already mentioned in this issue.

Clicking again and again and putting hope in it does not solve the issue.

closed 2021-05-12 by stale[bot]