home-assistant/operating-system #945: SSD only booting via USB 2.0. USB 3.0 Fails.

Teleportist, 2020-10-31:
HassOS 5.3
Home Assistant 0.116.4

RPi 4B 4GB Rev 1.2 (Firmware upgrade/latest bootloader - September 03/09/2020)

Eluteng USB to SSD Adapter
https://amzn.to/3mFA6jT

Kingston A400 120GB SSD
https://amzn.to/3kMf0Qb

Both adapter and drive are well tested and verified according to this compatibility list.
https://storage.jamesachambers.com/popular/

HassOS will only boot when the SSD drive is connected to either a USB 2.0 port, or when connected to a USB 3.0 port via a USB extension cable. This is the case regardless of connecting the RPi via ethernet or wreless.

agners, 2020-10-31:
What do you mean exactly by "via a USB extension cable"? It sound the other cable is bad then?

Teleportist, 2020-11-01:
> What do you mean exactly by "via a USB extension cable"? It sound the other cable is bad then?

Placing a USB (2.0) extension cable of 0.5m between the RPi and the SSD adapter lead+SSD. The idea was to eliminate any interference caused by close proximity, as the SSD adapter lead is only ~120mm long.

agners, 2020-11-02:
OK I see now, a USB 2.0 extension cable of course makes it act as a USB 2.0 device.

We do use U-Boot whereas RPi OS does not, so it could be that U-Boot has troubles enumerating this device.

So in the USB 3.0 case, is anything showing on the HDMI display? Do you have access to a USB to UART TTL adapter to access the serial console (https://github.com/home-assistant/operating-system/tree/dev/Documentation/boards/raspberrypi#serial-console)?

Teleportist, 2020-11-06:
I've now obtained a mini HDMI adapter, so can provide more info in the form of photos taken during start up.
I tried to cover the key messages below, but have also uploaded a complete video of the boot (6:40 until a login prompt appears!) just in case I've missed anything.

[https://youtu.be/Sm5rpPWXFfA](url)

Although I do eventually get to a login prompt, I'm never able to gain access via webpage or app. At this time the access light on my USB to SATA adapter goes solid and remains so.

Sorry, I don't have an adapter or the knowledge to access the serial console.

I hope someone can make sense of this. Thanks

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Teleportist, 2020-11-08:
OK, It seems I had a 'dodgy' Eluteng adapter.

According to Andrew G's post here, the incompatible ones can be identified by the Eluteng logo being on a sticker on the underside.
[https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/comment-page-18/#comment-7379](url)

My Kingspec SSD arrived today, which works perfectly from SSD.

Again thanks for all responses to my original post.
Closing.

closed 2020-11-08 by Teleportist

Jpsy, 2021-01-06:
> According to Andrew G's post here, the incompatible ones can be identified by the Eluteng logo being on a sticker on the underside.
> [https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/comment-page-18/#comment-7379](url).

@Teleportist please check your link. It is broken (although the link text is correct).

bcutter, 2021-01-06:
Yeah and where is that information on the sticker thing? Can't find the name you mentioned.

Teleportist, 2021-01-06:
Thanks jpsy - have fixed the link in my original post, but your quote still replicates the original broken one.

@bcutter - you'll need to scroll trough the comments, as that site has no direct way to link to individual posts. Good luck.

bcutter, 2021-01-06:
Well so that´s the comment (currently on page 17):

> I bought 4 Eluteng USB SATA adaptors from Amazon and they all worked perfectly. However I recently purchased one more of the same Eluteng adaptors from Amazon and this one didn’t work. It should have been exactly the same item, but closer inspection shows the USB A connector is moulded differently from the pictures of the Eluteng adaptor, and on the rear of the SATA connector the Eluteng label is applied with a sticker.

The proper ones that work report from `lsusb` that they use the ASMedia/ASM chipset, the so-called Eluteng adaptor that doesn’t work shows it’s got a JMicron chipset. It’s worth double-checking what you receive, either Eluteng have gone cheap(?) with their adaptors or there are cheap knock-offs being sold as Eluteng adaptors.

I checked mine and it reports as JMicron (JMS567) and it has the sticker. I can´t check if works but as the comment states it´s very likely it won´t. **What would you recommend - send it back immediately?**