home-assistant/operating-system #946: RAM busy, system suspended

Gawronnek, 2020-11-01:
[image]

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
docker_version | 19.03.11
hassio | true
host_os | HassOS 4.15
installation_type | Home Assistant OS
os_name | Linux
os_version | 4.19.127-v7
python_version | 3.8.6
supervisor | 2020.10.0
timezone | Europe/Warsaw
version | 0.117.2
virtualenv | false

[image]

pvizeli, 2020-11-01:
Explain a bit more about what the issue is. Currently, it looks like you run too much software on the hardware, maybe influxdb

Gawronnek, 2020-11-01:
That's all I have in the system, I don't know English well, so I don't know what to look for, but the situation repeats every few hours, from 3-4 days
[image]
[image]

agners, 2020-11-02:
Which container exactly is growing? Is this a new issue (e.g. did it work in previous versions of HAOS/HA)?

Gawronnek, 2020-11-02:
Probably containers homeasistant, I have been observing since yesterday, but it was full at night, and in order not to block the system, I did an automation that does a restart at 80%. I've noticed this for a few days, I didn't have time to look at it carefully before, but my colleague has HA 0.116.4 and he has this problem too. The memory fills up and you either have to wait 30-40 minutes or restart.

pvizeli, 2020-11-02:
So it's a memory issue with Core (or a custom component)