home-assistant/operating-system #947: Include alternate firmware boot files on Raspberry Pi builds

enegaard, 2020-11-02:
**HassOS release with the issue:**
4.15

**Supervisor logs:**
Not relevant

**Journal logs:**
Not relevant

**Kernel logs:**
Not relevant

**Description of problem:**
Home Assistant core contains support for the Raspberry Pi camera module, but in order to use the camera, the Raspberry Pi must be booted with alternate firmware and increased GPU memory. Currently this is a moderately difficult, error-prone manual task (documented at https://gist.github.com/enegaard/a57af286205914bd912270c89650fb1b#file-hassos-rpi-camera-howto-md).

Please include the start_x.elf and fixup_x.dat files on future Raspberry Pi 3 and earlier builds, and the start4x.elf and fixup4x.dat files on future Raspberry Pi 4 builds to make the process easier.

lpt2007, 2020-12-12:
Same problem here on rpi2.

If I put start_x=1 in config.txt rpi2 with latest HassOS Release-4 build 20 won't start.

I also try this, without success.
[https://raspberrypi.stackexchange.com/questions/46851/raspberry-pi-2-camera-and-alpinelinux/51440#51440](url)

RubenKelevra, 2020-12-30:
I agree. I was expecting a plug and play functionality with my RasPi 4B, after plugging the camera in. Wondering why there's a need for manual configuration at all. Is it so hard to detect the camera module present?

And why do I have to mess with the operating system?

ajross, 2021-01-24:
Until this is fixed, can the documentation at least be updated on the "Getting Started" page so it's clear that the camera won't work if you're using the Raspberry Pi images?

denics, 2021-01-24:
@frenck this is still active in HAssOS 5.10 is there a specific reason why the modules are not included into the firmware for RPI ? You even propose a "MotionEye" add-on that I would love to use with my RPI camera.

denics, 2021-05-12:
HAssOS 5.13 and the problem is still there dear stale bot

bschatzow, 2021-05-12:
Not fixed, so not stale to me

U32Frank, 2021-05-12:
Please make this a thing

L4RGER, 2021-06-15:
Yep this needs to be a thing, please
We’d be so greatful

agners, 2021-06-17:
With Operating System Release 6 `start4x.elf` and `start4cd.elf` are now included on Raspberry Pi 4 and `startx.elf` as well as `start_cd.elf` on Raspberry Pi 3.

closed 2021-06-17 by agners

enegaard, 2021-06-17:
Yay! Thanks Stefan.