home-assistant/operating-system #950: USB Hub not working after update to 5.4

pepsonEL, 2020-11-02:
Hi. I have Odroid N2 with operating system 5.2 dev. I also has connected to my ODROID N2 two hubs USB like this:
https://pl.aliexpress.com/item/33007409726.html?spm=a2g0o.productlist.0.0.7a2a2379aP08xH&algo_pvid=50d7a3c0-7935-49e8-9b9f-e749e0a7ab07&algo_expid=50d7a3c0-7935-49e8-9b9f-e749e0a7ab07-2&btsid=2100bdcf16043317618737152ea4c6&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

and it works ok to today. Today i update operating system to version 5.4 and aftre this system no boot. Home assistant not boot when HUBs is connected. I discoinect it and it boot ok. I downgrade to version 5.2 and it works again with connected HUBs as i show. What is a problem ? Please help me.... and please implement support to this hub. Tanks.

agners, 2020-11-02:
Is the screen showing anything?

What happens when you plug the Hub out, boot with 5.4, and then plug it back in?

We did update the kernel, so there might be an issue there, but it is hard to tell. Can you try 5.3? You can update to 5.3 using `ha os update --version 5.3`.

pepsonEL, 2020-11-03:
I dont have screen connected to my Odroid N2. Today i try update to 5.3 and give you info,

pepsonEL, 2020-11-03:
On version 5.3 is the same problem. Odroid N2 can not boot when USB Hubs are connected. When disconected Odroid Boot ok and Home Assistant runs ok. Then i plug USB Hubs and reboot only Home Assistant and Hubs working correct. Only under boot Hubs stop boot.

pepsonEL, 2020-11-04:
Is any solution to add support for it ?

closed 2021-05-12 by stale[bot]