home-assistant/operating-system #951: scp broken on 4.15: /usr/bin/dbclient: No such file or directory

deftdawg, 2020-11-02:
**HassOS release with the issue:**

4.15 (5.4 Dev does not boot from SD)

**Description of problem:**

When attempting to [manually pull a snapshot into /mnt/data/supervisor/backup/](https://github.com/home-assistant/frontend/issues/7316#issuecomment-720727246) I discovered that scp is broken

```
# scp test:test test
/usr/bin/dbclient: No such file or directory
```

it has been suggested that `ln -s /sbin/dropbear /usr/bin/dbclient` could resolve the problem from googling, however with / being a read-only squashfs that is impossible.

agners, 2020-11-02:
When you are not using a port then your scp will not connect to the dropbear instance of Home Assistant Operating System but from the Terminal & SSH Add-On. Just tested that with the latest 8.9.1 version of the Add-On and it seems to work here.

deftdawg, 2020-11-03:
@agners not sure we're talking about the same thing?

I was trying to connect from the host OS (shell outside of docker, not from within any containers) to another physical machine running OpenSSHd to pull a backup snapshot over... 

The host OS has scp, however it complains with that message when I try to run it.

agners, 2020-11-03:
@deftdawg oh I see, that way around! I get the same thing. I agree, that would be nice fixed, will look at it!

pvizeli, 2020-11-03:
Well, everything is a container on HassOS. There are Add-ons which sync backups over ssh/rsync/samba/cloud on different location. The SSH access to OS is called "debug access" which some reason.

However, bugs need to be fixed :)

pvizeli, 2020-11-03:
https://www.home-assistant.io/hassio/haos_common_tasks/#copying-your-snapshots-to-another-location

deftdawg, 2020-11-03:
Yup, there are other ways to do it once home assistant is installed for sure.   Even in my case, I was able to use the homeassistant docker container to scp the file and then mv it between volume mounts...  I just had to shell into containers until I found one that worked.

Not sure if this is the buildfile for the host, if so, getting this block to run might help https://github.com/home-assistant/operating-system/blob/dev/buildroot/package/dropbear/dropbear.mk#L20

If it's based on Alpine apk add dropbear-dbclient would do it

agners, 2020-11-10:
This is addressed by the upcoming 5.5 pre-release.

closed 2020-11-10 by agners