home-assistant/operating-system #965: Blank screen with SSD USB drive

urosgruber, 2020-11-06:
I'm trying to install HA latest version with 4.15 OS on RPi4. With SD card everything work as expected but I'm heaving a problem with SSD thumb drive with 256GB in size. 

First of all I run a test with a regular RPi image and I can confirm that device is able to boot and work without any issues. But trying to boot with HA image (any version 32 or 64 bit) it just waits on black screen. For a second I can see boot screen from RPi but soon after it see USB boot device it hangs.

I've tried RPi image writer as well as balenaEtcher and result is the same. I also tried all 4 USB ports without any changes.

Any help how to debug what might be a problem is highly appreciated.

LaStrada, 2020-11-06:
Don't you need to install 5.x (beta) version to be able to boot the rpi4 with an external drive?

urosgruber, 2020-11-06:
@LaStrada Don't know. I was able to boot with regular 32GB USB drive, even on RPi3.

LaStrada, 2020-11-06:
RPI3 and 4 doesn't work the same way. I know there was a lot of talk about the RPI4 not supporting booting from external drives for a long time. Check the first [5.0 beta](https://github.com/home-assistant/operating-system/releases/tag/5.0) release notes. Current latest beta is 5.4.

urosgruber, 2020-11-06:
@LaStrada I tried with 32bit version and it was actually the same. But now I was able to boot in 64 bit version after couple of writes on USB drive. Always different error, from not finding docker image, to connectivity problems. Right now it seems stable enough and I also needed to add `dtparam=sd_poll_once=on` to config.txt to silence some errors about CF card missing.

olivierov, 2020-11-08:
Hi,
With the v4.14 on RPi4 I have the same issue with the following console output (on USB-stick drive and on USB-SSD drive). Works on RPi3

> U-Boot 2020.01 (Oct 22 2020 - 01:54:44 +0000)
> 
> DRAM:  3.9 GiB
> RPI 4 Model B (0xc03112)
> MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
> In:    serial
> Out:   serial
> Err:   serial
> Net:   Net Initialization Skipped
> No ethernet found.
> Card did not respond to voltage select!
> Card did not respond to voltage select!
> No ethernet found.
> missing environment variable: pxeuuid
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/00000000
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/0000000
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/000000
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/00000
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/0000
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/000
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/00
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/0
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/default-arm-bcm283x-rpi
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/default-arm-bcm283x
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/default-arm
> No ethernet found.
> missing environment variable: bootfile
> Retrieving file: pxelinux.cfg/default
> No ethernet found.
> Config file not found
> No ethernet found.
> HassOS>

With the v5.4, it gives better results:

> U-Boot 2020.10 (Oct 23 2020 - 06:15:16 +0000)
> 
> DRAM:  3.9 GiB
> RPI 4 Model B (0xc03112)
> MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
> In:    serial
> Out:   serial
> Err:   serial
> Net:   eth0: genet@7d580000
> PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
> starting USB...
> Bus xhci_pci: Register 5000420 NbrPorts 5
> Starting the controller
> USB XHCI 1.00
> scanning bus xhci_pci for devices... 3 USB Device(s) found
>        scanning usb for storage devices... 1 Storage Device(s) found
> Card did not respond to voltage select!
> Card did not respond to voltage select!
> 
> Device 0: Vendor: TS-RDF5  Rev: TS38 Prod: SD  Transcend   
>             Type: Removable Hard Disk
>             Capacity: 121942.0 MB = 119.0 GB (249737216 x 512)
> ... is now current device
> Scanning usb 0:1...
> Found U-Boot script /boot.scr
> 2382 bytes read in 3 ms (775.4 KiB/s)
> #'# Executing script at 02400000
> 
> Device 0: Vendor: TS-RDF5  Rev: TS38 Prod: SD  Transcend   
>             Type: Removable Hard Disk
>             Capacity: 121942.0 MB = 119.0 GB (249737216 x 512)
> ... is now current device
> loading env...
> 
> usb read: device 0 block # 1214464, count 32 ... 32 blocks read: OK
> #'# Error: bad CRC, import failed
> Found valid slot A, 2 attempts remaining
> storing env...
> 
> usb write: device 0 block # 1214464, count 32 ... 32 blocks written: OK
> Loading kernel
> 15591936 bytes read in 200 ms (74.3 MiB/s)
>  Starting kernel
> #'# Flattened Device Tree blob at 2eff5a00
>    Booting using the fdt blob at 0x2eff5a00
>    Using Device Tree in place at 000000002eff5a00, end 000000002f002ff6
> 
> Starting kernel ...
> 
> [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
> [    0.000000] Linux version 4.19.127-v8 (builder@81d76475aa15) (gcc version 9.3.0 (Buildroot -gc152fdeb)) #1 SMP PREEMPT Fri Oct 23 05:57:25 UTC 2020
> [    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
> [    0.000000] efi: Getting EFI parameters from FDT:
> [    0.000000] efi: UEFI not found.
> [    0.000000] cma: Reserved 64 MiB at 0x0000000037000000
> [    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x470 with crng_init=0
> [    0.000000] percpu: Embedded 24 pages/cpu s59160 r8192 d30952 u98304
> [    0.000000] Detected PIPT I-cache on CPU0
> [    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
> [    0.000000] CPU features: enabling workaround for EL2 vector hardening
> [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
> [    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id= cgroup_enable=memory coherent_pool=1M 8250.nr_uarts=1 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:59:49:0F vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
> ... continuing to boot ...

Hope this helps to understand what goes wrong with hassOS from v4.14 to v4.16, but may be previous revisions are also impacted.

Regards
Olivier

jfgisbert, 2020-11-09:
No v4.X OS version will work on a Pi4G+SSD or USB. As stated in lots of threads, you need a 5.X version to be able to boot on a Rpi4+SSD. As of today, all 5.X versions are pre-release ones, but I am running 5.1beta on my Rpi4 2G + SSD without issues and lightning fast.

closed 2021-05-12 by stale[bot]