home-assistant/operating-system #966: Many addons are missing from store on fresh install

doudz, 2020-11-06:
**HassOS release with the issue:**

Home Assistant 0.117.4
Supervisor Version 227
HassOS 4.15

**Supervisor logs:**

```
20-11-06 14:00:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-11-06 14:00:22 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/core/samba/config.json: does not match regular expression @ data['map'][5]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/core/configurator/config.json: does not match regular expression @ data['map'][2]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/core/ssh/config.json: does not match regular expression @ data['map'][5]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/core/check_config/config.json: does not match regular expression @ data['map'][2]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/a0d7b954/plex/config.json: does not match regular expression @ data['map'][0]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/a0d7b954/ssh/config.json: does not match regular expression @ data['map'][3]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/a0d7b954/motioneye/config.json: does not match regular expression @ data['map'][0]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/a0d7b954/vscode/config.json: does not match regular expression @ data['map'][3]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/a0d7b954/ftp/config.json: does not match regular expression @ data['map'][3]. Got 'media:rw'
20-11-06 14:00:24 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/a0d7b954/node-red/config.json: does not match regular expression @ data['map'][1]. Got 'media:rw'
20-11-06 14:00:24 INFO (MainThread) [supervisor.store] Load add-ons from store: 56 all - 0 new - 0 remove
```
[image]

**Journal logs:**

**Kernel logs:**

**Description of problem:**
On a fresh install on Rpi3 many official addons are missing like samba, configurator

ludeeus, 2020-11-06:
The supervisor you have was not aware of the new "media" directory.
Update the supervisor with  `ha su update --version=2020.11.0` and it will work.

doudz, 2020-11-06:
hum ok, I'll try
it's a bit strange on a fresh install no ?

doudz, 2020-11-06:
(well I installed 4.10 and then upgrade to 4.15)

doudz, 2020-11-06:
> The supervisor you have was not aware of the new "media" directory.
> Update the supervisor with `ha su update --version=2020.11.0` and it will work.

it doesn't work...
```
20-11-06 14:34:01 INFO (MainThread) [supervisor.supervisor] Update Supervisor to version None
20-11-06 14:34:01 INFO (SyncWorker_10) [supervisor.docker.interface] Pull image homeassistant/armv7-hassio-supervisor tag None.
20-11-06 14:34:13 ERROR (SyncWorker_10) [supervisor.docker.interface] Can't install homeassistant/armv7-hassio-supervisor:None -> 404 Client Error: Not Found ("manifest for homeassistant/armv7-hassio-supervisor:None not found: manifest unknown: manifest unknown").
20-11-06 14:34:13 ERROR (MainThread) [supervisor.supervisor] Update of Supervisor fails!
```

pvizeli, 2020-11-06:
I would suggest if that is anyway a fresh install, flash 4.15 or 4.16 directly

doudz, 2020-11-06:
@pvizeli I think you're right, 4.15 install is in progress

doudz, 2020-11-07:
Using version 4.15 solved the problem

closed 2020-11-07 by doudz