home-assistant/operating-system #970: Development 5 build 4 not running on Raspberry PI4 8GB

amrheing, 2020-11-07:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

the goal was to boot the raspyberry and hass from a usb attached ssd disk.

i bought the 8GB pi for future performance. 

i tried a lot, also with a guy who wrote this article and followed the article step by step.

https://homeassistantforum.de/thema/pi4b-nur-mit-einer-usb-ssd-betreiben-ohne-sd-karte/

kind regards, gerald

amrheing, 2020-11-07:
[image]

here the foto... not sure if it was attached before

agners, 2020-11-09:
This is a known issue and due to a bad firmware. I also added this to the known issue section of the release: https://github.com/home-assistant/operating-system/releases/tag/5.4.

There will be a new 5.x pre-release this week.

agners, 2020-11-10:
5.5 pre-release is building right now and should be available soon.

closed 2020-11-10 by agners

amrheing, 2020-11-13:
Hi,
i tested with 5.5 and had the same issue
regards, gerald