home-assistant/operating-system #974: Intel Nuc with Centrino Advanced-N 6235, wifi card not detected

omriasta, 2020-11-09:
**HassOS release with the issue:**

4.14, 4.15, 4.16
**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

➜  ~ dmesg | grep 088e
[    0.557301] pci 0000:02:00.0: [8086:088e] type 00 class 0x028000
**Description of problem:**
It appears that this PCI card is not detected. Card is working when I live boot ubuntu. 
lspci shows this as "8086:088"
lsmod returns:
➜  ~ lsmod
Module                  Size  Used by    Tainted: G  
xfrm_user              36864  1 
snd_hda_codec_hdmi     57344  1 
i915                 1863680  1 
snd_hda_intel          40960  1 
snd_intel_nhlt         16384  1 snd_hda_intel
x86_pkg_temp_thermal    20480  0 
snd_hda_codec         114688  2 snd_hda_codec_hdmi,snd_hda_intel
e1000e                217088  0 
snd_hwdep              16384  1 snd_hda_codec
snd_hda_core           77824  3 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec
snd_pcm                98304  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
cdc_acm                32768  2 

Shouldn't iwlwifi be installed according to the documentation? I do not see the this module loaded at all.
Is there any way to load this module or force it so I can use the wireless card?

➜  ~ nmcli device wifi rescan
Error: No Wi-Fi device found.

omriasta, 2020-11-10:
Thanks, I just found the documentation for the supported Wifi cards and ordered an ac7260 from the list. Will I have to reinstall or will it be detected if I just pop it in?

omriasta, 2020-11-16:
FYI, 7260 card installed, the driver gets loaded but getting errors:
"failed to wake nic for hcmd"
and:
"error sending statistics_cmd"

nmcli can see the card and am able to create a connection however does not work when removing the ethernet cable.

david-schwartz-by, 2021-04-30:
Can this be looked at?

agners, 2022-01-12:
The firmware `iwlwifi-6000g2b-5.ucode` and `iwlwifi-6000g2b-6.ucode` is present on OS 7.1 (and probably already in previous release).

closed 2022-01-12 by github-actions[bot]