home-assistant/operating-system #977: IntelNUC 4.16 unsupported

tremebundo, 2020-11-10:
I have just installed a fresh image on a Intel NUC and I get a bunch of errors and "unsupported platform"
These are de log messages:

`20-11-10 12:42:56 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
20-11-10 12:42:56 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/amd64-hassio-cli with version 2020.10.1 - 172.30.32.5
20-11-10 12:42:56 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-observer with version 2020.10.1
20-11-10 12:42:56 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-11-10 12:42:57 INFO (SyncWorker_1) [supervisor.docker.observer] Starting Observer homeassistant/amd64-hassio-observer with version 2020.10.1 - 172.30.32.6
20-11-10 12:42:57 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/amd64-hassio-multicast with version 3
20-11-10 12:42:57 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
20-11-10 12:42:57 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast homeassistant/amd64-hassio-multicast with version 3 - Host
20-11-10 12:42:57 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-11-10 12:43:02 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
20-11-10 12:43:02 INFO (MainThread) [supervisor.plugins] dns does not have the latest version None, updating
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Updating image homeassistant/amd64-hassio-dns:2020.11.0 to homeassistant/amd64-hassio-dns:None
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Downloading docker image homeassistant/amd64-hassio-dns with tag None.
20-11-10 12:43:02 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install homeassistant/amd64-hassio-dns:None -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:40699->[::1]:53: read: connection refused").
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins.dns] CoreDNS update failed
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins] Can't update dns to None, the Supervisor healthy could be compromised!
20-11-10 12:43:02 INFO (MainThread) [supervisor.plugins] audio does not have the latest version None, updating
20-11-10 12:43:02 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image homeassistant/amd64-hassio-audio:17 to homeassistant/amd64-hassio-audio:None
20-11-10 12:43:02 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image homeassistant/amd64-hassio-audio with tag None.
20-11-10 12:43:02 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/amd64-hassio-audio:None -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:53233->[::1]:53: read: connection refused").
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins.audio] Audio update failed
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins] Can't update audio to None, the Supervisor healthy could be compromised!
20-11-10 12:43:02 INFO (MainThread) [supervisor.plugins] cli does not have the latest version None, updating
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Updating image homeassistant/amd64-hassio-cli:2020.10.1 to homeassistant/amd64-hassio-cli:None
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Downloading docker image homeassistant/amd64-hassio-cli with tag None.
20-11-10 12:43:02 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install homeassistant/amd64-hassio-cli:None -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:37279->[::1]:53: read: connection refused").
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins.cli] CLI update failed
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins] Can't update cli to None, the Supervisor healthy could be compromised!
20-11-10 12:43:02 INFO (MainThread) [supervisor.plugins] observer does not have the latest version None, updating
20-11-10 12:43:02 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image homeassistant/amd64-hassio-observer:2020.10.1 to homeassistant/amd64-hassio-observer:None
20-11-10 12:43:02 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image homeassistant/amd64-hassio-observer with tag None.
20-11-10 12:43:02 ERROR (SyncWorker_0) [supervisor.docker.interface] Can't install homeassistant/amd64-hassio-observer:None -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:50298->[::1]:53: read: connection refused").
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins.observer] HA observer update failed
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins] Can't update observer to None, the Supervisor healthy could be compromised!
20-11-10 12:43:02 INFO (MainThread) [supervisor.plugins] multicast does not have the latest version None, updating
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Updating image homeassistant/amd64-hassio-multicast:3 to homeassistant/amd64-hassio-multicast:None
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Downloading docker image homeassistant/amd64-hassio-multicast with tag None.
20-11-10 12:43:02 ERROR (SyncWorker_1) [supervisor.docker.interface] Can't install homeassistant/amd64-hassio-multicast:None -> 500 Server Error: Internal Server Error ("Get https://registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:37630->[::1]:53: read: connection refused").
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins.multicast] Multicast update failed
20-11-10 12:43:02 ERROR (MainThread) [supervisor.plugins] Can't update multicast to None, the Supervisor healthy could be compromised!
20-11-10 12:43:02 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/intel-nuc-homeassistant with version landingpage
20-11-10 12:43:02 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
20-11-10 12:43:02 INFO (MainThread) [supervisor.homeassistant] Update pulse/client.config: /data/tmp/homeassistant_pulse
20-11-10 12:43:02 INFO (SyncWorker_1) [supervisor.docker.homeassistant] Starting Home Assistant homeassistant/intel-nuc-homeassistant with version landingpage
20-11-10 12:43:02 INFO (MainThread) [supervisor.hassos] Detect HassOS 4.16 / BootSlot A
20-11-10 12:43:02 INFO (MainThread) [supervisor.store.git] Cloning add-on https://github.com/home-assistant/hassio-addons repository
20-11-10 12:43:02 INFO (MainThread) [supervisor.store.git] Cloning add-on https://github.com/hassio-addons/repository repository
20-11-10 12:43:07 INFO (MainThread) [supervisor.store] Loading add-ons from store: 66 all - 66 new - 0 remove
20-11-10 12:43:07 INFO (MainThread) [supervisor.addons] Found 0 installed add-ons
20-11-10 12:43:07 INFO (MainThread) [supervisor.snapshots] Found 0 snapshot files
20-11-10 12:43:07 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
20-11-10 12:43:07 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
20-11-10 12:43:07 ERROR (MainThread) [supervisor.homeassistant.api] Can't update Home Assistant access token!
20-11-10 12:43:07 ERROR (MainThread) [supervisor.core] NetworkManager is not correctly configured
20-11-10 12:43:07 INFO (MainThread) [__main__] Running Supervisor
20-11-10 12:43:07 WARNING (MainThread) [supervisor.core] System running in a unsupported environment!
20-11-10 12:43:07 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good
20-11-10 12:43:07 INFO (MainThread) [supervisor.supervisor] Update Supervisor to version None
20-11-10 12:43:07 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image homeassistant/amd64-hassio-supervisor with tag None.
[... cut ...]
20-11-10 12:44:54 INFO (SyncWorker_3) [supervisor.docker.interface] Cleanup images: ['homeassistant/intel-nuc-homeassistant:landingpage']
20-11-10 12:55:21 INFO (MainThread) [supervisor.host.info] Updating local host information
20-11-10 12:55:21 INFO (MainThread) [supervisor.host.services] Updating service information
20-11-10 12:55:21 INFO (MainThread) [supervisor.host.network] Updating local network information
20-11-10 12:55:22 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information`

agners, 2020-11-10:
I remember having seen that on a RPi3 when plugging in Ethernet a bit late. Does that problem persist after reboot?

chrisdrackett, 2020-11-15:
running into this as well, but didn't update to 4.16 yet, only core to `0.117.6`

PavelPawlowski, 2020-11-18:
Have the same problem.. even update to 4.17 did not help.

agners, 2020-11-26:
@chrisdrackett so when this suddenly appeared it is likely Supervisor related which automatically updates. What Supervisor version are you running? Is the problem still persisting?

agners, 2020-11-26:
This is very likely a duplicate of https://github.com/home-assistant/supervisor/issues/2256. In case the problem still persists with the new Supervisor release (currently on the beta channel), please reopen.

closed 2020-11-26 by agners