home-assistant/operating-system #982: Cant Update to 5.5 via CLI

townsmcp, 2020-11-11:
I have tried to update to 5.5 from 5.4 on RPI4 via CLI by running:
```
ha os update --version 5.5
```
This has worked for previous releases (updating to 5.4 etc) however when I run the update today, I get an error message:
[image]

When I look at the logs I see the below message.
As can be seen in the logs, the updater is expecting the file name to be 'hassos_rpi4-64-5.5.raucb' however on the assets for 5.5 on https://github.com/home-assistant/operating-system/releases there is no file titled 'hassos_rpi4-64-5.5.raucb', there is however a file titled 'hassos_rpi4-5.5.raucb'.
Does this mean there is no more 64 bit version for RPI4? Or is the file name incorrect on the release? Previous version of 32bit 5.x on RPI4 did not work with SSD.

**HassOS release with the issue:**
Updating to 5.5 on RPI 4

**Supervisor logs:**
```
20-11-11 13:17:58 INFO (MainThread) [supervisor.api.security] /os/update access from a0d7b954_ssh
20-11-11 13:17:58 INFO (MainThread) [supervisor.hassos] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/5.5/hassos_rpi4-64-5.5.raucb
```

**Description of problem:**
Cant update to 5.5 on RPI 4.

Jpsy, 2020-11-11:
I can confirm this problem.

bpwats, 2020-11-11:
I was able to succesfully update to version 5.5 using the command line. Various versions will gradually appear.

townsmcp, 2020-11-11:
@bpwats I assume you weren't using 64 bit? 
Are you on an RPI?
Are you booting and using SSD rather than SD card?

bpwats, 2020-11-11:
Yes, RPi 4 4GB / 64-bit HassOS image (so not the latest 8GB model), Samsung SSD using StarTech USB3 to Sata connection. RPi firmware upgraded to latest firmware to allow booting from SSD. No SD-card in RPi.

townsmcp, 2020-11-11:
Interesting. I first tried with 5.0 which wouldn’t take 32bit. Nor 5.1. At which point I went with 64bit and stayed with that since.

What command did you enter to update your RPI? 
I don’t understand how you could have updated 64bit 5.4 to what seems to be 32bit 5.5 (being as the 64bit name is missing from the list)

townsmcp, 2020-11-11:
How strange! The 64bit version is now on the assets list. And the update command has just completed successfully

closed 2020-11-11 by townsmcp