home-assistant/operating-system #983: HassOS 5.5 won't boot with wifi disabled

non092, 2020-11-11:
I have updated my Rpi4b 8GB running HassOS 5.3 64bit today to 5.5 and it couldn't boot after update
As it is running headless and I currently have no HDMI cable to check what was going on I just did a clean install and restored my snapshot.
After getting everything back up and running I edited config.txt with the settings I customized before.

This is how I discovered it wouldn't boot after adding dtoverlay=disable-wifi, removing the setting allow HassOS to boot fine

agners, 2020-11-11:
Just tried that here, I can confirm this on RPi4 when `enable_uart` is commented out. Hm, I think that is related to U-Boot picking up that UART which isn't initialized. You can either enable UART from firmware side with `enable_uart=1` which will make U-Boot not crash, or disable it entirely using `dtparam=uart0=off`.

What is your use case of disabling WiFi?

non092, 2020-11-12:
Not sure my use case is valid anymore, as I don’t use wifi I just thought it is best to leave it disabled.
Also on my previous Pi3 I had under voltage issues which I thought could be improved by disabling unnecessary hardware (WiFi,Audio)
I can live with it enabled, just thought that it would be a good idea to report the bug so someone else do not waste 2 hours on this 😊

closed 2021-05-12 by stale[bot]