home-assistant/operating-system #987: HassOS 5.5 - RPi 4/4GB - no boot from USB

thespawnat, 2020-11-13:
**HassOS release with the issue:**
5.5 Beta

**Supervisor logs:**
not available / not booting

**Journal logs:**
not available / not booting

**Kernel logs:**
not available / not booting

**Description of problem:**
tried today to switch from manual installation with "default rpi 4 / 32 bit image debian" to hass-os, but it won't boot from my USB stick, only from micro sd card.

storage is a 128 GB SanDisk USB 3.0 Stick, which worked really fine with default raspbian image.

I tried release 5.0 and 5.5 with same result - blank screen

I've tried the 32-bit (need to use GPIO) and 64-bit for testing purpose, wrote the image from windows 10 with balena etcher and win32diskimager and on linux mint with dd, but no difference. I can see that the bootloader 1-2 seconds, that it detects the usb stick and afterwards is a blank screen

agners, 2020-11-13:
Hm, it seems that U-Boot has trouble enumerating your USB device. Do you happen to have a USB UART TTL adapter? If so, it would be interesting to see the serial output using the serial console (see https://github.com/home-assistant/operating-system/tree/dev/Documentation/boards/raspberrypi#serial-console).

thespawnat, 2020-11-13:
yes I have one, will try tomorrow to create an output of the boot sequence

regards

sreckoo, 2020-11-13:
I'm having same issue booting from SSD drive (NVMe). All I get is black screen.

Thanks and regards

amrheing, 2020-11-14:
[image]

i have the same issue with the 8GB RPI4

amrheing, 2020-11-14:
i filmed the boot process and was able to get this screen captured. after that the screen get the colored....
[image]

amrheing, 2020-11-14:
[image]

thespawnat, 2020-11-14:
enabled as documented the serial output and connected with putty / 115200 

looks like as it "stops" after dram output. will try later a 8GB 3.0 USB Stick from Kingston.

[image]

thespawnat, 2020-11-14:
did some additional testing today. the 64-bit image now boots from Kingston Data Traveler 100 G3 8 GB and SanDisk UltraFit 128 GB USB 3.1. With the 32-bit image, both stop after "DRAM: 3.9 GiB" and the raspi gets really hot. 

I re-flashed the 64-bit image with balena etcher 2 times in a row, to get it booting. but for the 32-bit image, it won't work.. .

thespawnat, 2020-11-14:
everything seems to work fine, except gpio. can't get it working. is it still unsupported in 5.5 as in the v4 on 64bit?

sreckoo, 2020-11-17:
I also managed to boot 64-bit version 5.5 from SSD (Samsung 850 EVO - MZ-N5E500 and Ewent EW7023 enclosure). It doesn't want to work with 32-bit version.

agners, 2020-11-17:
Hm, I tested the 32-bit version on 8GB just a few days ago, but I admit I did not test it on the 4GB version. Will test that.

Since the Raspberry Pi OS starts supporting 64-bit as well we plan to support it as well or even recommend the 64-bit variant.

amrheing, 2020-11-17:
Hi,

i also recreated my ssd now with the raspberry imager.

it doesn't work!

SSD SanDisk SDSSDA 128GB
ELUTENG usb3-SATA Adapter
hassos_rpi4-64-5.5.img

i there anything i can analyze to help you?

kind regrds, gerald

[image]

ShonP40, 2020-11-17:
Getting the same issue with the 32-bit HassOS
64-bit works though

amrheing, 2020-11-17:
that is strange, as mentioned before 32bit should not work and the raspy should become hot???

How did you get the 64bit working?

ShonP40, 2020-11-17:
Just downloaded the latest 64-bit RPI 4 HASSOS 5.5 img and flashed it onto my WD SSD using the RPI Imager.

agners, 2020-11-17:
Some USB drive to cause problems with Home Assistant OS or/and RPi 4 in general.

@amrheing is your drive working with Raspberry Pi OS? Also make sure to update to the latest EEPROM using SD card and Rasperry Pi OS.

If Raspberry Pi OS works and Home Assistant OS not, then the serial port output would be interesting if you have access to a USB to TTL UART adapter (as described in https://github.com/home-assistant/operating-system/issues/987#issuecomment-726789957).

amrheing, 2020-11-18:
Hi, 
i tried again and it doesen't work. Now i ordered a ttl-usb cable and will do deeper analyzes for you.
regards, gerald

caylor93, 2020-11-18:
I'm also having issues with the 5.5 on RPI 4/4GB.  I performed a “ha os update --version 5.5” to upgrade from 5.4 and when it rebooted I now get that it is looking for the SD CARD and will not boot. The initial boot mode (from power on) shows USB MSD and says "USB MSB timed out after 20 secs" and then defaults to boot mode SD 0 and ends with a "Insert SD Card".  This set up was working fine with 5.4 and the EEPROM is fully updated.

agners, 2020-11-20:
@thespawnat I can confirm here, 32-bit image running on 2GB and 4GB version of RPi from booting from USB flash drive fails to boot. This is what I get on the serial console:

```
U-Boot 2020.10 (Nov 11 2020 - 12:21:04 +0000)

DRAM:  1.9 GiB
<freeze>
```

I think I have a work around, but I am a bit worried that it negatively affects other combinations. I am also not sure who is at fault here, it could be the firmware or U-Boot. I brought this topic up on the U-Boot mailing list, see https://lists.denx.de/pipermail/u-boot/2020-November/433207.html.

agners, 2020-11-20:
@caylor93 I am a bit confused, 5.4 did not boot at all due to a firmware issue... Are you sure you have been using 5.4? What is your boot device?

caylor93, 2020-11-20:
> @caylor93 I am a bit confused, 5.4 did not boot at all due to a firmware issue... Are you sure you have been using 5.4? What is your boot device?

Maybe it was a fluke that it booted, but I was definitely using 5.4 with a Samsung 970 EVO 500GB NVMe M.2 SD and a Pluggable USB adapter on an external USB 3.0 powered hub.  Funny thing was, it was the first time I had gotten that combo to boot at all.  I was able to get it to boot once on a USB 2.0 port and was able to download a backup, but then it goes into an infinite boot loop.  When I loaded a fresh copy of 5.5 to the drive to restore the backup, it never boots in any configuration.

nickidw, 2021-01-01:
I have a similar problem on RPI4 4GB. In order to troubleshoot a potential issue with an SDCard, I tried writing a HassOS image (5.8 and 5.9, 32bit) to a USB drive, and booting from it. If I power the pi on without the USB or an sdcard, it displays the bootloader. If I then put in the usb drive, it briefly changes to a black screen, and thereafter to a screen that looks like it is in graphics mode, but nothing showing on the screen. It is not black, but there appears to be some backlight on the screen. I was able to boot Raspbian OS successfully with the same usb drive.

If I power on the raspberry with the USB drive plugged in, the end result is the same, it just does not display the rpi bootloader screen because it seems to boot somehow.

I updated the bootloader to a Sept version, and also to a recent December version. No difference.

agners, 2021-01-04:
@nickidw does it work with 64-bit? This issue was about 32-bit failing while 64-bit working. 32-bit now should be working as well as 64-bit. But there are other issues with some drives, so you might hit that problem.

closed 2021-01-04 by agners

nickidw, 2021-01-04:
@agners I only tried 32 bit, since I have a 4GB Pi. I'll search and troubleshoot further and log a new issue, since it seems there is still something not working for 32bit on a recent version.

stone0904, 2021-01-04:
@agners: hi Stefan, I had similar issues with rpi 4/4Gb with 32 bit and 64 bit. Wait until the boot loader came back with the error missing Sd card for boot, plug my ssd in my rpi then the image starts. 
Additional information: On my Mac I can't mount the /boot partition.
With the same image on a sd card 32 GB no issues to start my rpi. 
BTW. The same hardware runs with official raspberry os perfectly.