home-assistant/operating-system #992: Could SkiffOS be useful as a layer on top of Buildroot?

paralin, 2020-11-15:
Hi all,

Home-assistant is great. I was wondering if my project Skiff might be of use for the OS part.

https://github.com/skiffos/skiffos

It is a layer of scripts + modular configuration packages for configuring Buildroot to produce an identical output across many underlying supported device configs.

If you added it as a sub-module, and then added a custom configuration layer "hass/os" or so, you could include either automated docker-based setup or bake the OS directly into the buildroot tree.

This is of course very similar to what the home-assistant/operating-system project already does.

If this isn't of interest to the community feel free to close. Thanks!

agners, 2020-11-15:
@paralin interesting project, and indeed very similar architecture and goals. I guess it comes pretty quickly down to if we still have the amount of control we need for the OS and if it really does lower the maintenance effort.

Do I understand it right that you are using a initramfs with the complete go runtime and Docker container engine in it? Isn't that wasting quite some memory since lots of code which might not be needed during runtime needlessly sits in memory?

What is the update mechanism you are using? I see OTA, but not exactly what it is or how it works.

paralin, 2020-11-16:
@agners The iniitramfs model is ultimately optional and can be overridden depending on the package.

However, the memory usage with it is negligible, and I've found that this approach provides a level of reliability that booting with a mutable root filesystem does not.

For example: the kernel modules are loaded in a separate squashfs, which slims down the initramfs component to just 100Mb, and slimming it even further is possible if really necessary (although I haven't seen a case yet where it was).

This works fine even on something as small as a Pi 0.

For updates, you just replace 3 files: rootfs.cpio.gz, Image, and modules.squashfs. There is a script here which does it across all the skiff boards: https://github.com/skiffos/SkiffOS/blob/master/scripts/push_image.bash

You still have 100% control over the OS. Especially with the ability to create unlimited custom configuration layers, which can contain buildroot + kernel configs, patches, external buildroot packages, target filesystem root overlays.

The advantage here is having a project where continuous work + testing is being done on the baseline configurations for each target board. Then, each application can layer its specific requirements on top.

More CI is coming soon for automated E2E boot + run testing on target boards.

paralin, 2020-11-16:
FWIW I'm running home-assistant on Skiff already like this:

```
docker run --init \
  -d --name="home-assistant" \
  -e "TZ=America/Los_Angeles"  \
  -v /mnt/persist/hass/config:/config \
  -v /mnt/persist/hass/media:/media \
  --net=host \
  homeassistant/raspberrypi4-homeassistant:stable
```

paralin, 2021-02-25:
@agners Hey, Skiff is still chugging along, building some momentum. I will work to add an `apps/hass` package soon for integrating home-assistant, but you could also add it as a submodule of this project and keep adding custom configs on top here.

You can see an example of how the submoduling works, with buildroot_ext support: https://github.com/skiffos/skiff-ext-example

paralin, 2021-10-02:
This is still a interesting possibility - instead of using Buildroot directly, could use [this format](https://github.com/skiffos/skiff-ext-example) and support all of the different boards Skiff supports.

sairon, 2026-09-17:
Thanks for the suggestion and for the follow-ups over the years. SkiffOS is an interesting project, but HAOS has grown quite a bit since 2020 and is now tightly built around its own Buildroot tree, RAUC-based updates, and the board support we maintain in-tree. Switching to a shared layer like SkiffOS isn't something we plan to pursue, so I'm closing this to keep the tracker focused on things we intend to act on.

closed 2026-09-17 by sairon