home-assistant/operating-system #993: Add drivers used for NUC to OVA. For users of hypervisor and PCI pass-trough

kolins-cz, 2020-11-15:
**HassOS release with the issue:**
 .ova release version 0.117.6

**Description of problem:**
I am running HassOS in ESXi on Intel NUC 5gen. I would like to use NUC integrated Wi-Fi/Bluetooth and sound card. PCI Passtrough is working fine in vmware, but image built for virtual machines does not have the drivers included.

According to **intel_nuc_defconfig** and **nuc kernel.config** there are missing features and drivers forvirtual environment in NUC build. So installing NUC image in virtual would probably lead to some issues or degraded performance.

I could probably install buildroot and build custom version of HassOS but it would break automatic updates.

Perhaps new custom board added?

agners, 2020-11-15:
Just to make sure I understand that right, you like to forward the NUC integrated Wi-Fi/Bluetooth (looks like a M.2 Intel Wireless-AC 7265 + Bluetooth 4.2 PCIe/USB device) to the virtual instance running HassOS?

Hm, I see you basically need the virtualization drivers for pass-through from `ova` but the device drivers from `intel_nuc`. Yeah I am not sure, I am not a huge fan of too many board configuration. It think I would prefer to extend the `ova` with drivers for devices commonly forwarded, like WiFi/Bluetooth.

kolins-cz, 2020-11-15:
Exactly. I would like to forward Intel Wireless and Sound card to HassOS. I believe that for PCI-passtrough itself there are no special drivers needed. When I do **lspci** on HassOS, I can see correct VID/PID. It would be probably enough to add drivers to .OVA image. I believe it will not slow down startup or increase image size significantly for other users not using this function.

CosminC82, 2021-01-09:
Hello,

Same here, I am running Hass OS in kvm on an Intel NUC that has the Intel Corporation Centrino Advanced-N 6235 wifi/BT card. I have it passthrough to Hass OS, and it's being seen in lspci, but can't use it because of no drivers available.

closed 2021-05-12 by stale[bot]