home-assistant/operating-system #994: Periodic fstrim execution

alexborro, 2020-11-16:
**HassOS release with the issue:**

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
The fstrim is needed to release no longer used blocks for LVM systems. Currently HASSIO has the fstrim application but there is no way to easily execute it periodically since it needs root permission and there is no way to execute it from HA. A tool like Cron will solve this problem.

Just as reference, my HASSIO VM was using 18Gb of space. After executing fstrim, it dropped down to 11Gb. Almost 40% of the space was wasted, allocated in the filesystem but no longer used. 

I currently have automated backups, but need to execute fstrim manually - it is not good for the system maintenance.

agners, 2020-11-16:
Hm, `fstrim` actually also make sense on flash storage like eMMC or USB SSD drives. We should consider adding `fstrim` in general.

Fedora uses systemd timer which seems a good fit for this: https://fedoraproject.org/wiki/Changes/EnableFSTrimTimer

alexborro, 2020-11-16:
Yes, that is true. It will benefit specially those who have smaller SD cards / storage for HA.
Trimming becomes critical when the storage is almost full, so the memory controller needs to know which blocks are free.

It would be nice to have a scheduled service to run fstrim in an user specified period.

HASSIO already has fstrim, we just need the scheduler.

jrbenito, 2020-11-19:
A way to schedule it would very helpful. I do run it manually, bu I systematically forget to do it. For instance, today I run a backup of my system, the VM image, compacted was 7.1GB. I then run fstrim manually and did another backup and it was only 2.8GB compacted.  This is much much less data to transfer to a backup storage besides the other benefits already discussed.

agners, 2021-02-08:
It seems that systemd is doing this now automatically on release 5:

```
...
Feb 08 01:40:02 ha-intel-nuc fstrim[67443]: /mnt/data: 105.6 GiB (113412075520 bytes) trimmed on /dev/sda8
...
```

The timer is setup to do it weekly (see `/usr/lib/systemd/system/fstrim.timer`). It can also be manually triggered by using `systemctl start fstrim.service` in the OS console.

To me that seems good enough for now. If there are good reasons we need different behavior/functionality, please create a new issue.

closed 2021-02-08 by agners

jrbenito, 2021-02-19:
That is good enough for me!

Thanks a lot for this.