home-assistant/operating-system #997: Errors with Intel 7260 wifi card

omriasta, 2020-11-16:
**HassOS release with the issue:**

4.16
**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

```

➜  ~ dmesg | grep iwl 
[    2.365740] iwlwifi 0000:02:00.0: enabling device (0100 -> 0102)
[    2.384036] iwlwifi 0000:02:00.0: loaded firmware version 17.3216344376.0 op_mode iwlmvm
[    2.432720] iwlwifi 0000:02:00.0: Detected Intel(R) Dual Band Wireless AC 7260, REV=0x144
[    2.462826] iwlwifi 0000:02:00.0: base HW address: 00:16:6f:e7:5e:73
[    2.678474] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[    2.789091] iwlwifi 0000:02:00.0 wlp2s0: renamed from wlan0
[   76.027180] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   76.027324] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   76.047631] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   76.047749] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   82.026161] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   82.026334] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   82.048610] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   82.048883] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   84.953127] WARNING: CPU: 1 PID: 0 at drivers/net/wireless/intel/iwlwifi/pcie/trans.c:2089 iwl_trans_pcie_grab_nic_access+0x1e4/0x220 [iwlwifi]
[   84.953135] Modules linked in: xfrm_user snd_hda_codec_hdmi x86_pkg_temp_thermal iwlmvm i915 snd_hda_intel snd_intel_nhlt snd_hda_codec cdc_acm iwlwifi snd_hwdep snd_hda_core e1000e(O) snd_pcm
[   84.953181] RIP: 0010:iwl_trans_pcie_grab_nic_access+0x1e4/0x220 [iwlwifi]
[   84.953274]  ? iwl_pcie_txq_inc_wr_ptr+0xf0/0xf0 [iwlwifi]
[   84.953284]  iwl_read_prph+0x2f/0x80 [iwlwifi]
[   84.953297]  iwl_trans_pcie_log_scd_error+0x135/0x210 [iwlwifi]
[   84.953309]  ? iwl_pcie_txq_inc_wr_ptr+0xf0/0xf0 [iwlwifi]
[   84.953320]  iwl_pcie_txq_stuck_timer+0x31/0x50 [iwlwifi]
[   84.953543] iwlwifi 0000:02:00.0: iwlwifi transaction failed, dumping registers
[   84.953552] iwlwifi 0000:02:00.0: iwlwifi device config registers:
[   84.953841] iwlwifi 0000:02:00.0: 00000000: 08b18086 00100000 028000bb 00000000 00000004 00000000 00000000 00000000
[   84.953853] iwlwifi 0000:02:00.0: 00000020: 00000000 00000000 00000000 40708086 00000000 000000c8 00000000 00000100
[   84.953864] iwlwifi 0000:02:00.0: 00000040: 00020010 10008ec0 001b0c10 0006ec11 10110000 00000000 00000000 00000000
[   84.953874] iwlwifi 0000:02:00.0: 00000060: 00000000 00080812 00000005 00000000 00010001 00000000 00000000 00000000
[   84.953885] iwlwifi 0000:02:00.0: 00000080: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.953895] iwlwifi 0000:02:00.0: 000000a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.953905] iwlwifi 0000:02:00.0: 000000c0: 00000000 00000000 c823d001 0d000000 00804005 00000000 00000000 00000000
[   84.953915] iwlwifi 0000:02:00.0: 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.953926] iwlwifi 0000:02:00.0: 00000100: 14010001 00100000 00000000 00462031 00002001 00002000 00000014 40000001
[   84.953936] iwlwifi 0000:02:00.0: 00000120: 0000000f f7c00460 00000000 00000000 00000000 00000000 00000000 00000000
[   84.953947] iwlwifi 0000:02:00.0: 00000140: 14c10003 ffe75e73 00166fff 15410018 00000000 0001000b 0141cafe 00f01e1f
[   84.953956] iwlwifi 0000:02:00.0: iwlwifi device memory mapped registers:
[   84.953999] iwlwifi 0000:02:00.0: 00000000: ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff
[   84.954009] iwlwifi 0000:02:00.0: 00000020: ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff
[   84.954021] iwlwifi 0000:02:00.0: iwlwifi device AER capability structure:
[   84.954056] iwlwifi 0000:02:00.0: 00000000: 14010001 00100000 00000000 00462031 00002001 00002000 00000014 40000001
[   84.954065] iwlwifi 0000:02:00.0: 00000020: 0000000f f7c00460 00000000
[   84.954073] iwlwifi 0000:02:00.0: iwlwifi parent port (0000:00:1c.2) config registers:
[   84.954264] iwlwifi 0000:00:1c.2: 00000000: 1e148086 00100407 060400c4 00810010 00000000 00000000 00020200 200000f0
[   84.954274] iwlwifi 0000:00:1c.2: 00000020: f7c0f7c0 0001fff1 00000000 00000000 00000000 00000040 00000000 00120305
[   84.954285] iwlwifi 0000:00:1c.2: 00000040: 01428010 00008000 00110000 03123c12 70110042 0014b200 01480000 00000000
[   84.954295] iwlwifi 0000:00:1c.2: 00000060: 00000000 00000016 00000000 00000000 00010002 00000000 00000000 00000000
[   84.954306] iwlwifi 0000:00:1c.2: 00000080: 00019005 fee02004 00004022 00000000 0000a00d 20448086 00000000 00000000
[   84.954316] iwlwifi 0000:00:1c.2: 000000a0: c8020001 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954326] iwlwifi 0000:00:1c.2: 000000c0: 00000000 00000000 00000000 00000000 01000000 00000b02 81118000 00000000
[   84.954337] iwlwifi 0000:00:1c.2: 000000e0: 00000300 00000000 00000001 00000000 00000000 00000000 08050f87 00000000
[   84.954347] iwlwifi 0000:00:1c.2: 00000100: 00000000 00000000 00000000 00060011 000011c1 00002000 00000000 00000000
[   84.954357] iwlwifi 0000:00:1c.2: 00000120: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954367] iwlwifi 0000:00:1c.2: 00000140: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954377] iwlwifi 0000:00:1c.2: 00000160: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954388] iwlwifi 0000:00:1c.2: 00000180: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954398] iwlwifi 0000:00:1c.2: 000001a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954408] iwlwifi 0000:00:1c.2: 000001c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954418] iwlwifi 0000:00:1c.2: 000001e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   84.954427] iwlwifi 0000:00:1c.2: 00000200: 00000000 00000000 00000000
[   85.010396] iwlwifi 0000:02:00.0: Queue 11 is active on fifo 2 and stuck for 10000 ms. SW [54, 130] HW [90, 90] FH TRB=0x05a5a5a5a
[   88.020509] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   88.020635] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   88.048370] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   88.048488] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   94.021272] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   94.021394] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[   94.049480] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[   94.049556] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  100.020616] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  100.020719] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  100.047532] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  100.047593] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  106.020536] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  106.020636] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  106.046213] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  106.046290] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  112.024323] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  112.024372] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  112.049313] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  112.049357] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  118.026298] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  118.026349] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  118.055342] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  118.055402] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  124.025288] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  124.025382] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  124.044289] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  124.044353] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  130.024525] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  130.024785] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
[  130.045341] iwlwifi 0000:02:00.0: Failed to wake NIC for hcmd
[  130.045417] iwlwifi 0000:02:00.0: Error sending STATISTICS_CMD: enqueue_hcmd failed: -5
```

**Description of problem:**
Wifi Card is detected and ip address is set, nmcli shows the connection as active however the card still fails. I tried setting powersave=0 and powersave=2 as per the instructions but that did not help.

agners, 2020-11-16:
What system are you running on?

agners, 2020-11-16:
It seems as if PCIe somehow fails. Maybe a hardware failure?

omriasta, 2020-11-19:
Looks like you were right, I tried this using live ubuntu and got the same errors, I should have a replacement card soon to test. I don't think it was the slot because I had an intel 6235 card before which is not supported in HassOS but it worked fine with no errors in ubuntu live. I will check in again when the new card arrives.

agners, 2020-11-20:
Ok, thanks for the update. Please reopen the issue in case it turns out not to be hardware issue.

closed 2020-11-20 by agners

omriasta, 2020-11-28:
Received a new card and all is well, thanks