home-assistant/operating-system #1000: HassOS hardware support?

newlund, 2020-11-17:
Hi!

I'm currently running HassOS in a proxmox VM but want to move to separate hardware in order to get more battery time from my UPS.
I tried to install HassOS on a Zotac Zbox ND22 I had lying around but I can't get it to boot with the NUC image. I have tried both 4.16 and 5.5 but I only get a blinking cursor.
The ND22 has a Nvidia Ion chipset which I guess is the problem. Or perhaps that it lacks UEFI?
Could you perhaps provide a more generic image with all the hardware support that comes with the linux kernel? The ND22 runs debian without any additional driver modules.
Or is this not possible due to the use of buildroot?

Thank you for your great work with HassOS!

BR/Nicklas

newlund, 2020-11-17:
After reading up a bit I guess it could be due to missing UEFI. The NUC build requires UEFI?
Would it be possible for you to provide an image similar to the NUC image but with U-Boot instead?

agners, 2020-11-17:
Yes we use UEFI for Intel NUC. It probably could be done using U-Boot or Barebox (what we use today for x86), but it would need to go back to BIOS/MBR based boot flow. I'd rather prefer moving forward and only support UEFI/GPT based boots.

newlund, 2020-11-17:
I understand you want to move forward but perhaps there are many like me that have old hardware lacking UEFI but otherwise is perfect for running HassOS? Since HA doesn't require that much CPU, old hardware would be perfect to reuse. That is, if it is not too power hungry.

agners, 2021-02-04:
Barebox does support legacy BIOS boot, so we could reuse that boot loader. But it would mean quite some implementation effort and maintenance. 

From my perspective I really don't see a point putting effort into it. UEFI started to get popular ~10 years ago, so even old hardware these days supports UEFI. I'd rather prefer to add more features to UEFI boot, like secure boot etc.

You can use Debian in a Supervised installation if you really want to use this hardware.

closed 2021-02-04 by agners

newlund, 2021-02-05:
Ok, I understand. Will get something more up to date.
Thank you for your excellent work! 👍