home-assistant/operating-system #1010: HassOS 5.x read/write issues

LeChuck2909, 2020-11-20:
**HassOS release with the issue:**

5.x

**Supervisor logs:**

20-10-31 09:05:12 ERROR (SyncWorker_3) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:05:12 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:05:12 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:06:04 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
20-10-31 09:06:04 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:06:04 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:06:04 ERROR (SyncWorker_5) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:06:04 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:06:04 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:07:04 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
20-10-31 09:07:04 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:07:04 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:07:04 ERROR (SyncWorker_0) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:07:04 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:07:04 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:08:04 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
20-10-31 09:08:04 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:08:04 INFO (SyncWorker_3) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:08:04 ERROR (SyncWorker_3) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:08:04 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:08:04 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:08:15 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Application found a problem with observer plugin!
20-10-31 09:08:15 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:08:15 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:08:15 INFO (SyncWorker_2) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:08:15 ERROR (SyncWorker_2) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:08:15 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:08:15 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Read-only file system
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Read-only file system
20-10-31 09:09:04 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
20-10-31 09:09:04 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:09:04 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:09:04 ERROR (SyncWorker_0) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:09:04 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:09:04 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:10:04 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
20-10-31 09:10:04 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:10:05 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:10:05 ERROR (SyncWorker_5) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:10:05 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:10:05 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:10:24 INFO (MainThread) [supervisor.host.info] Updating local host information
20-10-31 09:10:25 INFO (MainThread) [supervisor.host.services] Updating service information
20-10-31 09:10:25 INFO (MainThread) [supervisor.host.network] Updating local network information
20-10-31 09:10:26 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-10-31 09:11:05 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
20-10-31 09:11:05 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:11:05 INFO (SyncWorker_5) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:11:05 ERROR (SyncWorker_5) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:11:05 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:11:05 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:11:18 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Application found a problem with observer plugin!
20-10-31 09:11:18 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:11:18 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
20-10-31 09:11:18 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_observer application
20-10-31 09:11:18 ERROR (SyncWorker_0) [supervisor.docker] Can't create container from hassio_observer: 409 Client Error: Conflict ("Conflict. The container name "/hassio_observer" is already in use by container "fb0f9e9a33e88b62fbae8d8ec95318731de6e697e839a01d053fb77952a3b37e". You have to remove (or rename) that container to be able to reuse that name.")
20-10-31 09:11:18 ERROR (MainThread) [supervisor.plugins.observer] Can't start observer plugin
20-10-31 09:11:18 ERROR (MainThread) [supervisor.misc.tasks] Observer watchdog reanimation failed!
20-10-31 09:12:05 WARNING (MainThread) [supervisor.misc.tasks] Watchdog/Docker found a problem with observer plugin!
[... cut ...]
the log from the supervisor is not always like above, but it seems to be always mesages with read/write.

i did not starte an issue on git yet, becaue the problem for me is so difficult to explain and i dont want to open a complicated unprofessional issue. at one point i startet an thread in the forum though, because i was desperate.

on release of HA core 118 i read that HA OS 5 is considered to be close to stable, and this motivated me to open this issue, because my experience is oposite.

again sorry for this mess of a issue and thank you all, as i appreciate very much all of your work.

agners, 2020-11-23:
I have seen read errors on my SSD as well, for one I had to disable UAS mode using `usb-storage.quirks=aaaa:bbbb:u `. This is a general RPi4 issue, see https://www.raspberrypi.org/forums/viewtopic.php?t=245931.

One of my SSD even with disable UAS showed read errors. The only way I was able to get it reliable was using a powered USB Hub. Can you try this approach as well?

LeChuck2909, 2020-11-25:
so you mean, have a separate power supply for the ssd, right ? either by a powered USB Hub, or by a powered USB-to-SATA adapter, right ?

LeChuck2909, 2020-11-25:
by the way, home assistant is starting to become very slow again and doesnt show any history graphs anymore and this is the according supervisor log

`20-11-25 04:40:32 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-11-25 04:40:32 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository
20-11-25 04:40:32 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-11-25 04:40:34 INFO (MainThread) [supervisor.store] Loading add-ons from store: 67 all - 0 new - 0 remove
20-11-25 05:27:12 INFO (MainThread) [supervisor.host.info] Updating local host information
20-11-25 05:27:12 INFO (MainThread) [supervisor.host.services] Updating service information
20-11-25 05:27:12 INFO (MainThread) [supervisor.host.network] Updating local network information
20-11-25 05:27:13 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-11-25 05:40:19 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-11-25 06:44:34 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 257:None is not str
20-11-25 06:44:34 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
20-11-25 06:44:57 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-11-25 06:44:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
20-11-25 06:44:57 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_vscode
20-11-25 06:44:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
20-11-25 13:40:20 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-11-25 13:40:20 ERROR (MainThread) [supervisor.utils.json] Can't write /data/updater.json: [Errno 30] Read-only file system: '/data/updater.json'
20-11-25 13:40:34 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
20-11-25 13:40:34 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository
20-11-25 13:40:34 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/hassio-addons repository
20-11-25 13:40:34 ERROR (MainThread) [supervisor.store.git] Can't update https://github.com/hassio-addons/repository repo: Cmd('git') failed due to: exit code(255)
  cmdline: git fetch --depth=1 --update-shallow -v origin
  stderr: 'error: cannot open .git/FETCH_HEAD: Read-only file system'.
20-11-25 13:40:34 ERROR (MainThread) [supervisor.store.git] Can't update https://github.com/thomasmauerer/hassio-addons repo: Cmd('git') failed due to: exit code(255)
  cmdline: git fetch --depth=1 --update-shallow -v origin
  stderr: 'error: cannot open .git/FETCH_HEAD: Read-only file system'.
20-11-25 13:40:34 ERROR (MainThread) [supervisor.store.git] Can't update https://github.com/home-assistant/hassio-addons repo: Cmd('git') failed due to: exit code(255)
  cmdline: git fetch --depth=1 --update-shallow -v origin
  stderr: 'error: cannot open .git/FETCH_HEAD: Read-only file system'.
20-11-25 13:40:34 INFO (MainThread) [supervisor.store] Loading add-ons from store: 67 all - 0 new - 0 remove
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
20-11-25 13:53:56 INFO (MainThread) [supervisor.host.info] Updating local host information
20-11-25 13:53:57 INFO (MainThread) [supervisor.host.services] Updating service information
20-11-25 13:53:57 INFO (MainThread) [supervisor.host.network] Updating local network information
20-11-25 13:53:58 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
Failed to load cookie file from cookie: No such file or directory
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied
/usr/lib/bashio/jq.sh: line 26: cannot create temp file for here-document: Permission denied`

LeChuck2909, 2020-11-25:
this is the same root cause right ? again i read things like "read only" and "permission denied"

agners, 2020-11-25:
Yeah could be the same root cause. If it happens again, can you also check the "Log Provider" Host and see if that shows errors?

LeChuck2909, 2020-12-11:
ok, so since i added `usb-storage.quirks=aaaa:bbbb:u`, there is no issue anyome. HA with HA OS works butter smooth. and very fast.

as the noob that i am, maybe you can help me bring this a little into perspective (i know, that this question should be asked in rpi forums). when i think on USB in general, i think on "just works" "is compatible" and so on ... now here this is not the case. why extra lines for my SATA to USB adapter are necessary, or more precise, will this be different, on some point in the future ? as far as i could follow the conversation on rpi forum i limit the performance to ensure communication.

thanks a lot.

closed 2021-01-12 by LeChuck2909