home-assistant/operating-system #1011: HassOS 5.5 - RPi 4/4GB/SSD - boot problem with RFLink Arduino set

miviho, 2020-11-20:
With HassOS 5.x the RPi won't boot from SSD if the Arduino mega is connected to the USB. Boot from SD card or booting Raspbian from SSD with the Arduino connected is not a problem. Tried a different power supply (3A and 5A version), tested with Startech and Icy Box USB 3.0 SATA connector and Kingston SA400S37/240G SSD disk. Using a (powered) USB hub doesn't make any difference.

Latest Raspberry boot loader c305221a
[image]

Anyone with suggestion to try?

agners, 2020-11-23:
Hm, that is interesting. Likely U-Boot having troubles to enumerate the USB devices as soon as Arduino Mega is connected. Since Raspberry Pi OS doesn't use U-Boot, that would explain why it works with it.

Do you have a USB to UART TTL adapter? If so, a capture of the serial console would be interesting. See also: https://github.com/home-assistant/operating-system/tree/rel-4/Documentation/boards/raspberrypi#serial-console

miviho, 2020-11-23:
I don't have one but will have a look later this evening to get one so I can share the output.

miviho, 2020-11-25:
Here is the result:

U-Boot 2020.10 (Nov 11 2020 - 13:45:31 +0000)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03112)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... WARN halted endpoint, queueing URB anyway.
Unexpected XHCI event TRB, skipping... (3aef7cf0 00000000 13000000 03008400)
BUG at drivers/usb/host/xhci-ring.c:498/abort_td()!
BUG!
resetting ...

agners, 2020-11-25:
Hm, that's a new, haven't seen this problem before. It looks like U-Boot has problems talking to the hard drive via USB.

Is the output always the same or is it (slightly) different?

miviho, 2020-11-26:
It is always the same and only change when I plug in the USB Conbee,

DRAM:  3.9 GiB
RPI 4 Model B (0xc03112)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... WARN halted endpoint, queueing URB anyway.
Unexpected XHCI event TRB, skipping... **_(3aef7d20 00000000 13000000 04008400)_**
BUG at drivers/usb/host/xhci-ring.c:498/abort_td()!
BUG!
resetting ...

miviho, 2020-11-26:
[RPI boot - later Arduino.txt](https://github.com/home-assistant/operating-system/files/5601409/RPI.boot.-.later.Arduino.txt)

This is the dump when the Arduino is not connected at boot but later after [36.165734]

agners, 2020-11-26:
Unfortunately, I don't have a good solution for this. This is likely a bug in the USB stack of the U-Boot bootloader. Raspbian doesn't use U-Boot and probably doesn't see the problem due to that. To fix it I would need a device to reproduce it here, and it would take time...

I think the best approach is to boot from SD card and use the `datactl` command to move the data to the USB attached SSD storage. That way you don't need to boot from USB SSD and Home Assistant has the full bandwidth of the SSD for data read/writes.

miviho, 2020-11-27:
Who is responsible for the U-Boot bootloader? Perhaps it is worth to inform them about the bug?
In the meantime I will go for the move from SD to SSD.

tempesta-home, 2020-12-22:
Hi, I have the same problem, my pi4 doesn't boot or reboot with rflink connected.  
When I disconnect rflink, it suddendly starts with boot, it seems that it was somehow waiting for something that never arrives.
If I re-connect rflink after some seconds, even though HA core is still starting, everything works fine.
I dont have USB SSD, I'm trying to boot from SD card.
Hope that helps.

SergeyAnokhin, 2020-12-29:
I have the same problem: cant boot. 
PI4 + Miscrosd (my HA or clean downloaded from website). OpenZWave USB is OK, RFXCom is OK.
Raspbian cant boot with RFLink.

SergeyAnokhin, 2020-12-29:
Can fix this issue temporary : disconect RFLink before boot and reconnect after few seconds.

katonage, 2021-01-02:
Same issue here: Rpi4 with everything upgraded to the latest version. Need to pull out USB of the rflink for the system / home assistant to start. After that, usb can be plugged in and ha server restarted to have the rflink devices.
That is quite an issue as the automation center does not returns live after a power failure without manual user intervention...

katonage, 2021-01-29:
This ticket was originally an **ssd** boot issue, however may be related to that, it seems a few people are experiencing boot problems with the **rflink** arduino attached. Should we open a separate ticket for this specific setting?  @agners

mischung, 2021-01-30:
Hi guys,

just started working with my RF-Link gateway and got the same problem.

As long as the Arduino Mega (RF-Link) is connected to any USB 2.0 port, the Pi (4 - Rev 1.1 / 4GB / only SD with HA) will not start at all.
There is also absolutely no HDMI output when it's stuck within this reboot loop.

Would be happy helping with debugging and solving this issue.

mrsilva1986, 2021-02-01:
Hi,  have exactly the same issue. Already tried to boot from ssd or sd card the issue continues. Only happend with my new raspberry pi 4. My old 3 boots fine.
Did someone find a solution for this?

katonage, 2021-02-01:
Is it possible to connect the (Nodo shop kit) RFLink gateway with ethernet /TCP as a workaround?

mrsilva1986, 2021-02-02:
> Is it possible to connect the (Nodo shop kit) RFLink gateway with ethernet /TCP as a workaround?

How? The arduino mega board don't have a ethernet connection. The only workaround available at this moment is to start the raspeberry without the arduino connected and only connect it after boot. 
That is bad in case of need to reboot your homeassistant remotely or have a power cut.

mischung, 2021-02-02:
> How? The arduino mega board don't have a ethernet connection. 

There is a description for using TCP-Mode on the RFLink integration page of HA, but you will need another host with network access to enable this mode (e.g. an ESP8266). 

This would be a workaround, but it's not an option for me, because I want to limit the total amount of devices.

Maybe there is a possibility to enable USB power later during the boot-process? Just an idea...

mrsilva1986, 2021-02-02:
> This would be a workaround, but it's not an option for me, because I want to limit the total amount of devices.

Neither for me, just bouth a rasperry pi 4 to replace my 3 to have some performance improvement. Not to have 2 doing the same thing. 😞 
 
> Maybe there is a possibility to enable USB power later during the boot-process? Just an idea...

I think not, at least did not find any reference to that... But see that the bootloader on pi4 can be changed to only boot SDCard's. Did anyone already try that? I'm looking to that, but need, aparently, raspian os to do the change. 😩

agners, 2021-02-03:
The report on the U-Boot mailing list is probably worth linking: https://www.mail-archive.com/u-boot@lists.denx.de/msg390604.html

However, no hints/updates so far.

jacek903, 2021-02-04:
Hi, I have the same issue. Did someone tried the solution with cutting the power lines of usb cable or cut the usb power trace on the Arduino Mega and power Mega independently?

ninjax95, 2021-02-07:
> Same issue here: Rpi4 with everything upgraded to the latest version. Need to pull out USB of the rflink for the system / home assistant to start. After that, usb can be plugged in and ha server [...]

Same problem, do you have a solution ?

GieBek, 2021-02-09:
Hi everyone, sorry for my english.

I have been struggling with this problem for a long time, i found somewhere on the forums that it starts working after separating the power supply from the arduino but I have not tested it.
So friend jacek903 is following a good lead. For now, to solve the problem, i have RFlink and 2x arduino-multi-relay disconnected :(

Cahuete65, 2021-02-11:
Hello  tous,

Same problem for me:

RPI4 4GB - SSD - Arduino mega with RFLink (firmware R48)
Raspbian GNU/Linux 10 (buster) lite
To solve it and boot i disconnect the arduino usb or the power supply (9v 1A) and after few seconds i can plugg all
I try also booting with sd32 card and ardui => same issue
I will try to found a second usb cable and try  jacek903 idea :)
 bonne journée

Cahuete65, 2021-02-12:
I come back to tell that i change the usb cable (before cuting the powers lines) and now all is ok with the new usb cable :)
Solved for me, thank's all

ninjax95, 2021-02-12:
> I come back to tell that i change the usb cable (before cuting the powers lines) and now all is ok with the new usb cable :)
> Solved for me, thank's all

I change usb cable, i cut power pin and power with an adapter but same, don't works for me :(

Cahuete65, 2021-02-17:
> > I come back to tell that i change the usb cable (before cuting the powers lines) and now all is ok with the new usb cable :)
> > Solved for me, thank's all
> 
> I change usb cable, i cut power pin and power with an adapter but same, don't works for me :(

Hello ninjax95, i did not cut the wires i just changed the cable.

ninjax95, 2021-02-18:
> Hello ninjax95, i did not cut the wires i just changed the cable.

I understood but as it did not work I also cut the power wires. Do you have a reference of your cable, or something in particular? It just does data maybe?

katonage, 2021-02-20:
Is it maybe possible to switch off searching for USB devices, usb boot in the U-Boot config? If sy knows how to do it, pls describe.

RoosterX, 2021-03-08:
Yeah same problem here, RPI 4 4GB won't boot with Rflink device connected.

mSoftMS, 2021-04-29:
I have the same problem, RPI 4 4GB won't boot with arduino mega connected... Has anyone found a solution to this problem?

GieBek, 2021-05-06:
Hello, it's been so long now that I'm starting to worry. There is really no solution to this?

miviho, 2021-05-06:
All the workarounds didn't really work for me as an usefull solution for me. I gave up on the raspberry pi for now and switched to a cheap NUC running Home Assistant on Docker. IT works much more stable. I will keep following this thread but not sure if I go back to the raspberry PI ... the NUC is a nice piece of hardware together with docker.

katonage, 2021-05-06:
RPI4 is one of the most popular platform for HA. I think it should be solved somehow...
there should be definitely some configurable bootloader that can be instructed not to look for USB in this case...

bazirico, 2021-06-13:
Same problem here. @Cahuete65 seems to have solve this problem changing the usb cable. Does anyone else tried this and can give us the reference of the new usb cable ?

Can we help in any ways to solve this issue ?

ME-TYHO, 2021-06-29:
I have same problem. With RFlink disconnected, Home Assistant OS on RPI4 4GB start right away. With RFlink connected it won't start and the PI keeps rebooting.
Changing USB (longer and shorter) didn't do the trick
On my RPI3B+ i didn't have this problem.

When Home Assistant OS is booted i can just plug in RFlink and everything is worinking fine.

It seems like the pi tries to boot from the arduino but then crashes and restarts and repeats endlessly.

As work around is there a way to create a delay to the usb port and first wait for HASSOS to boot.

Julius112, 2021-07-09:
I can confirm the problem is also existent on a Helios64 NAS not running HassOS but running uboot. So I guess it's actually a uboot problem...

katonage, 2021-07-09:
I tried proposing the problem on discord chat with no luck. Anyone closer to uboot or the right knowledgeables?

Julius112, 2021-07-16:
I got a "workaround" on a different forum. You could switch to the https://docs.openmqttgateway.com/ and use RFLink over Ethernet/MQTT. This way you do not need the USB connection. (https://forum.armbian.com/topic/18587-fail-to-boot-with-serial-device-connected-to-usb/?do=findComment&comment=126811)

shahglobal, 2021-07-23:
Can rx/tx from rflink arduino be connected to tx/rx pins in pi4 and node red serial node to bring the data in/out?

111Sebu111, 2021-09-11:
I also have the same problem Rpi4 8GB with RFlink.

borwolf, 2021-09-21:
Same problem here. Rpi4 8GB with RFlink and RFXcom.

agners, 2021-10-05:
This is fixed with HAOS 6.5 (currently on beta channel).

closed 2021-10-05 by agners

mischung, 2021-10-05:
Just joined the beta channel and can confirm it is working with the RFLink module connected.

Thanks for your great work and support!