home-assistant/operating-system #1012: HassOS 5.x & 5.5

BlackTurtle123, 2020-11-21:
**Hardware Environment**

- [ ] Raspberry Pi4 8GB
- SSD with external power resource

**Home Assistant OS release:**
- [ ] Fresh installation of release 5.3 and 5.5

**Description of problem:**
When booting from ssd connected through USB, the installation doesn't work.
Booting from sd card does work. booting raspian from the same ssd does also work.

BlackTurtle123, 2020-11-21:
Update 5.2 does boot

agners, 2020-11-23:
Hm, that is weird, as U-Boot used in 5.2 basically has all the patches which ultimately went into U-Boot 2020.10, which is part of 5.3/5.5.

Does maybe reboot vs. cold boot make a difference?

What firmware are you using? Can you maybe try with the latest stable 2020-09-03 (update using Raspberry Pi OS).

BlackTurtle123, 2020-11-23:
I was using the latest rpi non stable release I think.
Cold or hot boot didn't make any change.

agners, 2020-11-24:
Ok, can you also check firmware?

BlackTurtle123, 2020-11-24:
I will check if I still got the rpi card. Around.
Something else I noticed today, the hassos goes into read-only mode. Have
to look into the details. A reboot/power reset fixes this.

BlackTurtle123, 2020-11-25:
> Ok, can you also check firmware?

Aug 19 2020 17:38:16 
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)

BlackTurtle123, 2020-11-25:
Also what I noticed is that my ssd usb, powered via external power. samsung evo 970, regulary goes into read only mode.
Nothing works anymore then, and only option left is to fully reboot everything. Not sure if it has something to do with this?

agners, 2020-11-25:
The read-only mode could be due to UAS mode issue. Are you using `usb-storage.quirks`? https://www.raspberrypi.org/forums/viewtopic.php?t=245931

As for the boot issue on 5.3/5.5: Do you use a custom config.txt?

Do you have a USB to UART TTL adapter? If so, a capture of the serial console would be interesting. See also: https://github.com/home-assistant/operating-system/tree/rel-4/Documentation/boards/raspberrypi#serial-console

BlackTurtle123, 2020-11-25:
I tried today to set the uas settings.
Let's see.
It didn't worked to make my usb3 port work. But if it doesn't crash anymore
i am already happy. Will keep posted.

BlackTurtle123, 2020-11-25:
Update:
The custom boot didn't helped the issue.
The system crashed with read errors just now again, within 4 hours.

BlackTurtle123, 2020-11-25:
Update 2: reflashed and tested the USB once again. 5.5 seems to work now. With usb2 port. And still these read-only file system errors.

Sadly don't have this adapter by the hand

agners, 2020-11-25:
Can you post logs which show the read errors?

BlackTurtle123, 2020-11-26:
Yes, will do. Sadly when this happens my system becomes dead and so can't ssh or access logs besides by screen. So can't do it now since by booting it of they become corrupt. However will makeba picture next time

BlackTurtle123, 2020-11-26:
[image]

After 22 hours roughly from no issues it appeared again. No change in hardware.

BlackTurtle123, 2020-11-29:
It happened again today. 
[image]

BlackTurtle123, 2020-11-29:
[image]

agners, 2020-11-30:
Kinda hard to tell, but the most likely cause is USB issues of some kind. You write "SSD with external power", maybe this power source is not good (enough?) What kind of SSD are you using? Maybe a different helps...

Unfortunately SSD boot with RPi4 is a bit brittle, and it works better on some SSDs and not so well on others... It's a bit a hit and miss.

BlackTurtle123, 2020-11-30:
I tried a different usb hub with power source.
I use a Samsung nvme. It seemq under Ubuntu there might also be problems with it. 
So i guess it's a broader topic. I now try to edit some boot params and we will see..

BlackTurtle123, 2020-12-05:
Does hassos can handle all the boot params Ubuntu and other os'es are able to use?

BlackTurtle123, 2020-12-09:
Another update, also HassOS 5.6 seems to freeze after a while an doesn't work on usb3 with this hardware. However the not working also happens with raspian os... so I also opened a ticket there. Let's see

agners, 2020-12-10:
@BlackTurtle123 thanks for the update! Also note that HAOS 5.6 uses the exact same git hash as the latest Raspberry Pi OS 2020-12-02/04 (https://github.com/raspberrypi/linux/commit/9797f1a4938c20139b00a25de93cc99efb5c291b).

BlackTurtle123, 2020-12-11:
I sended back a usb3 external power source after I seen that my ssd booted if directly connected. Until it wouls require to much power consumption. So it seems hardware related/external hub related. Now trying something out. Will report back if it crashed again like before with an unreachable disk.

BlackTurtle123, 2020-12-12:
Stable now since 2days and 9 hours. Never been this good. Let's see if it's just a lucker or not

closed 2021-05-12 by stale[bot]