home-assistant/operating-system #1013: Memory used reported incorrectly to Proxmox hypervisor

gdiciancia, 2020-11-22:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [X] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)
         Proxmox

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.yv
- Additional information (if accessible):

Version | 0.118.2
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.6
Operating System Family | Linux
Operating System Version | 5.4.77
CPU Architecture | x86_64

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

The "System" tab is reporting roughly 1Gb of memory used (Supervisor + CORE), however Proxmox reports 94.5% usage on 8Gb of memory. FYI, the guest agent seems to be working correctly since Proxmox reports the IP address correctly.

agners, 2020-11-23:
The System tab only takes Supervisor and Core into account, but there are other services running on the OS (Docker itself, CoreDNS, etc). Furthermore, Linux OS tries to make use of the available memory as much as possible by caching disk blocks (page cache). So if you give Linux 8GB of memory, over time it will use all of it as a page cache.

gdiciancia, 2020-11-23:
Ok, this is from Glances add-on (used+cached is 6.3Gb, proxmox is reporting 7.13Gb used)
MEM
14.1%
total:
7.77G
used:
1.1G
free:
6.67G
active:
2.44G
inactive:
4.09G
buffers:
371M
cached:
5.20G

agners, 2020-11-24:
Yeah that seems to add up: The total memory consumption of the OS is close to 8GB, which is what Proxmox reported.

So this is not an OS bug.

closed 2020-11-24 by agners