home-assistant/operating-system #1014: Random Mac Address on the Odroid XU4

xblackedemonx, 2020-11-22:
This is pretty much the same issue as this thread: https://github.com/home-assistant/operating-system/issues/551

but it's on the Odroid XU4.

Hi !

Adding @pvizeli for viz

HassOS release with the issue:

Host Operating System | HassOS 4.17
-- | --
Update Channel | stable
Supervisor Version | 2020.11.0
Docker Version | 19.03.12

Description of problem:
At every reboot, the Odroid box is getting a new (random ?) mac address.
Some digging yields that there may be a parameter to set in some boot config files : https://forum.libreelec.tv/thread/2308-odroid-c2-7-90-008-mac-address-changes-randomly-after-reboot/?postID=18264#post18264

Let me know if I can be of any help :)

Thanks !

agners, 2020-11-23:
Just tried to reproduce this on my, and it seems to work fine here:
```
# ip addr show dev eth0
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:1e:06:37:80:7a brd ff:ff:ff:ff:ff:ff
...
```

There are some indices on ODROID forum it seems that older devices did not have a stable MAC address flashed for some reason. If found this how-to set a stable MAC address: https://wiki.odroid.com/odroid-xu4/troubleshooting/change_mac

Another option might be to use a custom NetworkManager configuration file with `cloned-mac-address=stable`. Adding this section:

```
[ethernet]
cloned-mac-address=stable
```

See https://github.com/home-assistant/operating-system/blob/rel-4/Documentation/network.md.

xblackedemonx, 2020-11-23:
I will try your solution. thank you!

agners, 2020-11-29:
Consider this closed. Please reopen if the solution doesn't work for you.

closed 2020-11-29 by agners