home-assistant/operating-system #1024: Rpi3 with SSD, won't boot after OS update from 4.17 to 4.18

godinperson, 2020-11-26:
All I get is a black screen. On my SSD, I have a light, normally, that light would blink. Now, it is steady blue. Been working for the last 1-2 years.

Thanks

agners, 2020-11-26:
This is very weird, as you can see from the changelog, the 4.18 release did not change anything for the RPi's:
https://github.com/home-assistant/operating-system/releases/tag/4.18

Could it be that due to the reboot some other problem got uncovered? Maybe hardware failing due to age...?

godinperson, 2020-11-26:
I don't think so, it was super stable. Could it be because of disabling USB enumeration. I'm having weird issues with USB. On the Rpi3, I have a Aeotec z-stick and the SSD. Sometimes, the z-stick won't light up. Unplugging and plugging the power back will bring the z-stick back

godinperson, 2020-11-26:
Do you think I need to reinstall or there is some things we can try before that could help identify the issue?

Mariusthvdb, 2020-11-26:
got the same thing, even tried hard unplug, nothing helps so far, my Aeotec stick doesn't light up either...

Thrasher2020, 2020-11-26:
Same - trying to restore my snapshot with a clean install. Taking ages :(

donkawechico, 2020-11-27:
Same issue here on an RPi4. I don't know yet about black screen, but I can't ssh into it anymore and HA doesn't load. I just get the solid red LED light.

In a little bit I'll try hooking up a monitor to it to see about black screen

jaxonstagecrew, 2020-11-27:
Same issue here on my RPI4.  No UI or a way to SSH into terminal.  After an hour I had to give up and restore a snapshot.

imbrianj, 2020-11-27:
Same issue here - RPi4, 4gb with Nortek Zwave/Zigbee stick running off microsd card.  Did the update for 118.4 then immediately after, upgrade to 4.18 and failed to boot.  Unplugged, plugged back in - saw some disc activity - but router showed no host.  Formatting now to restore snapshot.

haddood, 2020-11-27:
same here RPI3B+ and SSD, the system did sometimes not boot before, but power recycling solved the issue. after todays upgrade   it does not boot no matter what I try.

excellapay, 2020-11-27:
RPI 4B, 4G with 64 OS Bricked.  It has downloaded over 10GB from the net.  I can never get into UI.  It just seems to boot and hang.

donkawechico, 2020-11-27:
Writing 4.17 to SD card brought back HA fresh install interface (showing "create new account" splash screen).

Tried onboarding with recent snapshot but the UI didn't show any sort of confirmation that it was working. After awhile, the UI stopped working, throwing 500. Wasn't sure if that meant that the snapshot loading was working or not, so after waiting 5-10 minutes I restarted the Pi4.

Pi4 booted back up, but the UI was still throwing 500. After about 15-20 minutes, UI finally came back.

c1xx, 2020-11-27:
Same issue here, update to 4.18 and after reboot my HASS on my RPi4 is not coming up. I hope that "snapshot" Backup contains all needed addons.

ollimagic, 2020-11-27:
Hello I have the same Problem RPI4 4 GB 128 GB SD Card dont but after update to 4.18 i need help any idea???

Mariusthvdb, 2020-11-27:
@donkawechico how do you only write the 4.17 back to the SD? Or do you mean the regular fresh install, to download from https://www.home-assistant.io/hassio/installation/ ?

ollimagic, 2020-11-27:
write back to 4.17 without destroy my installation?

c1xx, 2020-11-27:
I used 4.17 on other SD-Card, installed it and restored my backup from snapshot

ollimagic, 2020-11-27:
is there no other way??

c1xx, 2020-11-27:
> is there no other way??

I don't know currently about other ways....only the HASS Team can tell us other way(s)

agners, 2020-11-27:
Sorry folks for the troubles, I am trying to reproduce the issue on my end. The 4.18 release was on beta.

@ollimagic quite likely that there is a way, I need to understand the problem first.

ollimagic, 2020-11-27:
haha i can imagin i hope you will find it i have my sd card already on my
pc if you need a log tell me where i find it and i send it
after the update you can even see on the monitor only a black screen no
boot nothing happen

agners, 2020-11-27:
Unfortunately the boot script contained an error. The good news is, this is relatively simple tor recover: Plug the SD card/USB disk into a PC, make sure you can access the (first) boot partition with `config.txt` on it etc. and replace the file `boot.scr` with the one from the 4.17 release. Here are they as a zip file:
[boot.scr-4.17-rpi-32-bit.zip](https://github.com/home-assistant/operating-system/files/5606998/boot.scr-4.17-rpi-32-bit.zip)
[boot.scr-4.17-rpi-64-bit.zip](https://github.com/home-assistant/operating-system/files/5606999/boot.scr-4.17-rpi-64-bit.zip)

c1xx, 2020-11-27:
> Unfortunately the boot script contained an error. The good news is, this is relatively simple tor recover: Plug the SD card/USB disk into a PC, make sure you can access the (first) boot partition [...]

Thank you very much!!

Mariusthvdb, 2020-11-27:
> Unfortunately the boot script contained an error. The good news is, this is relatively simple tor recover: Plug the SD card/USB disk into a PC, make sure you can access the (first) boot partition with `config.txt` on it etc. and replace the file `boot.scr` with the one from the 4.17 release. 

how to do this on a Mac? My Finder always complains it can't read those sd-cards, so there's no way to overwrite anything at all, other than flashing the sd-card with Etcher?

MrYadro, 2020-11-27:
> > Unfortunately the boot script contained an error. The good news is, this is relatively simple tor recover: Plug the SD card/USB disk into a PC, make sure you can access the (first) boot partition with `config.txt` on it etc. and replace the file `boot.scr` with the one from the 4.17 release.
> 
> how to do this on a Mac? My Finder always complains it can't read those sd-cards, so there's no way to overwrite anything at all, other than flashing the sd-card with Etcher?
> 
> but besides that, a much more important question: shouldn't this whole bricking of the HA instance because of a small typo in the boot script be impossible.
> 
> Why is this **not guarded**? I mean some sort of self test before executing the script should be done? Even in a beta version.

ext4 partitions can't be mounted as RW on macOS, you have to use VM's with linux or Paragon thing for 30+ euros

pvizeli, 2020-11-27:
> ext4 partitions can't be mounted as RW on macOS

It's a fat32 partition

MrYadro, 2020-11-27:
> It's a fat32 partition
Oh, my bad...
[image]

Mariusthvdb, 2020-11-27:
> > ext4 partitions can't be mounted as RW on macOS
> 
> It's a fat32 partition

still, SD-card not readable on a Mac?

daconrad0, 2020-11-27:
So is this now fixed and can try another upgrade.  I had to start over and restore a snapshot last night?

godinperson, 2020-11-27:
> Unfortunately the boot script contained an error. The good news is, this is relatively simple tor recover: Plug the SD card/USB disk into a PC, make sure you can access the (first) boot partition [...]

Thanks, problem solved!

ollimagic, 2020-11-27:
thanks it was working

Rxich, 2020-12-01:
> ~ext4 partitions can't be mounted as RW on macOS, you have to use VM's with linux or Paragon thing for 30+ euros~

do you happen to have access to an Android cellular device?

Mariusthvdb, 2020-12-01:
afraid not... am up and running from my snapshots, so no worries there. My question is still of a 'good to know' position: How to only update/downgrade the OS on the SD card, and not the rest of the system.

agners, 2020-12-01:
> So is this now fixed and can try another upgrade. I had to start over and restore a snapshot last night?

It is fixed but there is no new release. Once 4.19 is out things should be fine again.

agners, 2020-12-01:
@Mariusthvdb `ha os update --version x.y` does update/downgrade the OS only.

Mariusthvdb, 2020-12-02:
ya, thanks. I am aware of the cl command of course, but unfortunately, this update bricked the instance so we couldn't use that..
hence my question how to update the OS only from the outside.

agners, 2020-12-10:
@Mariusthvdb not sure if this is still relevant, but just for the records: You can fix a an instance which got stuck due to this update using the method explained above: https://github.com/home-assistant/operating-system/issues/1024#issuecomment-734735837

closed 2020-12-10 by agners

Mariusthvdb, 2020-12-10:
> @Mariusthvdb not sure if this is still relevant, but just for the records: You can fix a an instance which got stuck due to this update using the method explained above: [#1024 (comment)](https://github.com/home-assistant/operating-system/issues/1024#issuecomment-734735837)

sure, but, as said, my questions was how to do that on a Mac, not a PC. Don't have a spare PC lying around for bricked HA Rpi instances ;-)  Anyways, we're back in business for the time being.