home-assistant/operating-system #1040: datactl move not using external drive?

UnderpantsGnome, 2020-12-01:
**Hardware Environment**

- [x] Raspberry Pi 4
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 4.16
- [ ] Updated from version x.y
- Additional information (if accessible):

```
arch: aarch64
channel: stable
docker: 19.03.12
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "4.16"
homeassistant: 0.118.4
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: HassOS 4.16
supervisor: 2020.11.0
supported: true
supported_arch:
- aarch64
- armv7
- armhf
```

**Supervisor logs:**

```20-12-01 00:18:21 INFO (MainThread) [supervisor.api] Stopping API on 172.30.32.2
20-12-01 00:18:21 INFO (MainThread) [supervisor.misc.hwmon] Stopped Supervisor hardware monitor
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
20-12-01 00:18:21 INFO (MainThread) [supervisor.core] Supervisor is down
20-12-01 00:18:21 INFO (MainThread) [__main__] Closing Supervisor
[s6-finish] sending all processes the TERM signal.
[s6-finish] sending all processes the KILL signal and exiting.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[00:18:57] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:18:58] INFO: Starting local supervisor watchdog...
20-12-01 00:19:01 INFO (MainThread) [__main__] Initializing Supervisor setup
20-12-01 00:19:01 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
20-12-01 00:19:01 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2020.11.0
20-12-01 00:19:01 INFO (MainThread) [__main__] Setting up Supervisor
20-12-01 00:19:01 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
20-12-01 00:19:02 INFO (MainThread) [supervisor.host.info] Updating local host information
20-12-01 00:19:02 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-01 00:19:02 INFO (MainThread) [supervisor.host.network] Updating local network information
20-12-01 00:19:03 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
[... 692 lines cut ...]
tmpfs                     1.9G      1.0M      1.9G   0% /run
tmpfs                     1.9G         0      1.9G   0% /sys/fs/cgroup
tmpfs                     1.9G      1.0M      1.9G   0% /etc/machine-id
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /mnt/overlay
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/docker
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/dropbear
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/modprobe.d
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/modules-load.d
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/udev/rules.d
/dev/mmcblk0p1           31.9M      3.1M     28.8M  10% /mnt/boot
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /root/.docker
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /root/.ssh
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/NetworkManager/system-connections
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/hostname
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/hosts
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/systemd/timesyncd.conf
/dev/mmcblk0p8           57.8G      3.3G     51.6G   6% /mnt/data
/dev/zram2               15.5M     44.0K     14.3M   0% /tmp
/dev/zram1               31.0M    128.0K     28.6M   0% /var
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /var/lib/NetworkManager
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /var/lib/bluetooth
/dev/mmcblk0p8           57.8G      3.3G     51.6G   6% /var/lib/docker
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /var/lib/systemd
/dev/mmcblk0p8           57.8G      3.3G     51.6G   6% /var/log/journal
```

agners, 2020-12-01:
Hm, it seems the system opted to use the SD card non-the-less.

Do you have that system still booted after that first reboot after using the `datactl` command? If so, can you use the OS console and run `journalctl -u hassos-data.service`, it should get a bit more logs about the moving process.

UnderpantsGnome, 2020-12-01:
Last night I decided to delete the partition and take another try, it was still doing the setting up the data partition after 14 hours, then... power outage, so I'll have to take another crack at it. 😞

agners, 2020-12-01:
So the process was stuck when running `datactl` or the reboot after?

It could be a disk size problem, I never tried with such a large disk.

UnderpantsGnome, 2020-12-01:
On the reboot after, this is where it's been sitting for the last 30 minutes or so.
[image]

UnderpantsGnome, 2020-12-01:
@agners here is the additional logging

```
# journalctl -u hassos-data.service
-- Logs begin at Sun 2020-09-20 10:43:58 UTC, end at Tue 2020-12-01 18:41:07 UTC. --
Sep 20 10:43:59 homeassistant systemd[1]: Condition check resulted in HassOS data partition being skipped.
-- Reboot --
Sep 20 10:43:59 homeassistant systemd[1]: Condition check resulted in HassOS data partition being skipped.
-- Reboot --
Sep 20 10:43:59 homeassistant systemd[1]: Starting HassOS data partition...
-- Reboot --
Sep 20 10:43:59 homeassistant systemd[1]: Condition check resulted in HassOS data partition being skipped.
-- Reboot --
Sep 20 10:43:59 homeassistant systemd[1]: Condition check resulted in HassOS data partition being skipped.
-- Reboot --
Sep 20 10:44:00 homeassistant systemd[1]: Condition check resulted in HassOS data partition being skipped.
-- Reboot --
Sep 20 10:44:00 homeassistant systemd[1]: Condition check resulted in HassOS data partition being skipped.
-- Reboot --
Sep 20 10:44:04 homeassistant hassos-data[196]: [INFO] Cleanup device /dev/sda!
Sep 20 10:44:05 homeassistant hassos-data[271]: ***************************************************************
Sep 20 10:44:05 homeassistant hassos-data[271]: Found invalid GPT and valid MBR; converting MBR to GPT format
Sep 20 10:44:05 homeassistant hassos-data[271]: in memory.
Sep 20 10:44:05 homeassistant hassos-data[271]: ***************************************************************
Sep 20 10:44:05 homeassistant hassos-data[271]: Warning! Secondary partition table overlaps the last partition by
Sep 20 10:44:05 homeassistant hassos-data[271]: 33 blocks!
Sep 20 10:44:05 homeassistant hassos-data[271]: You will need to delete this partition or resize it in another utility.
Sep 20 10:44:05 homeassistant hassos-data[271]: GPT data structures destroyed! You may now partition the disk using fdisk or
Sep 20 10:44:05 homeassistant hassos-data[271]: other utilities.
Sep 20 10:44:05 homeassistant hassos-data[196]: [INFO] Create new hassos-data partition
Sep 20 10:44:07 homeassistant hassos-data[273]: Creating new GPT entries in memory.
Sep 20 10:44:07 homeassistant hassos-data[273]: The operation has completed successfully.
Sep 20 10:44:08 homeassistant hassos-data[274]: Setting name!
Sep 20 10:44:08 homeassistant hassos-data[274]: partNum is 0
Sep 20 10:44:08 homeassistant hassos-data[274]: The operation has completed successfully.
Sep 20 10:44:08 homeassistant hassos-data[275]: No problems found. 2014 free sectors (1007.0 KiB) available in 1
Sep 20 10:44:08 homeassistant hassos-data[275]: segments, the largest of which is 2014 (1007.0 KiB) in size.
Sep 20 10:44:08 homeassistant hassos-data[196]: [INFO] Move hassos-data from /dev/mmcblk0p8 to /dev/sda1
Sep 20 11:31:27 homeassistant hassos-data[282]: dd: error writing '/dev/sda1': No space left on device
Sep 20 11:33:07 homeassistant hassos-data[196]: [ERROR] Data copy fails!
Sep 20 11:33:09 homeassistant hassos-data[322]: The operation has completed successfully.
Sep 20 11:33:09 homeassistant hassos-data[323]: partx: specified range <1:0> does not make sense
Sep 20 11:33:09 homeassistant systemd[1]: hassos-data.service: Main process exited, code=exited, status=1/FAILURE
Sep 20 11:33:09 homeassistant systemd[1]: hassos-data.service: Failed with result 'exit-code'.
Sep 20 11:33:09 homeassistant systemd[1]: Failed to start HassOS data partition.
```

UnderpantsGnome, 2020-12-01:
Based on the errors in that log output, I wiped the drive and set it up for GPT, then reran `datactl move /dev/sda` rebooted, it's been on the setting up data partition for a couple hours now. Is there a way I can modify the SD card and/or the external drive to make HA think it's already done?

agners, 2020-12-01:
Hm, the screenshot with the stack traces don't look good. It seems as if the bus the harddrive is connected to (USB?) or the harddrive itself had some issues writing data which made the "write" commands time out. Is that driver powered externally? Are you sure the harddrive is fine?

The log then sounds like Home Assistant OS tried to create a very small partition which then filled up immediately. I think ideally you should not have any partition table or partition since this is done by the `hassos-data.service` started on first boot.

I am planning to work on the feature tomorrow to improve its usability. Mainly I like to move more logic into datactl to catch problems before moving data.

UnderpantsGnome, 2020-12-01:
The drive is powered externally. I ran Gparted on it and it says it fixed a bunch of issues. I reattached it to the pi but HA still isn't using it. Tried `datactl` again but got

```
# datactl move /dev/sda
[ERROR] Can't be the same disk!
```

🤷‍♂️

agners, 2020-12-02:
Yeah I think that happens when there is still some parts of the partition table there. Can you try:

```
sgdisk --zap-all /dev/sda 
```

before using datactl?

UnderpantsGnome, 2020-12-02:
Here is what is currently on the external drive.

```
# cd supervisor/
# ls -l
-rw-------    1 root     root         19321 Dec  1 19:16 addons.json
drwxr-xr-x    2 root     root          4096 Sep 20 10:44 apparmor
-rw-------    1 root     root            70 Dec  1 19:07 audio.json
-rw-------    1 root     root           209 Dec  1 19:15 cli.json
-rw-------    1 root     root           345 Dec  1 19:17 config.json
-rw-------    1 root     root           401 Nov 30 22:46 discovery.json
-rw-------    1 root     root            92 Dec  1 19:07 dns.json
-rw-------    1 root     root            22 Nov 30 22:45 docker.json
-rw-------    1 root     root           550 Dec  1 19:16 homeassistant.json
-rw-------    1 root     root            34 Dec  1 19:17 ingress.json
-rw-------    1 root     root            73 Dec  1 19:15 multicast.json
-rw-------    1 root     root           214 Dec  1 19:15 observer.json
-rw-------    1 root     root           265 Dec  1 19:15 services.json
-rw-------    1 root     root           607 Dec  1 19:16 updater.json
```

Should I still `sgdisk --zap-all /dev/sda` it or is there another path?

agners, 2020-12-02:
There should be single partition with all data on it. Is the old data partition still there?

UnderpantsGnome, 2020-12-02:
here is the fdisk info.

```
# fdisk /dev/sda

Welcome to fdisk (util-linux 2.35.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/sda: 4.56 TiB, 5000981077504 bytes, 9767541167 sectors
Disk model: Expansion Desk
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: gpt
Disk identifier: 48B0E587-59ED-4D30-BA58-4B1AA8B7EEF1

Device     Start        End    Sectors  Size Type
/dev/sda1   2048 9767541133 9767539086  4.6T Linux filesystem

Command (m for help): i
Selected partition 1
         Device: /dev/sda1
          Start: 2048
            End: 9767541133
        Sectors: 9767539086
           Size: 4.6T
           Type: Linux filesystem
      Type-UUID: 0FC63DAF-8483-4772-8E79-3D69D8477DE4
           UUID: A52A4597-FA3A-4851-AEFD-2FBE9F849079
           Name: hassos-data
```

agners, 2020-12-02:
Yeah so `/dev/mmcblk0p8` is still there. It should be safe to remove the partition table and partition on the external drive using `sgdisk --zap-all /dev/sda` then.

While in theory the current script should work, it will take quite a while since your source partition is quite large. I am working on a rewrite of datactl which should be faster, see https://github.com/home-assistant/operating-system/pull/1046. It will be part of the next pre-release, so if you can wait couple of days it might be more efficient.

UnderpantsGnome, 2020-12-02:
Thanks @agners I'll hold off, I can live without the Plex stuff for a while 😉

UnderpantsGnome, 2020-12-21:
Hey @agners just checking in to see if any progress has been made on the `datactl` flow.

UnderpantsGnome, 2020-12-23:
I have a followup on this. I updated the OS to 5.9, reran the datactl move. After the reboot I can see that is it using the external drive, the new partition was created to fill the drive, but the file system is only the size of the SD card.

```
# datactl move /dev/sda
WARNING: All partitions on /dev/sda will be deleted!
Enter "yes" to confirm: yes
Checking that no-one is using this disk right now ... OK

Disk /dev/sda: 4.55 TiB, 5000981077504 bytes, 9767541167 sectors
Disk model: Expansion Desk
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: gpt
Disk identifier: 48B0E587-59ED-4D30-BA58-4B1AA8B7EEF1

Old situation:

>>> Script header accepted.
>>> Created a new GPT disklabel (GUID: BB9860C9-8D12-FD4E-9DCC-14B4483930DB).
/dev/sda1: Created a new partition 1 of type 'Linux filesystem' and of size 4.5 TiB.
Partition #1 contains a ext4 signature.
The signature will be removed by a write command.
/dev/sda2: Done.

New situation:
Disklabel type: gpt
Disk identifier: BB9860C9-8D12-FD4E-9DCC-14B4483930DB

Device     Start        End    Sectors  Size Type
/dev/sda1   2048 9767541133 9767539086  4.5T Linux filesystem

Filesystem/RAID signature on partition 1 will be wiped.

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.

Disk /dev/sda has been prepared to be used as data drive and the data
move has been scheduled for the next reboot. Please reboot the device now and
make sure to leave the disk connected to the system from now on.
``` 
```
# fdisk /dev/sda

Command (m for help): p
Disk /dev/sda: 4.55 TiB, 5000981077504 bytes, 9767541167 sectors
Disk model: Expansion Desk
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: gpt
Disk identifier: BB9860C9-8D12-FD4E-9DCC-14B4483930DB

Device     Start        End    Sectors  Size Type
/dev/sda1   2048 9767541133 9767539086  4.5T Linux filesystem
```
```
# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0   4.5T  0 disk
`-sda1        8:1    0   4.5T  0 part /mnt/data
mmcblk0     179:0    0  59.5G  0 disk
|-mmcblk0p1 179:1    0    32M  0 part /mnt/boot
|-mmcblk0p2 179:2    0    24M  0 part
|-mmcblk0p3 179:3    0   256M  0 part /
|-mmcblk0p4 179:4    0    24M  0 part
|-mmcblk0p5 179:5    0   256M  0 part
|-mmcblk0p6 179:6    0     8M  0 part
|-mmcblk0p7 179:7    0    96M  0 part /mnt/overlay
`-mmcblk0p8 179:8    0  58.8G  0 part
zram0       254:0    0 949.1M  0 disk [SWAP]
zram1       254:1    0    32M  0 disk /var
zram2       254:2    0    16M  0 disk /tmp
```
```
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               100.5M    100.5M         0 100% /
devtmpfs                  1.8G         0      1.8G   0% /dev
tmpfs                     1.9G         0      1.9G   0% /dev/shm
tmpfs                   759.3M      1.2M    758.1M   0% /run
tmpfs                     4.0M         0      4.0M   0% /sys/fs/cgroup
tmpfs                   759.3M      1.2M    758.1M   0% /etc/machine-id
/dev/mmcblk0p1           31.9M      3.4M     28.6M  11% /mnt/boot
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /mnt/overlay
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/docker
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/dropbear
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/modprobe.d
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/modules-load.d
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/udev/rules.d
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /root/.docker
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /root/.ssh
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/NetworkManager/system-connections
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/hostname
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/hosts
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /etc/systemd/timesyncd.conf
/dev/zram2               15.5M     56.0K     14.3M   0% /tmp
/dev/zram1               31.0M    132.0K     28.6M   0% /var
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /var/lib/NetworkManager
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /var/lib/bluetooth
/dev/mmcblk0p7           89.0M      1.7M     80.5M   2% /var/lib/systemd
/dev/sda1                57.8G      7.0G     47.9G  13% /mnt/data
/dev/sda1                57.8G      7.0G     47.9G  13% /var/lib/docker
/dev/sda1                57.8G      7.0G     47.9G  13% /var/log/journal
```

agners, 2020-12-30:
Sorry for the delay, and thanks for the detailed reports! The file system should grow with the next release (fixed with #1133). You can grow the file system manually using `/Iib/systemd/systemd-growfs /mnt/data`.

closed 2020-12-30 by agners