home-assistant/operating-system #1043: 5.6 won't boot on Tinkerboard S using the stock image

eltron247, 2020-12-02:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [X] ASUS Tinker S
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 5.6
- [ ] Updated from version x.y

**Description of problem:**
Im attempting a fresh install of a default 5.6 on the tinkerboard S by flashing the emmc through balena. It flashes fine but upon trying to boot, nothing. The red LED stays on so I believe its still a Uboot issue but I can't be sure. Both ethernet led's blink at random as well but the board never establishes a link to my DHCP server or respond to ARP requests.
I've had 5.3 working for months so the hardware is fine. I'm able to flip back and forth between 5.3 and 5.6 with the same issue each time on 5.6 not booting at all and 5.3 working fine. I've also tried plugging in a monitor locally to diagnose but that wont connect either. I'm flashing 4.18 now to rule out where the issue was introduced.

agners, 2020-12-02:
Tinker Board S support is only working in dev (so rel-5), see https://github.com/home-assistant/operating-system/pull/845.

There is a boot script issue in 5.6, it will be fixed in the next release. Thanks for reporting!

closed 2020-12-02 by agners

eltron247, 2020-12-02:
I'll keep an eye out for it. Thanks!