home-assistant/operating-system #1045: Has os 5.6 only only boots by unplug and plugin SSD

snippem, 2020-12-02:
- [ x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ x] Fresh installation of release 5.6
- [ ] Updated from version 5.5
- Additional information (if accessible):

snippem, 2020-12-02:
Hi all.

Yesterday i updated from OS 5.5 to OS 5.6 with all the other versions never had the problem that my ssd won't boot at powering up.
Reinstalled a clean version off hassos 5.6 does not solve this issue.
So i presume something changed with OS 5.6 that prevents the pi4 from booting up. Changing the sequence in booting (first power on the pi then plug in the ssd ) lets HA boot but that is not the nicest way to boot haha.
I have 2 exactly the same RPI4 4GB with a kingston 128GB usb ssd so i can test whats working or not.
From what i have read here https://github.com/home-assistant/operating-system/pull/939
I have to wait for a new firmware version for the raspberry pi4 that is now still in beta? or is there something else i can do?
It's not the end of the world i know but just interested if the future updates wil solve this problem.

agners, 2020-12-02:
#939 is part of OS 5.6, so you should get the new firmware with 5.6. Can you check which firmware your RPi4 is using now? (Should show up on the initial screen on HDMI, the screen should stay there when you unplug the SSD/SD card).

This firmware is supposed to improve the situation yes, but since you are saying that 5.5 and before did not show the problem it seems the opposite is the case here?

@tmm1 have you noticed something like that with the 2020-10-28 firmware?

snippem, 2020-12-02:
> #939 is part of OS 5.6, so you should get the new firmware with 5.6. Can you check which firmware your RPi4 is using now? (Should show up on the initial screen on HDMI, the screen should stay there [...]

Will check this later what firmware the raspberry with hassos 5.6 is on as i am working atm.
Butt initially the rpi4's were flashed the first official firmware with support for ssd.

snippem, 2020-12-03:
[image]

Here is what i got when booting up replugging the SSD with power on makes it boot.

tmm1, 2020-12-03:
Can you try to plug into the blue USB3 port?

snippem, 2020-12-03:
> Can you try to plug into the blue USB3 port?

Always used the usb 3 ports never had any this before.
Also I tried it on 2 different pi's with the same result.

tmm1, 2020-12-03:
Is this problem only for soft reboot or also hard reboot?

I think you will need to file an issue with the rpi team with details about this SSD model and adapter. See for example: https://github.com/raspberrypi/rpi-eeprom/issues/241

snippem, 2020-12-03:
> Is this problem only for soft reboot or also hard reboot?
> 
> I think you will need to file an issue with the rpi team with details about this SSD model and adapter. See for example: https://github.com/raspberrypi/rpi-eeprom/issues/241
This issue  i have is the first time till 5.5 everything worked nothing changed in the hardware.
Also defective hardware is ruled out because I have 2 setups that are copy of eachother.
Both give the exact same result I know this could be due to the bootloader.
On the other hand it worked before so what breaked the setup the bootloader or the new hassos 5.6

tmm1, 2020-12-03:
Yes I understand. There must be some bug in the new eeprom bootloader. If you provide them with the proper information for debugging, hopefully it can be fixed by them.

What model SSD are you using? Is it in an enclosure? What brand/model enclosure? The USB chipset on that enclosure is likely not interacting well with the EEPROM bootloader USB driver.

snippem, 2020-12-04:
> Yes I understand. There must be some bug in the new eeprom bootloader. If you provide them with the proper information for debugging, hopefully it can be fixed by them.
> 
> What model SSD are you using? Is it in an enclosure? What brand/model enclosure? The USB chipset on that enclosure is likely not interacting well with the EEPROM bootloader USB driver.
> 
> 
> 
> > On the other hand it worked before so what breaked the setup the bootloader or the new hassos 5.6
> 
> Now that your eeprom is already updated to Oct 2020 build, you can try to boot a hassos 5.5 install and see if it works. I expect it will fail in the same way.

Tried it and indeed 5.5 with the latest firmware also failed.
The drive I have is a kingspec Z3 128GB (Ali express 😎)
Had another SSD this time a verbatim vx500 120GB drive and that booted 5.6.
If it ain't broken don't try to fix it hope that raspberry will fix the issue. For now it's not worth the hassle as home assistant controls my whole home and rebooting and then tell my wife you have to replug the drive ain't going to happen 😂

carras, 2020-12-05:
I have an Orico with JMS580 chip from Jmicron. It was booting up for 5.4 and 5.5 And now is not booting up at all in any version of HassOS. Do you know if there is anything I can try to make it work again?

tmm1, 2020-12-05:
@carras What does it show on screen? Is it the same as above?

carras, 2020-12-06:
> @carras What does it show on screen? Is it the same as above?

[image]

I don't think it's the same. And another SSD enclosure works.

tmm1, 2020-12-06:
Can you folks share the output of `lsusb -vvv` from regular Raspbian with the drive attached

carras, 2020-12-07:
> lsusb -vvv

Bus 002 Device 002: ID 152d:0580 JMicron Technology Corp. / JMicron USA Technology Corp. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp.
  idProduct          0x0580 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xc0
      Self Powered
    MaxPower                8mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              6 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15

BlackTurtle123, 2020-12-09:
> Hi all.
>
> Yesterday i updated from OS 5.5 to OS 5.6 with all the other versions never had the problem that my ssd won't boot at powering up.
> Reinstalled a clean version off hassos 5.6 does not solve [...]

Had the same after updating.
Flashed the ssd with new image from 5.6 and could boot again.

carras, 2020-12-10:
Do we know what cause this issue? Is it U-boot issue?

PS that trick to start the pie and then insert the usb drive works for me. Hope we get a better solution.

carras, 2020-12-11:
I just update to HassOS 5.7, then a tried from a fresh image; but none of those works and the issue persists.

Only by turn the RP4 on, then plugin the USB drive, I can make it boot.

carras, 2020-12-11:
@agners In #922 mentioned:

"this is due to the UAS incompatibility with Raspberry Pi. You have to add a usb-storage.quirks for your device. See: https://www.raspberrypi.org/forums/viewtopic.php?t=245931"

Is this related to this issue?

tmm1, 2020-12-11:
Does it work on the black USB2 port?

The bug is in the rpi eeprom and needs to be reported to the rpi team for investigation.

carras, 2020-12-11:
> Does it work on the black USB2 port?
> 
> The bug is in the rpi eeprom and needs to be reported to the rpi team for investigation.

No, it doesn't work on USB2, it's the same.

carras, 2020-12-11:
https://github.com/raspberrypi/rpi-eeprom/issues/262#issuecomment-743356585

I've try the EEPROM on the beta channel from dec11 and it doesn't work either.

Now, I'm trying to go back with EEPROM version because It used to work, maybe going back a few version solve the problem until a better solution.

Holland1, 2020-12-14:
I have the same issue using HassOS 5.8. Plugging in power with the SSD, it won't boot. But booting without SSD and plugging the SSD in a few seconds later, works.

tmm1, 2020-12-14:
Which SSD and enclosure model number?

Holland1, 2020-12-14:
@tmm1 

SSD: Kingston SA2000M8/250G A2000 NVMe PCIe M.2 2280 250GB SSD 
Enclosure: ORICO USB 3.1 Type-C (10Gbps) M.2 NVMe SSD-enclosure Caddy PCIe M2 Adapter

carras, 2020-12-14:
> @tmm1
> 
> SSD: Kingston SA2000M8/250G A2000 NVMe PCIe M.2 2280 250GB SSD
> Enclosure: ORICO USB 3.1 Type-C (10Gbps) M.2 NVMe SSD-enclosure Caddy PCIe M2 Adapter

I think that enclosure have either the JMS583 or JMS580 controller from Jmicron. I have the JMS580 also in a ORICO with the same issue.

Holland1, 2020-12-14:
@carras 

It's the JMS583 chip

Holland1, 2020-12-14:
Some additional info.

While running 5.4 the system would boot just fine without the workaround.
After the release of the 5.8 stable, I installed 5.8 over 5.4 dev. 
The issue started after moving to 5.8.
I flashed the drive with 5.4 dev again, but this time 5.4 dev also refused to boot
So apparently the boot regression issue for the JMS583 controller was introduced in one of the firmware updates between 5.4 and 5.8.

carras, 2020-12-14:
> While running 5.4 the system would boot just fine without the workaround.
> After the release of the 5.8 stable, I installed 5.8 over 5.4 dev.

The 5.4 and older has the cool feature of update your Raspberry Pi 4 EEPROM to an older version that is not working for your SSD enclosure. So we you use another HassOS image the Raspi firmware is already updated.

We should open a issue here: https://github.com/raspberrypi/rpi-eeprom

There we need to find compatibility with our SSD enclosures.

carras, 2020-12-14:
I've open a issue about those controller, please join me: https://github.com/raspberrypi/rpi-eeprom/issues/266

agners, 2020-12-15:
We have now several reports where the 2020-10-28 firmware actually caused issues with USB SSD drives. I am so far not aware of any USB SSD drive which works on 2020-10-28 but did not work with 2020-09-03. From what I know this will make some drives not work with soft reboot (see https://github.com/raspberrypi/rpi-eeprom/issues/241), but it seems to mostly affect spinning disks.

I am inclined to move back to 2020-09-03 for the next stable build.

@tmm1 thoughts?

@carras thanks for pushing this on RPi side! It is important to get it resolved in the firmware so we can move to a future firmware once fixed.

p4p3r, 2020-12-15:
Any report of this issue from non-JMicron users?
I'm running 5.2 on a RPi4 4Gb booting from SSD. I'll try updating to 5.8 tomorrow.
The current enclosure has a JM579 chip, but I also have enclosures with JM578, as well as VIA and ASM chips.
I'll try them all and report back.

tmm1, 2020-12-15:
>I am inclined to move back to 2020-09-03 for the next stable build.

👍 I think this makes sense until RPi team can figure out what's going on with these enclosures

agners, 2020-12-15:
I see that they also released a new stable (2020-12-14).

@carras had you a chance to test with that firmware? You can update the firmware using a SD card with Raspberry Pi OS. Then make sure to delete `pieeprom.sig` and `pieeprom.upd` from the first (boot) partition of the SSD to avoid a downgrade.

Dreamoffice, 2020-12-16:
I have exactly the same problem. I will do this eeprom update now and hope to get it back running with the SSD

carras, 2020-12-16:
> Then make sure to delete `pieeprom.sig` and `pieeprom.upd` from the first (boot) partition of the SSD to avoid a downgrade.

@agners I've try "stable" and "beta" but without doing this. I guess that avoid the auto-update of the Raspberry Pi firmware from the HassOS.

Can you tell me the exact paths to those files? I'm unable to find them.

carras, 2020-12-16:
Can you guys try the solution I found in: https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-746890101

**Solution**: with the command "sudo -E rpi-eeprom-config --edit" you edit the EEPROM BOOTLOADER then you the next configuration:
USB_MSD_PWR_OFF_TIME=0

@agners Is in necessary to have that USB POWER CUT by default on HassOS?

Dreamoffice, 2020-12-16:
> Can you guys try the solution I found in: [raspberrypi/rpi-eeprom#266 (comment)](https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-746890101)
> 
> **Solution**: with the command "sudo -E rpi-eeprom-config --edit" you edit the EEPROM BOOTLOADER then you the next configuration:
> USB_MSD_PWR_OFF_TIME=0
> 
> @agners Is in necessary to have that USB POWER CUT by default on HassOS?

it works for me

THANKS A LOT

agners, 2020-12-16:
@carras thanks for the update. Does `USB_MSD_PWR_OFF_TIME=1000` work for you? Since that was the default in the 2020-09-03 firmware...

My fear is when I set it to 0, then 10 other USB SSD stop working :see_no_evil:

carras, 2020-12-16:
For those using HassOS, I've only seen this solution to work on this EEPROM release:

Dec 11 2020 11:15:17
version c3f26b6070054bca030366de2550d79ddae1207a (release)
timestamp 1607685317
update-time 1608150584
capabilities 0x0000001f

*I've tried with a fresh HassOS 5.8 and the raspberry in Sep 03 firmware and it didn't work. No update from HassOS or boot. Then I use the "sudo raspi-config" command to move the firmware to "latest" branch then I used the solution; so It worked.

carras, 2020-12-16:
> @carras thanks for the update. Does `USB_MSD_PWR_OFF_TIME=1000` work for you? Since that was the default in the 2020-09-03 firmware...
> 
> My fear is when I set it to 0, then 10 other USB SSD stop working 🙈

@agners USB_MSD_PWR_OFF_TIME=1000 is default, so I've been testing that for a long time lol.

Apparently you should stick with that default because other more popular enclosure or HDs are working better. @timg236 said it here: https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-746913626

The problem I see is, what happen when you update the firmware in the HassOS? Is that config going to be remove so I have to put the SD card again and add the "USB_MSD_PWR_OFF_TIME=0" again?

agners, 2020-12-16:
@carras well, RPi foundation claims its the default since 2020-09-03 which used to work in your case right? Soooo I wonder if they adjusted that default in the beta without telling... Easy to test: Just set it to 1000 explicitly.

Because otherwise I wonder why the default of 1000 works with 2020-09-03, but doesnt't with newer firmwares...

vodoc, 2020-12-16:
Trick with USB_MSD_PWR_OFF_TIME=0 works for me. Thank you. But only with 64bit version. I have no luck with booting 32bit any 5.x version. I have 2G board version.

tmm1, 2020-12-16:
> Because otherwise I wonder why the default of 1000 works with 2020-09-03, but doesn't with newer firmwares...

In the 2020-10-28 firmware there were a lot of changes to the USB boot operation, so the effect of those variables is not the same before and after

https://github.com/raspberrypi/rpi-eeprom/pull/246

>Don't timeout a USB MSD device after USB_MSD_LUN_TIMEOUT if there are no other MSD devices or LUNs to tries. This avoids unnecessary timeouts on very slow to initialise disk drives e.g. USB HDDs designed for backups.

tmm1, 2020-12-16:
> My fear is when I set it to 0, then 10 other USB SSD stop working 🙈

I'm guessing this is the case. The 0 option will work for some drives but not others.

For example with the 2020-09-03 eeprom, it was necessary for me to use `USB_MSD_PWR_OFF_TIME=1500` to make several types of USB HDD boot up.

I think the best solution will be when the Pi team receives the problematic hardware and can develop a proper fix.

To speed up this process, someone could enable NETCONSOLE in the eeprom and capture the logs required via wireshark. But its a complicated process, so it might be easier to wait for the Pi team to do it.

carras, 2020-12-16:
@agners Nor 1500, 1000 or 500 work. And there is a consistent message in the failed login about a 1000ms usb reset.

The auto-update of the firmware is a great feature that works. The problem is that some firmwares are compatible with some enclosures. There will be a better firmware I guess.

But can it be an opt-out? does FREEZE_VERSION=1 works? What about the config file after an EEPROM update?

tmm1, 2020-12-16:
> But can it be an opt-out? does FREEZE_VERSION=1 works? What about the config file after an EEPROM update?

Yes FREEZE_VERSION will work. There is no config file. The config values are stored inside the eeprom image itself.

carras, 2020-12-16:
@tmm1 when there is an update the configuration will stay the same? Or do I need to update it everytime?

tmm1, 2020-12-16:
If you set FREEZE_VERSION then the eeprom is frozen and will not self-update.

Teleportist, 2020-12-16:
Raspberry 4, 4GB

Upgrading from HassOS 5.3 to 5.8 (64-bit)

This has been rock solid for well over a month. Now after a shutdown or power cycle booting HassOS on a SSD (Kingspec Z3 SSD) from USB 3 now fails. The drive access light illuminates for a second, once, then stays off. This happened straight after the upgrade to 5.8.

Booting will resume if the drive is unplugged from the USB port, then reinserted.

carras, 2020-12-16:
@tmm1 yes, I know that. But without using the Freezing, will I lost the configuration in the next update?

tmm1, 2020-12-17:
Depends if the next HassOS release has eeprom update or not. We may be removing it for now because of the issues.

carras, 2020-12-17:
> Can you guys try the solution I found in: https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-746890101
>
>
>
> **Solution**: with the command "sudo -E rpi-eeprom-config --edit" you edit [...]

@Teleportist this is the solution. Your enclosure use jmicron controller.

carras, 2020-12-17:
@snippem can you edit the first to make the solution more clear?

carras, 2020-12-17:
> Depends if the next HassOS release has eeprom update or not. We may be removing it for now because of the issues.

@tmm1 But if it does come with the update, will that modify my configuration "USB_MSD_PWR_OFF_TIME=0"?

tmm1, 2020-12-17:
Yes.

carras, 2020-12-17:
@agners Here you have the proper release notes: https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md

You can see there they back ported that USB power off time to the sep 03 release later on.

p4p3r, 2020-12-17:
Ok I'm not sure how useful this is at this point, but here's what happened with the 4 enclosures I have.

Tested using a RPi4 1Gb, fresh OS 5.8 64bit install
Working:
- Inateck FEU3NS-1E, reported as ASM1153E VendorID: 0x174c Product ID: 0x55aa
- Inateck FE2005, advertised as using JMicron JMS567, reported as JMicron Vendor ID: 0x152d Product ID: 0x0578
**Not working:**
- Inateck FE2010, advertised as using JMicron JMS578, reported as Toshiba VendorID: 0x0080 ProductID: 0xa001

Tested using a RPi4 4Gb, OS update from 5.2 64bit to 5.8
Working:
- POSUGEAR HD-SATA, reported as JMicron JMS579 VendorID 0x152d ProductID 0x0578

The interesting thing about the one that I found not working is that the "power led" on the enclosure is lit but just barely while the boot process is stuck and it become really bright after the unplug/plug that also allow the boot process to continue. Which makes perfect sense given it's now clear that the firmware is keeping the port powered off...

boesing, 2020-12-17:
@agners Thanks for the advice in #1095 
That worked for me.

## For others who stumble over this

I've used SD card with Raspberry Pi OS, updated anything in there `apt-get update && apt-get full-upgrade` and downgraded my Pi 4 to that version I was on before upgrading Hass OS.
I've installed my SSD on November 21st and thus I took the commit from November 19th.

A list of available commits can be found [here](https://github.com/raspberrypi/firmware/commits/master)

So I executed `raspi-update <commit>` to downgrade.

After rebooting, I added some suggestion from raspberry/raspi-eeprom#266 regarding the `USB_MSD_PWR_OFF_TIME` and fixed the boot order as somehow that changed aswell.

Verify `vcgencmd bootloader_config`, it should contain `BOOT_ORDER=0xf41`. If this looks different, you might want to change that using `raspi-config` (Advanced -> Boot Order).

Finally, I've plugged the SSD into my macbook, mounted the FAT32 Partition (Partition 1) and removed `pieeprom.sig` and 
`pieeprom.upd`.

Finally, unmount the SSD, shutdown raspberry pi, remove SD-Card, plug in SSD and my homeassistant started up again without having to wipe my SSD.

Thanks for the amazing guidance @agners!

Teleportist, 2020-12-18:
> > Can you guys try the solution I found in: [raspberrypi/rpi-eeprom#266 (comment)](https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-746890101)
> > **Solution**: with the command "sudo -E rpi-eeprom-config --edit" you edit the EEPROM BOOTLOADER then you the next configuration:
> > USB_MSD_PWR_OFF_TIME=0
> > @agners Is in necessary to have that USB POWER CUT by default on HassOS?
> 
> @Teleportist this is the solution. Your enclosure use jmicron controller.

Thank you Carras. I will try this, but need to wait for a new SD card to be delivered, as I don't have a spare card, to install Rasp Pi OS on, right now.

Unless the bootloader can be edited via SSH whilst still running HassOS?

[18 more replies]

closed 2020-12-19 by agners