home-assistant/operating-system #1047: HASSOS 5.6 - mDNS (.local) broken

dsegantx, 2020-12-02:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

I've been running 5.x for a month or so and upgraded through several point releases as well has having run 4.x for months.  Always been able to find my instance via browsers with homeassistant.local.  My environment is primarily Apple.

I have other devices that present/are found as xxxxx.local.

Running on Rpi 4 64 bit. 

Upgraded to 5.6 and lost homeassistant.local.  Was able to access via ip address within my home network.  Tried flushing cache on computers and rebooting router but no luck.  Saw same problem on iPhone.  Yet all devices could see my other devices.local - just not homeassistant.local.

Backed down to 5.5 and problem went away.

Something in 5.6 changed...

agners, 2020-12-02:
With 5.6 we use systemd-resolved. Home Assistant OS supplies a default which enables mdns, but if you are upgrading from a previous version, this isn't taken into account. The problem will be fixed with the next release where we set the default in NetworkManager's configuration file: https://github.com/home-assistant/operating-system/commit/c48184cf918f04cbfe08672cf35c40cc71751cea

For now you can get mDNS going again by manually add `mdns` to the default network configuration (as done in the OS default https://github.com/home-assistant/operating-system/commit/25a0dd30823e5863b4d90f2556d05149099be864#diff-5d157295d65d48b5b9ad0494c7721e202aab74d7fff65dd084d26c1a32155696).

dsegantx, 2020-12-02:
> With 5.6 we use systemd-resolved. Home Assistant OS supplies a default which enables mdns, but if you are upgrading from a previous version, this isn't taken into account. The problem will be fixed with the next release where we set the default in NetworkManager's configuration file: c48184c
> For now you can get mDNS going again by manually add mdns to the default network configuration (as done in the OS default 25a0dd3#diff-5d157295d65d48b5b9ad0494c7721e202aab74d7fff65dd084d26c1a32155696).

Thanks for the clarification.  Should I close this issue or do you need it to cover the fix?

agners, 2020-12-02:
Since the fix is already committed the issue can be closed.

closed 2020-12-02 by agners

dsegantx, 2020-12-11:
FWIW- I remained on 5.5 and once 5.7 was available I tested as follows:

Upgrade 5.5 to 5.6.  Problem still exist (expected)
Upgrade 5.6 to 5.7.   Problem eliminated and able to use *.local domain name.

Backed down to 5.5 and then upgraded directly to 5.7 - Problem eliminated and able to use *.local domain

So good work - 1041 solved it.  Thanks.