home-assistant/operating-system #1050: Hassos ova for arm cpu ESXi Arm Edition?

shirou93, 2020-12-02:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Why not release hassos ova for arm? Esxi released ESXi Arm Edition and RPI4 8gb make sense.

Is it possible convert rpi.img to working vmdk?

agners, 2020-12-03:
@shirou93 patch welcome :-D Yeah you can convert the disk image to a vmdk and try create a machine which boots. The utility `qemu-img` can do that.

ipha, 2020-12-07:
I tried building a generic arm vm image, but I couldn't get a bootloader working. Best I can tell barebox doesn't support arm EFI.

GRUB could be an option, but I don't know if it supports the A/B partitions.

shirou93, 2020-12-11:
@ipha @agners  i try but not booting. Maybe bootloader was broken when convert img to vmdk.

agners, 2020-12-12:
Hm, yeah bootloader is a problem: We use U-Boot in all Arm builds. It seems that ESXi Arm is expecting a UEFI bootloader such as Grub2. Gleening at this blog post it seems that booting something else is possible in theory, but it seems that requires some hacking: https://blogs.vmware.com/arm/2020/10/23/uefi-less-vms-with-esxi-arm-fling-1-1/

myericbaba, 2020-12-21:
anyone try to follow this guideline for HA?
https://www.virtuallyghetto.com/2020/10/how-to-run-raspberry-pi-os-as-a-vm-on-esxi-arm.html

NeverEatYellowSwissSnow, 2021-12-11:
I am interested in this too. Would be so nice to run HA OS on ESXi ARM!

agners, 2022-02-23:
As noted in #1549, there are now images named `generic-aarch64` which support UEFI boot on arm64. I haven't tested them on ESXi for ARM, I am keen to here if the image also works on ESXi.

There are development builds available at https://os-builds.home-assistant.io/8.0.dev20220223/.

Grandma-Betty, 2022-02-23:
> As noted in #1549, there are now images named `generic-aarch64` which support UEFI boot on arm64. I haven't tested them on ESXi for ARM, I am keen to here if the image also works on ESXi.
> 
> There are development builds available at https://os-builds.home-assistant.io/8.0.dev20220223/.

Just tried it with the same settings as HAOS runs on ESXi-ARM, without any luck.
I have used the **.vmdk** disk image as well as the **.img** file from source.
I also tried to play around with hard disk controller settings too, no luck.
It's not booting.

agners, 2022-02-23:
@Grandma-Betty you did use the `generic-aarch64` images right? Did anything come up at all (Grub boot loader maybe?)

Grandma-Betty, 2022-02-23:
Yes, I used the said images.
No, it was not even reaching a boot loader. With the **.vmdk** image I was not even able to start the VM itself from the Hypervisor.
With the **.img** file the VM did of course start but was waiting forever for PXE boot.

agners, 2022-02-23:
Can you configure the UEFI BIOS in ESXi somehow? Maybe secure boot or something?

tweetysat, 2022-03-03:
Also tired with both (latest) generic vmdk and ova vmdk.
After adding the disk, changed sata to ide and verified that the boot is EFI, the vm is not starting.
It tells me : Failed - Disconnected from virtual machine.
And :

> Power On VM
> Key
> haTask-2-vim.VirtualMachine.powerOn-4294853845
> 
> Description
> Power On this virtual machine
> 
> Virtual machine
> [HomeAssistant OS](https://192.168.0.200/ui/#/host/vms/2)
> State
> Failed - Disconnected from virtual machine.
> 
> Errors
> Disconnected from virtual machine.
> Failed to establish transport connection (9): There is no VMware process running for config file /vmfs/volumes/5f4e56d3-d3071bfa-bf11-dca632e04cec/HomeAssistant OS/HomeAssistant OS.vmx.
> Remote disconnected

Grandma-Betty, 2022-05-06:
> Hm, yeah bootloader is a problem: We use U-Boot in all Arm builds. It seems that ESXi Arm is expecting a UEFI bootloader such as Grub2. Gleening at this blog post it seems that booting something [...]

Any news on this? Seems pretty interesting!

agners, 2022-05-07:
No updates, I don't plan to look into ESXi Arm myself.

Home Assistant OS uses regular GRUB2 built for aarch64. The bootloader is present on the ESP (EFI System Partition) as it is common in UEFI boot flows. ESXi really should be able to pick up that GRUB2 and start booting, it's unclear to me why it can't.

Unfortunately I haven't seen a useful log so far. If that is really all ESXi is logging, then its pretty useless to be honest :cry: 

I recommend using KVM, as it works and offers also logging which helps us developer to debug the problem.

Grandma-Betty, 2022-05-12:
@agners Well, I was able to get the time now to try again. I located more logfiles which I attached. Maybe this gives you any hints in any way?
I was trying to boot with the HAOS _.img_ file btw. (latest version from [here](https://os-builds.home-assistant.io/8.0.dev20220510/))
Thank you!
[vmware.log](https://github.com/home-assistant/operating-system/files/8678732/vmware.log)
[vmware-1.log](https://github.com/home-assistant/operating-system/files/8678733/vmware-1.log)
[vmware-2.log](https://github.com/home-assistant/operating-system/files/8678734/vmware-2.log)
[vmware-3.log](https://github.com/home-assistant/operating-system/files/8678735/vmware-3.log)

Grandma-Betty, 2022-05-12:
Yes, those are multiple boots. I tried to boot with IDE controller first, but the VM does not even start with IDE. Seems like the "Other 4.x or later Linux (64-bit) template does not support IDE controllers:

[image]

At last I tried SATA controller and then the VM started but never reached the GRUB2 bootloader.
The latest logs should be with the SATA controller setting.

agners, 2022-05-12:
The last log (vmware-3.log) does has that vdmk file still, I am not sure what that is/if that interferes maybe:

```
...
ide0:0.fileName = "HAOS_ARM_test.vmdk"
...
```

Is `HAOS_ARM_test` the name of the VM? Is that something automatically generated?

Grandma-Betty, 2022-05-12:
`HAOS_ARM_test` is the name I have given manually to the VM.
Here are the new logs after creating the VM and only booting once:
[vmware.log](https://github.com/home-assistant/operating-system/files/8679047/vmware.log)

agners, 2022-05-12:
It seems the system doesn't find anything bootable. However, there is clearly a EFI System Partition, a long with the proper boot files.

```
2022-05-12T12:33:33.598Z| vcpu-0| I125: [msg.Backdoor.OsNotFound] No operating system was found. If you have an operating system installation disc, you can insert the disc into the system's CD-ROM drive and restart the virtual machine.
```

There could be two reasons
a) The VM is trying to boot some other method (not UEFI)
b) It is trying to boot via UEFI, but somehow does not accept the default location `/EFI/BOOT/BOOTAA64.EFI`. It migth be related to #1760. If that is the case, booting a Live Ubuntu CD-ROM (for Aarch64, not sure if that exists) and try to write to use the work around suggested in https://github.com/home-assistant/operating-system/issues/1760#issuecomment-1107654728 might help (replace `bootx64.efi` with `bootaa64.efi`).

Do you see something on the screen, like a BIOS or something?

Grandma-Betty, 2022-05-12:
Yes, after a while not booting the system goes automatically into BIOS.
But I can force it too to enter the BIOS:
[image]

agners, 2022-05-12:
Yeah from the logs and that screenshot it looks like its an EFI BIOS. So that leaves us with option b): For some reason it does not pick up `/EFI/BOOT/BOOTAA64.EFI`.

As reported by #1760, some x86-64 UEFI BIOS require an entry in EFI variables *for internal disks*. You can try creating such variables using a Live Distribution and efibootmgr as well.

Alternatively, maybe you can make the drive to be considered as "external disk". Can you choose a USB as virtual disk controller? Or maybe attach the vmdk file as a CD-ROM?

claplace, 2022-05-14:
Hi @agners and @Grandma-Betty, I am the ESXi ARM dev :-) so I created a new VM, set the type to Linux/Other4.x, removed the new hard drive, added the existing haos.vmdk, and started the VM. It worked!

Okay there's one thing I've done, it's **importing** the vmdk into ESXi. If you want the vmdk to be on the native ESXi VMFS partition, it needs to be imported to be usable. Here, from a NFS storage:
```
vmkfstools -i /
vmfs/volumes/nfs/haos_generic-aarch64-8.0.dev20220223.vmdk  /vmfs/volumes/datastore1/haos/haos.vmdk
```

If you're using the vmdk directly from NFS this shouldn't be necessary.

claplace, 2022-05-14:
In other tries I noticed that using a NVMe virtual drive doesn't work, linux doesn't find the root partition. But as a SATA drive it's fine.

Grandma-Betty, 2022-05-14:
@claplace Nice! So importing the .vmdk via esxi web interface to esxi datastore will not work?

@agners I know the aarch64 images are still in development state but for testing purposes it would be nice to have my existing HA setup. So is there a reliable process to migrate from x86 to aarch64 version of home assistant?

claplace, 2022-05-14:
@Grandma-Betty I tried by uploading the vmdk file via esxi web interface, and it's only doing that: uploading the file, not importing it as a disk. I get the error "_[Failed to power on virtual machine haos. Unsupported or invalid disk type 2 for 'sata0:0'. Ensure that the disk has been imported. Click here for more details.]_". You still need that vmkfstool step for it to be usable.

iSCSI works, but the block device will be formatted as VMFS, so the vmkfstool step is also required.

agners, 2022-05-16:
> I know the aarch64 images are still in development state but for testing purposes it would be nice to have my existing HA setup. So is there a reliable process to migrate from x86 to aarch64 version of home assistant?

The backup only backups data, and a reference to the Core/Add-ons versions. Restoring a backup on a different architecture should work without further doing.

That said, it could be that some add-ons store "architecture specific" data, but that would be up to that add-on.

Grandma-Betty, 2022-05-16:
@agners Nice, thanks. I just mentioned the new HA release from today which contains some changes about GRUB2. Did the aarch64 image from [here](https://github.com/home-assistant/operating-system/releases/tag/8.0) leave its development state?

Grandma-Betty, 2022-05-16:
Just wanted to report back that I migrated now successfully from HAOS (x86) to HAOS (aarch64) with the native backup feature of HAOS. Some minor manual steps were required to make all add-ons work again but nothing major.
Only thing missing is open-vm-tools included in the aarch64 version. As far as I remember open-vm-tools is part of the native HAOS x86 virtual image but not of the aarch64 .vmdk file.

So far I had nothing that I was not able to migrate. In case anyone else is interested, I'm currently running the following add-ons/integrations which I have all migrated successfully from x86 to aarch64 version of HAOS:

Add-ons:
- Zigbee2MQTT
- Mosquitto
- Samba backup
- Samba share
- NGNIX SSL proxy
- ESPHome
- Frigate NVR
- Terminal & SSH

Integrations:
- EmulatedRoku
- Kodi
- Logitech Harmony Hub
- HACS
- Season

@agners @claplace Many thanks to both of you guys for your investigations, you really made my day!
The issue can be closed now I guess.

agners, 2022-05-16:
> Did the aarch64 image from [here](https://github.com/home-assistant/operating-system/releases/tag/8.0) leave its development state?

Yes, HAOS 8.0 is considered stable, and `generic-aarch64` image is part of it. That said, since there aren't that many users, its certainly not the most well tested image :sweat_smile: 

> Only thing missing is open-vm-tools included in the aarch64 version. As far as I remember open-vm-tools is part of the native HAOS x86 virtual image but not of the aarch64 .vmdk file.

> EDIT: uh, I don't even know if HA is based on debian packaging system...

HAOS is based on Buildroot. Buildroot has [openvmtools available as a package](https://github.com/home-assistant/buildroot/tree/2022.02.x-haos/package/openvmtools), so it should be relatively easy to add.

I'll let you know once I'll create a 9.0 dev build with it included tomorrow. If that works, we can backport it to rel-8 branch and include it in OS 8.1.

Grandma-Betty, 2022-05-17:
> Yes, HAOS 8.0 is considered stable, and `generic-aarch64` image is part of it. That said, since there aren't that many users, its certainly not the most well tested image 😅

No worries. Ganny will be happy to bite the bullet for y'all youngsters 💪

> I'll let you know once I'll create a 9.0 dev build with it included tomorrow. If that works, we can backport it to rel-8 branch and include it in OS 8.1.

Granny is here to test and report back.

Grandma-Betty, 2022-05-19:
@agners Does this image already include open-vm-tools and your latest NVMe driver commits?

https://os-builds.home-assistant.io/9.0.dev20220518/

agners, 2022-05-19:
Hm, open-vm-tools should be part of that build. Can you get OS shell access (using `login`) and check if the udev rules are present?

```
ls -la /usr/lib/udev/rules.d/99-vmware-scsi-udev.rules
```

Grandma-Betty, 2022-05-19:
Nope, there's not even the `/usr/lib/udev` directory available:
[image]

It's completely missing.

agners, 2022-05-19:
Hm, I see, the build system prevents the package from being selected under AArch64 :see_no_evil: should be a easy fix.

Grandma-Betty, 2022-05-19:
Let me know when the next image is ready.

Grandma-Betty, 2022-05-20:
I can confirm open-vm-tools and nvme boot works now with the latest 8.1 update.
Thanks a lot @agners for your support mate!
Bischen geile Siech ;-)

Grandma-Betty, 2022-05-21:
Just a minor thing I found:
Isn't the system supposed to show "Yes" for the "Virtual environment" entry when using a .vmdk as a installation source? I mean, where else would you use a .vmdk file for than within a virtual environment? Or am I misunderstanding the meaning of this?
[image]

mondface, 2022-05-21:
Now I also run home assistant 8.1 with "vmware tools" on my esxi on arm.
Thank you!

[image]

Grandma-Betty, 2022-05-21:
@mondface You're welcome in any way, but don't thank me, thank @agners and @claplace, they're the real heroes here ;-)

claplace, 2022-05-21:
oh I need to upgrade to 8.1 then!

Detecting VMware virtualization is possible 6 looking at the device tree. e.g. I added those lines to systemd: https://github.com/systemd/systemd/blob/df423851fcc05cf02281d11aab6aee7b476c1c3b/src/basic/virt.c#L134

mondface, 2022-05-21:
I have a zwave-me stick, but it doesn't run smoothly. USB I have to assign 3.1 for the vm.
how did you implement that?

[image]

Grandma-Betty, 2022-05-22:
@mondface Well I don't use zwave, but it's kinda common that you have to use a USB 3.1 controller. I was never able for any distribution running on ESXi ARM to make USB 2.0 controler working reliably. But since USB 3.1 controller never gave any issues it's not really a disadvantage to me. I think the USB backwards compatibility is always given so I don't care. You just have to know it, that's all ;-)

I remember once writing an issue about this according to Debian distro which may also be interesting to you:
https://flings.vmware.com/esxi-arm-edition/comments/18497#comment_18505

Grandma-Betty, 2022-05-22:
@mondface @claplace Off-topic but the following issue also might be interesting to you as I am not the only one struggling with NUT on ESXi since months now:
https://github.com/networkupstools/nut/issues/1004
This might be useful to be aware of as you can read there are also users struggling with NUT runnig on Home Assistant running on ESXi. Also one of the latest posts of the user running 29 ESXi servers which are all affected in the same way about the USB issue proofs that there is something wrong.

Grandma-Betty, 2022-05-23:
@agners As the aarch64 image went stable now it would make sense to include it here, wouldn't it?
https://www.home-assistant.io/installation/
I mean the more people are aware of it, the more people will use it, the more it gets the chance to get "matured".
I can only tell it works like a charm in my case and I really did many(!) tests over the weekend with the add-ons/integrations I have mentioned above. Nothing to complain about compared to my previous x86 virtual setup. In fact I am really more happy now with HA running on a Raspberry Pi based ESXi hypervisor.

agners, 2022-05-23:
@Grandma-Betty @claplace the "Virtual Environment" information is from Home Assistant Core, and indicates if the installation makes use of a Python Virtual Environment. So it is unrelated to the virtualization technology we are discussing.

@Grandma-Betty I'd still count aarch64 virtualization quite niche still. If we would add it to the official documentation would add maintenance load on our end. At this point I don't think we would accept it to the official documentation.

I recommend starting with a [Community Guide](https://community.home-assistant.io/c/community-guides/51) for now. A good title and it will be the first Google search result :smile:  Tips like the USB 2.0/3.1 issue etc, can then be collected there. The https://github.com/home-assistant/operating-system/issues/1050#issuecomment-1133594465 is probably a good start already.

I'll close the issue here as there is nothing open on HAOS side at this point. If you create a community guide, feel free to link it here. If more work is to be done on HAOS side, please create a new issue.

closed 2022-05-23 by agners

Grandma-Betty, 2022-05-31:
> I recommend starting with a [Community Guide](https://community.home-assistant.io/c/community-guides/51) for now. 

https://community.home-assistant.io/t/installing-haos-on-vmware-flings-esxi-arm/426604

thiscantbeserious, 2022-12-11:
Just in case: I used the aarch64 successfully including USB-Passtrought with a Zigbee-Stick (HAOS 9.3) on Pimox 7 - with a Raspberry Pi 4/8GB ... works surprisingly smooth. Used the installer from here: 

https://github.com/tteck/Proxmox/blob/main/vm/haos-vm-v4.sh

Simply replaced ova with generic-aarch64 image in it and afterwards coverted the disks to qcow2, also had to remove the EFI boot and readd it.

Works flawlessly - even with just 2 cores assigned

tteck, 2022-12-12:
There's actually a PiMox script to install HAOS at https://tteck.github.io/Proxmox/

(https://github.com/tteck/Proxmox/raw/main/vm/pimox-haos-vm-v4.sh)

thiscantbeserious, 2022-12-14:
> There's actually a PiMox script to install HAOS at https://tteck.github.io/Proxmox/
> 
> (https://github.com/tteck/Proxmox/raw/main/vm/pimox-haos-vm-v4.sh)

Dang, didn't see that, thanks! ... but it works anyway. :) 

> > Might be a better Option then ESiX in terms of performance, specifically on something as limited as ARM SBCs.
> 
> Not sure about that. The ESXi hypervisor does not consume much resources. But indeed it is a very interesting option and I'm very impressed by the progressive process of virtualization generally. Also it's nice that PiMox is open source and it does not have any license limitations as the annoying 180 days ESXi ARM limitation, which you only can get around when assigning a license via vCenter to it.
> 
> Because of license reasons I was considering too about jumping my homelab from vSphere to a Proxmox environment, but the missing ARM part for Proxmox was holding me back currently. Now I can think it over again. So thanks for sharing!

Just be carefull: Any BIG.little architecture will give you troubles - as long as its a Raspberrypi with 4 Cores of the same type it wont have any troubles.

thiscantbeserious, 2022-12-26:
@tteck if you modify your create_lxc.sh script to actually be able to select a locally stored images you can actually use all lxc scripts also with aarch64/arm64 ... currently running a hard coded variant over here to test them ... zigbee2mqtt and homeassistant lxc

tteck, 2022-12-26:
@thiscantbeserious Maybe use `uname -m` to detect the architecture. If aarch64/arm64 is detected, download the appropriate image. 🤷🏻‍♂️