home-assistant/operating-system #1056: VMDK Image not start OS

Saentist, 2020-12-03:
**Hardware Environment**
- [ ] OVA (VMWARE)

**Home Assistant OS release:**
- [ ] Fresh installation of release 4.16
```https://github.com/home-assistant/operating-system/releases/download/4.16/hassos_ova-4.16.vmdk.gz```

**Description of problem:**
Starting stop on this point all time. I'm using VMWare Workstation 16
[image]

thfrlw, 2020-12-04:
I've just run into pretty much the same issue

agners, 2020-12-04:
Hm, it seems the Linux kernel doesn't see the harddrive. What type of disk controller are you emulating in this machine configuration?

Saentist, 2020-12-04:
> Hm, it seems the Linux kernel doesn't see the harddrive. What type of disk controller are you emulating in this machine configuration?

SCSI with is default.
In manual is not specified with must be selected.

```
For Vmware Workstation create a new virtual machine, select “Custom”, 
make it compatible with the default of Workstation and ESX, 
Choose “I will install the operating system later”, 
select “Linux” -> “Other Linux 5.x or later kernel 64-bit”, 
give it at least 2 GB RAM and 1vCPU, 
select “Use Bridged Networking” then “Use an existing virtual disk” and select the VMDK file above, 
after creation of VM go to “Settings” and “Options” then “Advanced” and select “Firmware type” to “UEFI”.```

Saentist, 2020-12-04:
Image is for **SATA**
Manual need update, or better export of virtual mashine in place of hdd image.

agners, 2020-12-04:
@Saentist so it worked with SATA? Yeah I think we are missing a SCSI driver or two which are common in virtualization. But if SATA works fine, no problem using it.

Saentist, 2020-12-05:
Yes work with SATA mode connected disk image.
Manual do not say nothing about disk connecting mode.

jxbrown, 2020-12-17:
I just updated HASSIO from 4.16 to 5.8 on VMWare workstation. The update went fine but after I rebooted I get the following error.

[image]

I'm using a SATA HD configuration in my settings.  Easily reproducible. 

[image]

closed 2021-05-12 by stale[bot]