home-assistant/operating-system #1062: Watchdog service

urosgruber, 2020-12-06:
**Hardware Environment**

- [x] Raspberry Pi [4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

This is more of a question since I can't seem to find any info about this. I'm looking of how to setup watchdog service for HA on RPi4 in case it dies so that it gets restarted automatically. I have connection with uptimerobot but this still means a manual intervention. I've tested watchdog service with regular RPi4 image but I don't know how to deal with this with HA operating system.

Thanks for any input on this.

frenck, 2020-12-06:
It already has a watchdog...

urosgruber, 2020-12-06:
Does it need to be enabled somehow? The thing is my HA crashed couple of times but nothing really happened. I can see `/dev/watchdog` but no configuration.

frenck, 2020-12-06:
The Supervisor plays the watchdog in this case. It should detect it is down and should try to bring it up again.

urosgruber, 2020-12-06:
I checked today and nothing specific was visible on the screen but it was unresponsive. But I forgot to plugin keyboard to check If console works. This is happening randomly and I wasn't able to find the real cause of this. I'm running 5.5 with rpi4 64 bits.

agners, 2020-12-06:
Just to make sure we all talk about the same thing: There are hardware watchdogs and software watchdogs. HW watchdogs are usually exposed to user space via `/dev/watchdog` in Linux. Userspace programs can open the device and then need to periodically answer to it. If not, Linux will reboot the system. Of course this then only helps if the system actually behaves again after reboot...

Today, systemd can use the HW watchdog devices by default to make sure systemd is alive. Systemd in turn can enable a SW watchdog via `WatchdogSec=`. With that services need to periodically checkin on systemd to make sure it is alive. If not, it will restart the service. I don't think we make use of those features at the moment. I need to investigate if/how well this can be tied to Docker.

Currently `hassos-supervisor.service` starts the supervisor, which in turn supervises all add-ons and Home Assistant Core.

A kernel panic btw, is not covered by hardware watchdog per se, but we recently added an option to automatically reboot on kernel panic: https://github.com/home-assistant/operating-system/pull/937.

To the particular problem at hand, I think it is important to understand where the problem actually is. Hardware watchdog doesn't help if the software it checks (e.g. systemd) still works, and the problem is higher up...

> I checked today and nothing specific was visible
So black screen? Or just a login screen indicating a working system?

> But I forgot to plugin keyboard to check If console works.
Yes, screen can go to standby after a while, so make sure to wake it up with a few keystrokes.

Is the system responding on the network (e.g. ping)?

urosgruber, 2020-12-07:
Thanks @agners for such detailed response.

So I was asking abut HW watchdog so that in case HA freezes it will restart the board. I'll do my best to gather as much infor as possible when this repeats but it might take some time. From what I saw it didn't respond on TCP connection. And on the screen there was full of "**brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled**". I doubt this helps anything right now.

closed 2021-05-12 by stale[bot]

jflattery, 2021-12-12:
Someone has built an [add-on](https://community.home-assistant.io/t/home-assistant-add-on-hardware-watchdog-service/365894) for this, but I really think it should be built into the OS.

jflattery, 2021-12-12:
> The Supervisor plays the watchdog in this case. It should detect it is down and should try to bring it up again.

@frenck  And what if the system freezes / locks up which prevents this?

Related: https://diode.io/raspberry%20pi/running-forever-with-the-raspberry-pi-hardware-watchdog-20202/

agners, 2021-12-13:
Have your read https://github.com/home-assistant/operating-system/issues/1062#issuecomment-739565175?

> Related: https://diode.io/raspberry%20pi/running-forever-with-the-raspberry-pi-hardware-watchdog-20202/

This seems rather ad-hoc using shell scripts for that particular use case. What if a watchdog does not support a particular interval? What if the system boots fine but Supervisor doesn't start for some other reason? Lastly, any reason they did not use [the systemd built-in hardware Watchdog support](http://0pointer.de/blog/projects/watchdog.html)?

> Someone has built an add-on for this, but I really think it should be built into the OS.

What if the addon works, but Core just crashes? Getting watchdog right is not trivial.

We already have several mechanism which should prevent hangs/lockups. E.g. kernel crash should lead to a reboot immediately.

Do you have a particular case at hand where you would have expected the Watchdog to kick in? Any details about that case (what was the culprit of the problem etc?)