home-assistant/operating-system #1063: Communication with physical devices from Conbee/Deconz not working

a94marbo, 2020-12-07:
**Hardware Environment**

- [ ] Raspberry Pi [4] B 2MB

**Home Assistant OS release:**
- [ ] Fresh installation of release 5.6

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

ZigBee devices under Deconz (latest verion) with a ConBee stick (latest FW) is shown in interface, and seem to work. Turning lights on and off changes the states in HA, but no communication wirh physical device is done a the light remains in the state it was, even if the state in HA is changed. Not possible to add devices either.
Devices work in HA 4.x branch. Fault not visable at all.

agners, 2020-12-07:
This sounds more like a deCONZ Add-On issue. When you tested with HAOS 4.x branch, di you use the same Add-On version? Can you try downgrading and see if that makes it work again on the same setup otherwise (using `ha os update --version 4.19`).

a94marbo, 2020-12-07:
I am using the same snapshot in both releases, i am restoring the same snapshot in 5.6 and 4.16. Zigbee do not work in 5.6

agners, 2020-12-07:
I cannot reproduce this here:
```
# ha os info                                                                                                                                                                                                                                                                                                                  
board: rpi4-64                                                                                                                                                                                                                                                                                                                
boot: B                                                                                                                                                                                                                                                                                                                       
update_available: false                                                                                                                                                                                                                                                                                                       
version: "5.6"                                                                                                                                                                                                                                                                                                                
version_latest: "4.19"      
```

```
# ha supervisor info
addons:
...
- description: Control a Zigbee network with ConBee or RaspBee by Dresden Elektronik                                                                                                                                                                                                                                          
  icon: true                                                                                                                                                                                                                                                                                                                  
  logo: true                                                                                                                                                                                                                                                                                                                  
  name: deCONZ                                                                                                                                                                                                                                                                                                                
  repository: core                                                                                                                                                                                                                                                                                                            
  slug: core_deconz                                                                                                                                                                                                                                                                                                           
  state: started                                                                                                                                                                                                                                                                                                              
  update_available: false                                                                                                                                                                                                                                                                                                     
  version: 6.5.0                                                                                                                                                                                                                                                                                                              
  version_latest: 6.5.0   
```

I can control the lights from Home Assistant as well as using a remote (the state updates in HA).

Can you check the Add-On log as well as the Home Assistant Core log.

What are you using as `device:` in the Add-On configuration?

a94marbo, 2020-12-08:
Post-boot logs: 5.5 (tested that as well, same result)

**Deconz log**

09:01:46:526 Current channel 15
09:01:46:558 Device TTL 2897 s flags: 0x7
09:01:46:690 New websocket 127.0.0.1:38702 (state: 3) 
09:01:47:150 0x000B57FFFED4959D error APSDE-DATA.confirm: 0xE1 on task
09:01:48:959 0x000B57FFFEA88540 error APSDE-DATA.confirm: 0xE1 on task
172.30.32.2 - - [08/Dec/2020:08:01:50 +0000] "GET /api/87A45B838D/groups?_=1607414504733 HTTP/1.1" 200 4271 "192.168.10.143:8123/api/hassio_ingress/xxxxxxxxxxxxxxxxxxxxxxx/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.198 Safari/537.36"
172.30.32.2 - - [08/Dec/2020:08:01:50 +0000] "GET /api/87A45B838D/lights?_=1607414504734 HTTP/1.1" 200 5360 "192.168.10.143:8123/api/hassio_ingress/xxxxxxxxxxxxxxxxxxxxxxx/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.198 Safari/537.36"
09:01:50:751 0x90FD9FFFFE17ECE5 error APSDE-DATA.confirm: 0xE1 on task
172.30.32.2 - - [08/Dec/2020:08:01:50 +0000] "GET /api/87A45B838D/sensors?_=1607414504735 HTTP/1.1" 200 3995 "192.168.10.143:8123/api/hassio_ingress/xxxxxxxxxxxxxxxxxxxxxxx/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.198 Safari/537.36"
172.30.32.2 - - [08/Dec/2020:08:01:50 +0000] "GET /api/87A45B838D/config?_=1607414504736 HTTP/1.1" 200 5926 "192.168.10.143:8123/api/hassio_ingress/xxxxxxxxxxxxxxxxxxxxxxx/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.198 Safari/537.36"
09:01:52:527 0xD0CF5EFFFEDA2F6C error APSDE-DATA.confirm: 0xE1 on task
09:01:54:335 0x90FD9FFFFE0F6603 error APSDE-DATA.confirm: 0xE1 on task
09:01:56:127 0x000D6FFFFEE75695 error APSDE-DATA.confirm: 0xE1 on task
09:01:57:952 0x0017880103F30CD4 error APSDE-DATA.confirm: 0xE1 on task
09:01:59:760 0x14B457FFFEFD456E error APSDE-DATA.confirm: 0xE1 on task
09:02:01:536 0x90FD9FFFFE7087F2 error APSDE-DATA.confirm: 0xE1 on task
172.30.32.2 - - [08/Dec/2020:08:02:01 +0000] "GET /websocket HTTP/1.1" 101 0 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.198 Safari/537.36"
09:02:01:860 Remove websocket 127.0.0.1:38702 after error Unknown error, close-code: 1000, reason: 
09:02:03:344 0x90FD9FFFFE6E7191 error APSDE-DATA.confirm: 0xE1 on task
09:02:05:152 0x000B57FFFED4B2DF error APSDE-DATA.confirm: 0xE1 on task
09:02:06:944 0x90FD9FFFFE6E789A error APSDE-DATA.confirm: 0xE1 on task
09:02:08:737 0x90FD9FFFFE0BCEA9 error APSDE-DATA.confirm: 0xE1 on task
09:02:10:529 0x90FD9FFFFE7087F2 error APSDE-DATA.confirm: 0xE1 on task
09:02:12:305 0x0017880103D072A5 error APSDE-DATA.confirm: 0xE1 on task
09:02:14:145 0x90FD9FFFFE6E7191 error APSDE-DATA.confirm: 0xE1 on task
09:02:15:937 0x000B57FFFED4959D error APSDE-DATA.confirm: 0xE1 on task
09:02:17:745 0x000B57FFFEA88540 error APSDE-DATA.confirm: 0xE1 on task
09:02:19:537 0x90FD9FFFFE17ECE5 error APSDE-DATA.confirm: 0xE1 on task
09:02:21:330 0xD0CF5EFFFEDA2F6C error APSDE-DATA.confirm: 0xE1 on task
09:02:23:122 0x90FD9FFFFE0F6603 error APSDE-DATA.confirm: 0xE1 on task
09:02:24:946 0x000D6FFFFEE75695 error APSDE-DATA.confirm: 0xE1 on task
09:02:26:738 0x0017880103F30CD4 error APSDE-DATA.confirm: 0xE1 on task
09:02:28:546 0x14B457FFFEFD456E error APSDE-DATA.confirm: 0xE1 on task
09:02:30:323 0x90FD9FFFFE7087F2 error APSDE-DATA.confirm: 0xE1 on task
09:02:32:147 0x90FD9FFFFE6E7191 error APSDE-DATA.confirm: 0xE1 on task
09:02:33:939 0x000B57FFFED4B2DF error APSDE-DATA.confirm: 0xE1 on task
09:02:35:747 0x90FD9FFFFE6E789A error APSDE-DATA.confirm: 0xE1 on task
09:02:37:539 0x90FD9FFFFE0BCEA9 error APSDE-DATA.confirm: 0xE1 on task
09:02:39:332 0x90FD9FFFFE7087F2 error APSDE-DATA.confirm: 0xE1 on task
09:02:41:124 0x0017880103D072A5 error APSDE-DATA.confirm: 0xE1 on task
09:02:42:964 0x90FD9FFFFE6E7191 error APSDE-DATA.confirm: 0xE1 on task
09:02:44:756 0x000B57FFFED4959D error APSDE-DATA.confirm: 0xE1 on task
09:02:46:516 Current channel 15
09:02:46:548 Device TTL 2837 s flags: 0x7
09:02:46:628 0x000B57FFFEA88540 error APSDE-DATA.confirm: 0xE1 on task
09:02:48:420 0x90FD9FFFFE17ECE5 error APSDE-DATA.confirm: 0xE1 on task
09:02:50:245 0xD0CF5EFFFEDA2F6C error APSDE-DATA.confirm: 0xE1 on task
09:02:52:053 0x90FD9FFFFE0F6603 error APSDE-DATA.confirm: 0xE1 on task
09:02:53:829 0x000D6FFFFEE75695 error APSDE-DATA.confirm: 0xE1 on task
09:02:55:637 0x0017880103F30CD4 error APSDE-DATA.confirm: 0xE1 on task
09:02:57:429 0x14B457FFFEFD456E error APSDE-DATA.confirm: 0xE1 on task
09:02:59:254 0x90FD9FFFFE7087F2 error APSDE-DATA.confirm: 0xE1 on task
09:03:01:046 0x90FD9FFFFE6E7191 error APSDE-DATA.confirm: 0xE1 on task
09:03:02:838 0x000B57FFFED4B2DF error APSDE-DATA.confirm: 0xE1 on task
09:03:04:662 0x90FD9FFFFE6E789A error APSDE-DATA.confirm: 0xE1 on task
09:03:06:422 0x90FD9FFFFE0BCEA9 error APSDE-DATA.confirm: 0xE1 on task
09:03:08:246 0x90FD9FFFFE7087F2 error APSDE-DATA.confirm: 0xE1 on task
09:03:10:039 0x0017880103D072A5 error APSDE-DATA.confirm: 0xE1 on task
09:03:11:847 0x90FD9FFFFE6E7191 error APSDE-DATA.confirm: 0xE1 on task
09:03:13:623 0x000B57FFFED4959D error APSDE-DATA.confirm: 0xE1 on task
09:03:15:431 0x000B57FFFEA88540 error APSDE-DATA.confirm: 0xE1 on task
09:03:17:239 0x90FD9FFFFE17ECE5 error APSDE-DATA.confirm: 0xE1 on task
09:03:19:032 0xD0CF5EFFFEDA2F6C error APSDE-DATA.confirm: 0xE1 on task
[... cut ...]
20-12-08 07:58:40 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
20-12-08 07:58:40 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
20-12-08 07:58:40 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-08 07:58:40 INFO (MainThread) [supervisor.host.network] Updating local network information
20-12-08 07:58:44 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-12-08 07:58:44 INFO (MainThread) [supervisor.host] Host information reload completed

a94marbo, 2020-12-08:
Post-boot log from 4.16 (for comparison)

**Deconz log:**

09:14:26:792 0x0017880103f30cd4 found group 0xFFF0
09:14:30:408 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x000B57FFFEA88540 (wait reading or unsupported)
09:14:30:569 ZCL configure reporting rsp seq: 159 0x000B57FFFEA88540 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:14:30:697 ZCL configure reporting rsp seq: 161 0x000B57FFFEA88540 for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
09:14:32:489 ZCL configure reporting rsp seq: 163 0x90FD9FFFFE6E7191 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:14:32:553 ZCL configure reporting rsp seq: 164 0x90FD9FFFFE6E7191 for ep: 0x01 cluster: 0x0008 attr: 0x0000 status: 0x00
09:14:34:825 Bind response success for 0xd0cf5efffeda2f6c ep: 0x01 cluster: 0x0000
09:14:34:985 ZCL configure reporting rsp seq: 165 0xD0CF5EFFFEDA2F6C for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
09:14:35:209 Bind response success for 0xd0cf5efffeda2f6c ep: 0x01 cluster: 0x0006
09:14:35:417 ZCL configure reporting rsp seq: 166 0xD0CF5EFFFEDA2F6C for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:14:35:641 Bind response success for 0xd0cf5efffeda2f6c ep: 0x01 cluster: 0x0008
09:14:35:641 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0xD0CF5EFFFEDA2F6C (wait reading or unsupported)
09:14:36:793 Bind response success for 0x000d6ffffee75695 ep: 0x01 cluster: 0x0000
09:14:36:905 ZCL configure reporting rsp seq: 168 0x000D6FFFFEE75695 for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
09:14:42:585 verified group capacity: 255 and group count: 1 of LightNode 0x14b457fffefd456e
09:14:42:585 0x14b457fffefd456e found group 0xFFF0
09:14:47:609 	0x0017880103D072A5 force poll (2)
09:14:47:673 	0x000D6FFFFEE75695 force poll (2)
09:14:47:690 	0x0017880103F30CD4 force poll (2)
09:14:47:722 ZCL attribute report 0x000D6FFFFEE75695 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
09:14:54:505 Bind response success for 0x000d6ffffee75695 ep: 0x01 cluster: 0x0006
09:14:54:633 ZCL configure reporting rsp seq: 176 0x000D6FFFFEE75695 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:14:54:857 Bind response success for 0xd0cf5efffeda2f6c ep: 0x01 cluster: 0x0000
09:14:55:929 Bind response success for 0x000d6ffffee75695 ep: 0x01 cluster: 0x0008
09:14:55:929 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x000D6FFFFEE75695 (wait reading or unsupported)
09:14:56:089 Bind response success for 0xd0cf5efffeda2f6c ep: 0x01 cluster: 0x0006
09:14:56:265 ZCL configure reporting rsp seq: 178 0xD0CF5EFFFEDA2F6C for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:14:57:257 Current channel 15
09:14:57:290 Device TTL 3257 s flags: 0x7
09:14:59:305 Bind response success for 0x000d6ffffee75695 ep: 0x01 cluster: 0x0000
09:15:08:345 Bind response success for 0x90fd9ffffe17ece5 ep: 0x01 cluster: 0x0000
09:15:08:505 ZCL configure reporting rsp seq: 182 0x90FD9FFFFE17ECE5 for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
09:15:14:841 Bind response success for 0x90fd9ffffe17ece5 ep: 0x01 cluster: 0x0006
09:15:15:002 ZCL configure reporting rsp seq: 184 0x90FD9FFFFE17ECE5 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:15:16:201 Bind response success for 0x90fd9ffffe17ece5 ep: 0x01 cluster: 0x0008
09:15:16:201 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x90FD9FFFFE17ECE5 (wait reading or unsupported)
09:15:16:361 Bind response success for 0x000d6ffffee75695 ep: 0x01 cluster: 0x0006
09:15:16:521 ZCL configure reporting rsp seq: 186 0x000D6FFFFEE75695 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:15:28:838 Bind response success for 0x000d6ffffee75695 ep: 0x01 cluster: 0x0006
09:15:28:998 ZCL configure reporting rsp seq: 189 0x000D6FFFFEE75695 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:15:29:206 Bind response success for 0x90fd9ffffe17ece5 ep: 0x01 cluster: 0x0000
09:15:37:333 Bind response success for 0x90fd9ffffe0f6603 ep: 0x01 cluster: 0x0006
09:15:37:494 ZCL configure reporting rsp seq: 191 0x90FD9FFFFE0F6603 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:15:38:694 Bind response success for 0x90fd9ffffe0f6603 ep: 0x01 cluster: 0x0008
09:15:38:870 ZCL configure reporting rsp seq: 192 0x90FD9FFFFE0F6603 for ep: 0x01 cluster: 0x0008 attr: 0x0000 status: 0x00
09:15:39:094 Bind response success for 0x90fd9ffffe17ece5 ep: 0x01 cluster: 0x0006
09:15:39:270 ZCL configure reporting rsp seq: 193 0x90FD9FFFFE17ECE5 for ep: 0x01 cluster: 0x0006 attr: 0x0000 status: 0x00
09:15:53:334 Bind response success for 0x90fd9ffffe0f6603 ep: 0x01 cluster: 0x0000
09:15:53:494 ZCL configure reporting rsp seq: 196 0x90FD9FFFFE0F6603 for ep: 0x01 cluster: 0x0000 attr: 0x4000 status: 0x00
09:15:57:254 Current channel 15
09:15:57:286 Device TTL 3197 s flags: 0x7
09:15:58:119 send move to color temperature 454 to 0x0017880103D072A5
09:15:58:131 delay sending request 194 dt 0 ms to 0x0017880103D072A5, ep: 0x0B cluster: 0x0006 onAir: 1
09:15:58:132 delay sending request 196 dt 0 ms to 0x0017880103D072A5, ep: 0x0B cluster: 0x0008 onAir: 1
09:15:58:132 delay sending request 197 dt 0 ms to 0x0017880103D072A5, ep: 0x0B cluster: 0x0300 onAir: 1
09:15:58:141 send move to color temperature 454 to 0x0017880103F30CD4
09:15:58:141 delay sending request 194 dt 0 ms to 0x0017880103D072A5, ep: 0x0B cluster: 0x0006 onAir: 1
[... cut ...]
20-12-08 08:15:05 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
20-12-08 08:15:05 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
20-12-08 08:15:06 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-08 08:15:06 INFO (MainThread) [supervisor.host.network] Updating local network information
20-12-08 08:15:09 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-12-08 08:15:09 INFO (MainThread) [supervisor.host] Host information reload completed

agners, 2020-12-09:
@a94marbo thanks for the logs, unfortunately I cannot really see something unusual (other than those errors which likely should not happen).

Which USB port are you using? Does maybe using one of the other two (non-UBS3, or vice versa) helps? I have my deCONZ on a USB 2.0 port.

Does the kernel log contain errors maybe (Supervisor -> System -> Log Provider "Host")?

a94marbo, 2020-12-09:
The conbee is an a usb3 port.
I will try to switch tomorrow and see if its a differance.

a94marbo, 2020-12-10:
Changing the Conbee to a USB2 port didnt change anything.

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.74-v8 (builder@bbdcfcf7bc08) (gcc version 9.3.0 (Buildroot -ga48888a2)) #1 SMP PREEMPT Wed Nov 11 13:25:05 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504576
[    0.000000]   DMA zone: 3788 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242432 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89432 r8192 d29352 u126976
[    0.000000] pcpu-alloc: s89432 r8192 d29352 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496692
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=68dacc27ef0049f0b19528adbb8370ce cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:3B:9D:DB vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37300000-0x3b300000] (64MB)
[    0.000000] Memory: 1828032K/2018304K available (10940K kernel code, 1180K rwdata, 3720K rodata, 1088K init, 1232K bss, 124736K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36107 entries in 142 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x320/0x4b0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000232] Console: colour dummy device 80x25
[    0.000663] printk: console [tty1] enabled
[    0.000718] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000753] pid_max: default: 32768 minimum: 301
[    0.000893] LSM: Security Framework initializing
[    0.001015] AppArmor: AppArmor initialized
[    0.001201] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001252] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.003533] ASID allocator initialised with 32768 entries
[    0.003748] rcu: Hierarchical SRCU implementation.
[    0.004473] EFI services will not be available.
[    0.004935] smp: Bringing up secondary CPUs ...
[    0.005811] Detected PIPT I-cache on CPU1
[    0.005876] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.006877] Detected PIPT I-cache on CPU2
[    0.006916] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.007850] Detected PIPT I-cache on CPU3
[    0.007890] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.008092] smp: Brought up 1 node, 4 CPUs
[    0.008180] SMP: Total of 4 processors activated.
[    0.008203] CPU features: detected: 32-bit EL0 Support
[    0.008226] CPU features: detected: CRC32 instructions
[    0.034614] CPU: All CPU(s) started at EL2
[    0.034684] alternatives: patching kernel code
[    0.036102] devtmpfs: initialized
[    0.046109] Enabled cp15_barrier support
[    0.046161] Enabled setend support
[    0.046590] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.046637] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[... 583 lines cut ...]
[  366.466190] ---[ end trace 2a283903a2af7064 ]---
[  500.070387] hassio: port 11(veth0d6196e) entered blocking state
[  500.070419] hassio: port 11(veth0d6196e) entered disabled state
[  500.070702] device veth0d6196e entered promiscuous mode
[... cut ...]
[  500.932312] hassio: port 11(veth0d6196e) entered forwarding state
[  503.470015] hassio: port 12(veth5cc5c03) entered blocking state
[  503.470040] hassio: port 12(veth5cc5c03) entered disabled state
[  503.470272] device veth5cc5c03 entered promiscuous mode
[  504.612518] eth0: renamed from veth7c6f7d2
[  504.629759] IPv6: ADDRCONF(NETDEV_CHANGE): veth5cc5c03: link becomes ready
[  504.629932] hassio: port 12(veth5cc5c03) entered blocking state
[  504.629948] hassio: port 12(veth5cc5c03) entered forwarding state

Cludch, 2020-12-14:
I seem to have run into this issue as well.
I am using a rpi4 64 bit and the latest HassOS release 5.8.
Everything was still working after upgrading the release and even to Core 2020.12.
However, after I used `datactl move` to move my data partition to an external ssd, no zigbee device seems to receive my commands.

a94marbo, 2020-12-14:
> Update 2: Everything is working now with no sdcard and the SSD connected to the pi via one usb 2 port.

What did you do to make it work? Does it only work with USB-SSD in a USB2 port, did you try an USB3 port?

Cludch, 2020-12-14:
I am still trying different things as I have more issues.
The last working state was with 0.118.5 and 5.8. After I updated to 2020.12 ZigBee commands stopped working.
I tried the USB 3 port earlier, but haven't tried since and as I works with USB 2, I will leave it like that for now. I want to first fix all the other issues I keep having since the updates.

idofr, 2020-12-14:
The latest OS release seems to have broken quite a lot of things for me.
My cc2531 has also stopped working.
zigbee2mqtt gives the following error

```
Zigbee2MQTT:error 2020-12-14 20:30:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-12-14 20:30:32: Failed to start zigbee
Zigbee2MQTT:error 2020-12-14 20:30:32: Exiting...
Zigbee2MQTT:error 2020-12-14 20:30:32: Error: SRSP - SYS - osalNvRead after 6000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.16.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-12-14T19_30_32_348Z-debug.log
2020-12-14T20:30:35: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-12-14T20:30:35: PM2 log: App [npm:0] starting in -fork mode-
2020-12-14T20:30:35: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.2 start /zigbee2mqtt-1.16.2
> node index.js
```

damanti-me, 2020-12-15:
Here also the same error messages... Deconz stopped working.

StevenJonSmith, 2020-12-18:
Hi,
I too am having ZigBee connectivity issues, after the latest updates. I have devices dropping off the network, and I have persistent message delivery errors when trying to turn devices on and off.

I had this issue as well on a previous version, but I ended up disabling the ZHA map add-on I'd manually installed and that seemed to solve it. I notice this feature is back but now integrated into ZHA, so is there a way to disable it?

HA.log:
> 2020-12-18 09:59:05 DEBUG (MainThread) [homeassistant.components.zha.entity] light.christmas_present_left: polling current state
2020-12-18 09:59:06 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x0D17:11:0x0300]: failed to get attributes '['color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster: [0x0d17:11:0x0300]: Message send failure
2020-12-18 09:59:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 09:59:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 09:59:14 DEBUG (MainThread) [homeassistant.components.zha.entity] light.christmas_present_right: polling current state
2020-12-18 09:59:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x97E4:11:0x0008]: received attribute: 0 update with value: 84
2020-12-18 09:59:15 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x97E4:11:0x0300]: failed to get attributes '['color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster: [0x97e4:11:0x0300]: Message send failure
2020-12-18 09:59:19 DEBUG (MainThread) [homeassistant.components.zha.entity] light.hallway_ceiling_light: polling current state
2020-12-18 09:59:20 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xD5CF:11:0x0300]: failed to get attributes '['color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster: [0xd5cf:11:0x0300]: Message send failure
2020-12-18 09:59:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 09:59:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 09:59:47 DEBUG (MainThread) [homeassistant.components.zha.entity] light.landing_ceiling_light: polling current state
2020-12-18 10:00:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 10:00:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 10:00:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 10:00:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 10:00:47 DEBUG (MainThread) [homeassistant.components.zha.entity] light.kitchen_ceiling_light_2: polling current state
2020-12-18 10:00:47 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8A79:11:0x0008]: received attribute: 0 update with value: 50
2020-12-18 10:00:47 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x8A79:11:0x0300]: failed to get attributes '['color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster: [0x8a79:11:0x0300]: Message send failure
2020-12-18 10:00:53 DEBUG (MainThread) [homeassistant.components.zha.entity] light.bedroom_ceiling_light: polling current state
2020-12-18 10:00:53 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x1117:11:0x0008]: received attribute: 0 update with value: 10
2020-12-18 10:01:11 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5B25:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_1
2020-12-18 10:01:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 10:01:12 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 10:01:21 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6D16:1:0x0500]: Updated alarm state: ZoneStatus.Alarm_2
2020-12-18 10:01:33 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x5B25:1:0x0500]: Updated alarm state: ZoneStatus.0
2020-12-18 10:01:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 10:01:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 10:01:43 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x6D16:1:0x0500]: Updated alarm state: ZoneStatus.0
2020-12-18 10:01:50 DEBUG (MainThread) [homeassistant.components.zha.entity] light.fish_tank: polling current state
2020-12-18 10:01:56 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF2FF:11:0x0300]: failed to get attributes '['color_temperature', 'current_x', 'current_y', 'color_loop_active']' on 'light_color' cluster: [0xf2ff:11:0x0300]: Message send failure
2020-12-18 10:01:56 DEBUG (MainThread) [homeassistant.components.zha.entity] light.planter: polling current state
2020-12-18 10:01:57 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xC4CB:11:0x0008]: received attribute: 0 update with value: 254
[... cut ...]
2020-12-18 10:18:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x9881:1:0x0b04]: async_update
2020-12-18 10:18:42 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x03C7:1:0x0b04]: async_update
2020-12-18 10:18:47 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](RaspBee II): Attempting to checkin with device - missed checkins: 1
2020-12-18 10:18:47 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x0000](RaspBee II): does not have a mandatory basic cluster

10bias, 2021-01-08:
I have the same issue. After moving my data with datactl on an external USB SSD my conbee stopped working. Are there any news about this?

damanti-me, 2021-01-08:
I solved it by using a usb connection cable the deconz stick.

StevenJonSmith, 2021-01-08:
Hi @10bias,
Unfortunately I haven't seen any updates. I do have a thread running in the community forum ([https://community.home-assistant.io/t/zha-with-raspbee-ii-message-errors/246224/15](https://community.home-assistant.io/t/zha-with-raspbee-ii-message-errors/246224/15)).

I have had some success by doing a full shutdown of my Pi, via the Supervisor menu, but I'm not sure if that is coincidence.

casperklein, 2021-01-08:
@10bias If your SSD is connected to an USB3 port, try it with USB2 and see if conbee works (also USB2). I had a similar issue and that "fixed" it for me.

a94marbo, 2021-01-08:
> @10bias If your SSD is connected to an USB3 port, try it with USB2 and see if conbee works (also USB2). I had a similar issue and that "fixed" it for me.
Running a SSD HD on USB2 is not really a fix, its a workaround at best.
However, I have tested running the HD and conbee in all combinations of USB slots, with the same (bad) result. I am suspecting a interferance between certain USB HD controllers and the conbee stick, but I am not anywhere close to an answer.

casperklein, 2021-01-08:
> Running a SSD HD on USB2 is not really a fix, its a workaround at best.

That is why I put fixed in quotation marks ;-) Another thing you can try is to test different EEPROM firmwares.

With the 11. december firmware I couldn't get it to work. However, with the older firmware from 3. Sep it "works" (at least on USB2).

    $ sudo rpi-eeprom-update

    BCM2711 detected
    VL805 firmware in bootloader EEPROM
    BOOTLOADER: up-to-date
    CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
     LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
     FW DIR: /lib/firmware/raspberrypi/bootloader/critical
    VL805: up-to-date
    CURRENT: 000138a1
     LATEST: 000138a1

10bias, 2021-01-09:
@casperklein i have already tried the USB2 port but this doesn't work. 
How can i update the firmware? I ssh into the hassos and than i make a `login` but there is no rpi-eeprom-update.

```
tobias@tobias-desktop:~$ ssh root@192.168.1.111 -p 22222
 _    _                                         _     _              _   
| |  | |                          /\           (_)   | |            | |  
| |__| | ___  _ __ ___   ___     /  \   ___ ___ _ ___| |_ __ _ _ __ | |_ 
|  __  |/ _ \| '_ ` _ \ / _ \   / /\ \ / __/ __| / __| __/ _` | '_ \| __|
| |  | | (_) | | | | | |  __/  / ____ \\__ \__ \ \__ \ || (_| | | | | |_ 
|_|  |_|\___/|_| |_| |_|\___| /_/    \_\___/___/_|___/\__\__,_|_| |_|\__|
                                                                         
Welcome to the Home Assistant command line.

For more details use 'help' and 'exit' to close.
If you need access to host system use 'login'.

ha > login
# sudo rpi-eeprom-update
/bin/ash: sudo: not found
# rpi-eeprom-update
/bin/ash: rpi-eeprom-update: not found

```

10bias, 2021-01-09:
Just an little update with an conbee stick it works. So i removed the rasbee und now i use the conbee stick.

casperklein, 2021-01-09:
You'll need Raspberry Pi OS for that. See [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md) and [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md), [here](https://community.home-assistant.io/t/installing-home-assistant-on-a-rpi-4b-8gb-with-ssd-boot/230948)

StevenJonSmith, 2021-01-10:
I am still have some issues, though they seem to be limited to a few specific devices now. Here is a log I got today:
`2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Sending Zigbee request with tsn 138 under 139 request id, data: b'018a00'
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_request (18, 139, 0, <DeconzAddressEndpoint address_mode=2 address=0x098B endpoint=11>, 260, 6, 1, b'\x01\x8a\x00', 2, 0)
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x126a00190012008b00028b090b04010600010300018a000200
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x126a0009000200228b
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.api] APS data request response: [2, <DeviceState.APSDE_DATA_REQUEST_SLOTS_AVAILABLE|2: 34>, 139]
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x0e6b000700a600
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.api] Device state changed response: [<DeviceState.128|APSDE_DATA_REQUEST_SLOTS_AVAILABLE|APSDE_DATA_CONFIRM|2: 166>, 0]
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.api] Command Command.aps_data_confirm (0,)
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.uart] Send: 0x046b0007000000
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.uart] Frame received: 0x046b0013000c00228b028b090b01e900000000
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.api] APS data confirm response for request with id 139: e9
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.api] Request id: 0x8b 'aps_data_confirm' for <DeconzAddressEndpoint address_mode=ADDRESS_MODE.NWK address=0x098b endpoint=11>, status: 0xe9
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy_deconz.zigbee.application] Error while sending 139 req id frame: TXStatus.MAC_NO_ACK
2021-01-10 13:10:31 DEBUG (MainThread) [zigpy.device] [0x098b] Delivery error for seq # 0x8a, on endpoint id 11 cluster 0x0006: message send failure
2021-01-10 13:10:31 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x098B:11:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: '[0x098b:11:0x0006]: Message send failure'
2021-01-10 13:10:31 DEBUG (MainThread) [homeassistant.components.zha.entity] light.kitchen_ceiling_light_1: turned off: [0x098b:11:0x0006]: Message send failure`

I am wondering if this issue needs to be created in homeassistant/core though?

Rendeik, 2021-02-26:
Hi
With RPI 3b + I have to re-pair with each HassOS update to see them online again.

In fact I'm still on HassOS 5.9, because I don't have time to do it now.

austwhite, 2021-12-12:
Is this issue only happening with Conbee/Conbee II devices?
I found on a few devices (odroid, Openbox VM) that the Conbee Sticks are very sensitive to external interference and I had to use a reasonably long extension USB cable to have the conbee stick (both Conbee and Conbee II) at least 50cm away from the computer / raspberry pi and also away from any WiFi routers.
You guys may have already done that of course.   
I have not had the same issue with other brands of ZigBee USB co-ordinators, only with the Conbee's and it seems to be interference in the conbee itself rather than the OS in my tests

closed 2022-03-12 by a94marbo