home-assistant/operating-system #1070: HassOS 4.17 Installed On Virtualbox Does Not Detect Soundcard

owenostrander, 2020-12-08:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [X] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

Installed in Virtualbox, using standard vdi installation.
Virtualbox Version 5.2.30 r130521 (Qt5.6.2)

Host system: Windows 10 PRO
Intel(R) Core(TM) i7-3770 CPU @ 3.40 Ghz
12.0GB RAM
64-bit Operating System, x64-based processor

Host System Sound, video and game Controllers:
Intel(R) Display Audio
Realtek High Definition Audio

Virtualbox Audio Settings:
Host Audio Driver: Windows DirectSound
Audio Controller: ICH AC97
Extended Features: Enable Audio Output (selected)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 4.16
- Additional information (if accessible):

Initially installed as 4.16, then immediately upgraded to 4.17.
Error was unknown at that time

System Health

Home Assistant Core Integration

version: 0.118.5
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.6
os_name: Linux
os_version: 5.4.77
arch: x86_64
timezone: America/Los_Angeles

Hass.io

host_os: HassOS 4.17
update_channel: stable
supervisor_version: 2020.12.6
docker_version: 19.03.12
disk_total: 38.7 GB
disk_used: 9.7 GB
healthy: true
supported: true
board: ova
supervisor_api: ok
version_api: ok
installed_addons: Samba share (9.3.0), Mosquitto broker (5.1), File editor (5.2.0), Duck DNS (1.12.4), Check Home Assistant configuration (3.6.0), ESPHome (1.15.3), Nginx Proxy Manager (0.8.2), Home Assistant Google Drive Backup (0.102.0), Grafana (5.3.6), InfluxDB (3.7.9), AirCast (2.4.1), MariaDB (2.2.1), Mopidy (0.2.7)

Lovelace

dashboards: 1
mode: storage
views: 8
resources: 12
[... 52 lines cut ...]
20-12-08 16:26:08 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-12-08 16:55:23 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
20-12-08 16:55:23 INFO (MainThread) [supervisor.resolution.check] System checks complete
20-12-08 16:56:11 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-12-08 16:56:47 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
20-12-08 16:56:51 INFO (MainThread) [supervisor.api.security] /snapshots access from cebe7a76_hassio_google_drive_backup
20-12-08 17:03:27 INFO (MainThread) [supervisor.host.info] Updating local host information
20-12-08 17:03:27 INFO (MainThread) [supervisor.host.services] Updating service information
20-12-08 17:03:27 INFO (MainThread) [supervisor.host.network] Updating local network information
20-12-08 17:03:28 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
20-12-08 17:03:28 INFO (MainThread) [supervisor.host] Host information reload completed
20-12-08 17:26:14 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
20-12-08 17:55:23 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
20-12-08 17:55:23 INFO (MainThread) [supervisor.resolution.check] System checks complete
20-12-08 17:56:17 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token

**Journal logs:**

**Kernel logs:**

**Description of problem:**

Home Assistant is not detecting the sound card on the host machine for output. 

[image]

agners, 2020-12-09:
Home Assistant OS seems not to have the AC97 drivers built-in. This is also a rather old audio codec.

Try using Intel HD Audio, this should work.

owenostrander, 2020-12-10:
> 
> 
> Home Assistant OS seems not to have the AC97 drivers built-in. This is also a rather old audio codec.
> 
> Try using Intel HD Audio, this should work.

Thank-you.
I tried this, but, unfortunately, it doesn't work here. 
Below is my audio log upon startup.
Note: "I: [pulseaudio] module-udev-detect.c: Found 0 cards."

D: [pulseaudio] core-util.c: setpriority() worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: Found user 'root' (UID 0) and group 'root' (GID 0).
W: [pulseaudio] main.c: Home directory of user 'root' is not '/var/run/pulse', ignoring.
W: [pulseaudio] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
I: [pulseaudio] main.c: Successfully changed user to "root".
I: [pulseaudio] main.c: This is PulseAudio 13.0
D: [pulseaudio] main.c: Compilation host: x86_64
D: [pulseaudio] main.c: Compilation CFLAGS: Not yet supported on meson
D: [pulseaudio] main.c: Running on host: Linux x86_64 5.4.77 #1 SMP Tue Nov 17 17:11:55 UTC 2020
D: [pulseaudio] main.c: Found 2 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: yes
D: [pulseaudio] main.c: Running from build tree: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: All asserts enabled.
I: [pulseaudio] main.c: Machine ID is 3506fc5b99ea4a6b875b1d30e97aadb3.
I: [pulseaudio] main.c: Using runtime directory /var/run/pulse.
I: [pulseaudio] main.c: Using state directory /var/lib/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-13.0/modules.
I: [pulseaudio] main.c: Running in system mode: yes
W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please make sure that you actually do want to do that.
W: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
I: [pulseaudio] main.c: System supports high resolution timers
D: [pulseaudio] memblock.c: Using private memory pool with 1024 slots of size 64.0 KiB each, total size is 64.0 MiB, maximum usable slot size is 65472
I: [pulseaudio] cpu-x86.c: CPU flags: CMOV MMX SSE SSE2 SSE3 SSSE3 SSE4_1 SSE4_2 
I: [pulseaudio] svolume_mmx.c: Initialising MMX optimized volume functions.
I: [pulseaudio] remap_mmx.c: Initialising MMX optimized remappers.
I: [pulseaudio] svolume_sse.c: Initialising SSE2 optimized volume functions.
I: [pulseaudio] remap_sse.c: Initialising SSE2 optimized remappers.
I: [pulseaudio] sconv_sse.c: Initialising SSE2 optimized conversions.
D: [pulseaudio] database-tdb.c: Opened TDB database '/var/lib/pulse/3506fc5b99ea4a6b875b1d30e97aadb3-device-volumes.tdb'
I: [pulseaudio] module-device-restore.c: Successfully opened database file '/var/lib/pulse/3506fc5b99ea4a6b875b1d30e97aadb3-device-volumes'.
I: [pulseaudio] module.c: Loaded "module-device-restore" (index: #0; argument: "").
D: [pulseaudio] database-tdb.c: Opened TDB database '/var/lib/pulse/3506fc5b99ea4a6b875b1d30e97aadb3-stream-volumes.tdb'
I: [pulseaudio] module-stream-restore.c: Successfully opened database file '/var/lib/pulse/3506fc5b99ea4a6b875b1d30e97aadb3-stream-volumes'.
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1 added for object /org/pulseaudio/stream_restore1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Ext.StreamRestore1.RestoreEntry added for object /org/pulseaudio/stream_restore1/entry0
I: [pulseaudio] module.c: Loaded "module-stream-restore" (index: #1; argument: "").
D: [pulseaudio] database-tdb.c: Opened TDB database '/var/lib/pulse/3506fc5b99ea4a6b875b1d30e97aadb3-card-database.tdb'
I: [pulseaudio] module-card-restore.c: Successfully opened database file '/var/lib/pulse/3506fc5b99ea4a6b875b1d30e97aadb3-card-database'.
I: [pulseaudio] module.c: Loaded "module-card-restore" (index: #2; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-port-available" (index: #3; argument: "").
I: [pulseaudio] module.c: Loaded "module-switch-on-connect" (index: #4; argument: "").
I: [pulseaudio] module-udev-detect.c: Found 0 cards.
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "").
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-13.0/modules/module-bluez5-discover.so': success
D: [pulseaudio] dbus-util.c: Successfully connected to D-Bus system bus c9e9504024e159134fbb0fef5fd17fdd as :1.69
I: [pulseaudio] module.c: Loaded "module-bluez5-discover" (index: #7; argument: "").
I: [pulseaudio] module.c: Loaded "module-bluetooth-discover" (index: #6; argument: "").
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #8; argument: "auth-anonymous=1 auth-cookie-enabled=0 socket=/data/external/pulse.sock").
I: [pulseaudio] module-default-device-restore.c: No previous default sink setting, ignoring.
I: [pulseaudio] module-default-device-restore.c: No previous default source setting, ignoring.
I: [pulseaudio] module.c: Loaded "module-default-device-restore" (index: #9; argument: "").
I: [pulseaudio] module.c: Loaded "module-rescue-streams" (index: #10; argument: "").
D: [pulseaudio] module-always-sink.c: Autoloading null-sink as no other sinks detected.
[... cut ...]
I: [pulseaudio] client.c: Created 0 "Native client (UNIX socket client)"
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.
D: [pulseaudio] protocol-native.c: Protocol version: remote 33, local 33
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
I: [pulseaudio] client.c: Freed 0 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.

nightcat91, 2020-12-29:
Hi,

I think I am at the same problem here...

Installed a fresh Hassos using the QCOW2 image on a Proxmox hypervisor.
The Host has three audio devices:
ich9-intel-hda
intel-hda
AC97

I think both of the intel-hda belong to the two HDMI outputs and the AC97 will probably be a "Realtek" analog audio device on the mainboard that gives me the 3.5mm audio headphone jack.

Passing the AC97 audio device to the VM does not show up in the hardware page of the supervisor.
Proxmox shows "Backend Driver: spice"

If more info is needed let me know.

It would be great if the OS would be able to support the AC97 audio device!

jforce, 2021-04-29:
Hi,
I am at the same problem here...
Installed a fresh Hassos using the QCOW2 image on qemu.
The Host has two audio devices:
00:03.0 Audio device: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller (rev 06)
00:1b.0 Audio device: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller (rev 05)

vandad51, 2021-12-14:
@owenostrander Did you find a solution to this? Has anyone?

agners, 2021-12-15:
For VirtualBox its crucial that you choose "Intel HD Audio" as Audio Controller. I created a PR to fix the instructions: https://github.com/home-assistant/home-assistant.io/pull/20708/files.

agners, 2021-12-15:
Also make sure to **not** select the "Null Audio Driver" as Host Audio Driver. This is a dummy driver which will lead to no sound :mute:

vandad51, 2021-12-16:
So I selected Intel HD Audio as the audio, and now I get this right at the bottom of my audio logs.

```

D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34
D: [pulseaudio] protocol-native.c: SHM possible: no
D: [pulseaudio] protocol-native.c: Negotiated SHM: no
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support
E: [alsa-source-Generic Analog] alsa-source.c: ALSA woke us up to read new data from the device, but there was actually nothing to read.
E: [alsa-source-Generic Analog] alsa-source.c: Most likely this is a bug in the ALSA driver 'snd_hda_intel'. Please report this issue to the ALSA developers.
E: [alsa-source-Generic Analog] alsa-source.c: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
I: [pulseaudio] client.c: Freed 0 "supervisor"
I: [pulseaudio] protocol-native.c: Connection died.
E: [alsa-sink-Generic Analog] alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write.
E: [alsa-sink-Generic Analog] alsa-sink.c: Most likely this is a bug in the ALSA driver 'snd_hda_intel'. Please report this issue to the ALSA developers.
E: [alsa-sink-Generic Analog] alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.

```

agners, 2021-12-16:
@vandad51 works here without problems and no such errors.

Did you try different Host outputs? What host operating system are you using? What is your VirtualBox version? What OS version?

owenostrander, 2021-12-16:
[image]

Please see the attached screen capture. I am getting sound, but, something is still not configured correctly. 
The audio is "studdering" ... which I suspect is due to the error referred to "corking" (Requesting rewind due to corking).
Also, the HA command line still doesn't list the sound card (Audio Device). 

If I need to open a new issue, could someone please let me know where to open one?

Thank-you

agners, 2021-12-16:
@owenostrander  since you are the original poster of this issue we can just reopen. Can you share the latest version details? What is your Host Audio Driver? To me it seems the VirtualBox is somehow not keeping up with processing the audio samples. Is your system busy?

owenostrander, 2021-12-17:
> @owenostrander since you are the original poster of this issue we can just reopen. Can you share the latest version details? What is your Host Audio Driver? To me it seems the VirtualBox is somehow not keeping up with processing the audio samples. Is your system busy?

I appreciate it. 

Host Machine:
Operating System: Windows 10 Pro
CPU: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz   3.40 GHz
RAM: 12.0 GB
The host is also a BlueIris Video server, but, CPU usage averages 20%, and memory is around 65%, so, I don't think the system is taxed to the degree that audio wouldn't work
[image]

Virtual Box: 
Version: 5.2.30
(see attached screen shots for settings)
[image]
[image]
[image]

HomeAssistant:
Host: Home Assistant OS 7.0
Core: core-2021.12.2

I am attempting to use Mopidy, to play tts messages over my whole-home audio system. 
In my previous set-up, I ran HomeAssistant, on the same machine, within VirtualBox (same version), but, VirtualBox hosted an Ubutnu Server, with HomeAssistant installed on that. Because that was considered an "unofficial installation", I switched to VirtualBox running HassOS directly, and, ever since then, the audio has never worked correctly. I'd love to get away from the Chromecast Audio I currently have to use. 

Thank-you again, and please let me know if I left anything you need out.

agners, 2022-01-12:
> In my previous set-up, I ran HomeAssistant, on the same machine, within VirtualBox (same version), but, VirtualBox hosted an Ubutnu Server, with HomeAssistant installed on that. 

Do you remember the VirtualBox Audio setting for that virtual machine? Was it something other than "Intel HD Audio"?

You are using a bit outdated VirtualBox version, and there have been fixes around Audio/HDA in releases after 5.2.30 (see https://www.virtualbox.org/wiki/Changelog-5.2).

I don't think I can do much from HAOS side. It VirtualBox has this Wiki page describing how to debug Audio issues with VirtualBox: https://www.virtualbox.org/wiki/AudioDebug

closed 2022-04-19 by github-actions[bot]