home-assistant/operating-system #1071: HassOs 5.6

BlackTurtle123, 2020-12-09:
**Hardware Environment**

- [ ] Raspberry Pi [/4]

- [ ] Fresh installation of release 5.6

**Supervisor logs:**
[image]

**Description of problem:**
Can't seem to update or download new docker containers due to above error

BlackTurtle123, 2020-12-09:
Update, weird thing, a manual docker pull did work docker pull homeassistant/aarch64-hassio-supervisor:2020.12.6

agners, 2020-12-09:
Hm, I have seen Docker Hub hick-ups in the past: Just random container pulling failing with errors, and 5 seconds later working... Have you been able to reproduce this? Otherwise, I think I'll close it and book it into the category "random Docker Hub hick-ups..."

BlackTurtle123, 2020-12-09:
I am now reinstalling hassos 5.6 again. Since i was trying out something
else and report back.

BlackTurtle123, 2020-12-09:
Yes. Did a clean install. Imported my old backup. Try to update supervisor.
Same docker hiccup

agners, 2020-12-11:
@BlackTurtle123 I still haven't come across this issue, but the fact that you have been able to reproduce seems like there is something wrong in some situations.

Is `systemctl status systemd-resolved` saying something if that happens?

BlackTurtle123, 2020-12-11:
I haven't checked that one out at that point. But i will do later when some
time pops up again.

BlackTurtle123, 2020-12-11:
What i can say that was weird tho, ping, curl and docker pull by cmd worked, altho docker pull for these 2 containers didnt worked. Even after i tested curl. It's for now running stable, and didnt encounter it during graphana and influx install.

Might it have been that my backup restore destroyed something?

agners, 2020-12-12:
I don't think that backup restore can have destroyed something.

Docker uses the hosts DNS lookup capabilities, which is now provided by systemd-resolved in release 5. That curl/ping worked but Docker pull didn't is indeed strange, it sounds as if Docker and/or systemd-resolved cached an unsuccessful lookup or something.

BlackTurtle123, 2020-12-12:
> I don't think that backup restore can have destroyed something.
>
> Docker uses the hosts DNS lookup capabilities, which is now provided by systemd-resolved in release 5. That curl/ping worked but [...]

Currently testing if my system can be up and running for a week to test stability. (for the other issue) after that I can retry it again.

BlackTurtle123, 2020-12-19:
[image]
It seems it happened again. 
I am atm on hassos 5.8
Docker pull did work again.

closed 2021-05-12 by stale[bot]