home-assistant/operating-system #1079: Hassos 5.7 only only boots from SDD on USB3 via unpowered USB3 HUB

locorotondo-gitter, 2020-12-11:
**Hardware Environment**

- [X] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 5.7
- [ ] Updated from version x.y
- Additional information (if accessible): 64 bit version

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:** HASSOS 5.7 does not boot from USB3.0 when SDD is directly attached to the USB3.0 port. It does boot from USB2.0 and also boots from USB3.0 if the SDD is connected via an unpowered USB3 hub. The raspi4 boots fine using the same device when flashed with Raspberry PI OS. If you need more info or some testing, just let me know.

Some extra information:
dmesg on my main linux box (with device attached) gives:
[19759.331258] usb 2-3: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd 
[19759.356724] usb 2-3: New USB device found, idVendor=152d, idProduct=0583, bcdDevice= 2.08 
[19759.356727] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3 
[19759.356728] usb 2-3: Product: USB to PCIE Bridge 
[19759.356729] usb 2-3: Manufacturer: JMicron 
[19759.356730] usb 2-3: SerialNumber: 0123456789ABCDEF 
[19759.396855] scsi host7: uas 
[19759.397402] scsi 7:0:0:0: Direct-Access     JMicron  Generic          0208 PQ: 0 ANSI: 6 
[19759.397932] sd 7:0:0:0: Attached scsi generic sg5 type 0 
[19761.404601] sd 7:0:0:0: [sdd] 488397168 512-byte logical blocks: (250 GB/233 GiB) 
[19761.404603] sd 7:0:0:0: [sdd] 4096-byte physical blocks 
[19761.404698] sd 7:0:0:0: [sdd] Write Protect is off 
[19761.404699] sd 7:0:0:0: [sdd] Mode Sense: 5f 00 00 08 
[19761.404911] sd 7:0:0:0: [sdd] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA 
[19761.405472] sd 7:0:0:0: [sdd] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes) [19761.424217]  sdd: sdd1 sdd2 sdd3 sdd4 sdd5 sdd6 sdd7 sdd8 
[19761.425563] sd 7:0:0:0: [sdd] Attached SCSI disk

lsusb -vvv for the device gives:
Bus 002 Device 003: ID 152d:0583 JMicron Technology Corp. / JMicron USA Technology Corp. JMS583Gen 2 to PCIe Gen3x2 Bridge 
Device Descriptor: 
 bLength                18 
 bDescriptorType         1 
 bcdUSB               3.20 
 bDeviceClass            0  
 bDeviceSubClass         0  
 bDeviceProtocol         0  
 bMaxPacketSize0         9 
 idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp. 
 idProduct          0x0583 JMS583Gen 2 to PCIe Gen3x2 Bridge 
 bcdDevice            2.08 
 iManufacturer           1 JMicron 
 iProduct                2 USB to PCIE Bridge 
 iSerial                 3 0123456789ABCDEF 
 bNumConfigurations      1 
 Configuration Descriptor: 
   bLength                 9 
   bDescriptorType         2 
   wTotalLength       0x0079 
   bNumInterfaces          1 
   bConfigurationValue     1 
   iConfiguration          0  
   bmAttributes         0x80 
     (Bus Powered) 
   MaxPower              896mA 
   Interface Descriptor: 
     bLength                 9 
     bDescriptorType         4 
     bInterfaceNumber        0 
     bAlternateSetting       0 
     bNumEndpoints           2 
     bInterfaceClass         8 Mass Storage 
     bInterfaceSubClass      6 SCSI 
     bInterfaceProtocol     80 Bulk-Only 
     iInterface              0  
     Endpoint Descriptor: 
       bLength                 7 
       bDescriptorType         5 
       bEndpointAddress     0x81  EP 1 IN 
       bmAttributes            2 
         Transfer Type            Bulk 
         Synch Type               None 
         Usage Type               Data 
       wMaxPacketSize     0x0400  1x 1024 bytes 
       bInterval               0 
       bMaxBurst              15 
     Endpoint Descriptor: 
       bLength                 7 
       bDescriptorType         5 
       bEndpointAddress     0x02  EP 2 OUT 
       bmAttributes            2 
         Transfer Type            Bulk 
         Synch Type               None 
         Usage Type               Data 
       wMaxPacketSize     0x0400  1x 1024 bytes 
       bInterval               0 
       bMaxBurst              15 
   Interface Descriptor: 
     bLength                 9 
     bDescriptorType         4 
     bInterfaceNumber        0 
     bAlternateSetting       1 
     bNumEndpoints           4 
     bInterfaceClass         8 Mass Storage 
     bInterfaceSubClass      6 SCSI 
     bInterfaceProtocol     98  
     iInterface             10 MSC USB Attached SCSI 
     Endpoint Descriptor: 
       bLength                 7 
       bDescriptorType         5 
       bEndpointAddress     0x01  EP 1 OUT 
       bmAttributes            2 
         Transfer Type            Bulk 
         Synch Type               None 
         Usage Type               Data 
       wMaxPacketSize     0x0400  1x 1024 bytes 
       bInterval               0 
       bMaxBurst               0 
       Command pipe (0x01) 
     Endpoint Descriptor: 
       bLength                 7 
       bDescriptorType         5 
       bEndpointAddress     0x82  EP 2 IN 
       bmAttributes            2 
         Transfer Type            Bulk 
         Synch Type               None 
         Usage Type               Data 
       wMaxPacketSize     0x0400  1x 1024 bytes 
       bInterval               0 
       bMaxBurst               0 
       MaxStreams             32 
       Status pipe (0x02) 
     Endpoint Descriptor: 
       bLength                 7 
       bDescriptorType         5 
       bEndpointAddress     0x83  EP 3 IN 
       bmAttributes            2 
         Transfer Type            Bulk 
         Synch Type               None 
         Usage Type               Data 
       wMaxPacketSize     0x0400  1x 1024 bytes 
       bInterval               0 
       bMaxBurst               3 
       MaxStreams             32 
       Data-in pipe (0x03) 
     Endpoint Descriptor: 
       bLength                 7 
       bDescriptorType         5 
       bEndpointAddress     0x04  EP 4 OUT 
       bmAttributes            2 
         Transfer Type            Bulk 
         Synch Type               None 
         Usage Type               Data 
       wMaxPacketSize     0x0400  1x 1024 bytes 
       bInterval               0 
       bMaxBurst              15 
       MaxStreams             32 
       Data-out pipe (0x04) 
Binary Object Store Descriptor: 
 bLength                 5 
 bDescriptorType        15 
 wTotalLength       0x002a 
 bNumDeviceCaps          3 
 USB 2.0 Extension Device Capability: 
   bLength                 7 
   bDescriptorType        16 
   bDevCapabilityType      2 
   bmAttributes   0x00000f0e 
     BESL Link Power Management (LPM) Supported 
   BESL value     3840 us  
 SuperSpeed USB Device Capability: 
   bLength                10 
   bDescriptorType        16 
   bDevCapabilityType      3 
   bmAttributes         0x00 
   wSpeedsSupported   0x000e 
     Device can operate at Full Speed (12Mbps) 
     Device can operate at High Speed (480Mbps) 
     Device can operate at SuperSpeed (5Gbps) 
   bFunctionalitySupport   1 
     Lowest fully-functional device speed is Full Speed (12Mbps) 
   bU1DevExitLat          10 micro seconds 
   bU2DevExitLat          32 micro seconds 
 SuperSpeedPlus USB Device Capability: 
   bLength                20 
   bDescriptorType        16 
   bDevCapabilityType     10 
   bmAttributes         0x00000001 
     Sublink Speed Attribute count 1 
     Sublink Speed ID count 0 
   wFunctionalitySupport   0x1100 
   bmSublinkSpeedAttr[0]   0x000a4030 
     Speed Attribute ID: 0 10Gb/s Symmetric RX SuperSpeedPlus 
   bmSublinkSpeedAttr[1]   0x000a40b0 
     Speed Attribute ID: 0 10Gb/s Symmetric TX SuperSpeedPlus 
can't get debug descriptor: Resource temporarily unavailable 
Device Status:     0x0000 
 (Bus Powered)

agners, 2020-12-11:
Hm, I guess the fact that a unpowered USB3 hub is enough shows that is not a power issue but something else.

I do have the same device (see #796) but I honestly only checked with powered USB3 hub. If unpowered USB3 hub works, then I wonder what kind of difference attaching directly makes.

locorotondo-gitter, 2020-12-11:
I can do a dmesg and lsusb - vvv *with* the USB hub attached to the drive to see if there is a difference in output. Will check and post if there are any differences.

vodoc, 2020-12-11:
Before fresh installation 5.7 or 5.6 
OR before update from 5.5 to 5.7
```
pi@raspberrypi:~ $ rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
HDMI_DELAY=0
BOOT_ORDER=0x41
```

after unsuccessful boot on 5.7
```
pi@raspberrypi:~ $ rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
HDMI_DELAY=0
```

line BOOT_ORDER=0x41 is absent

agners, 2020-12-11:
This is likely caused by the EEPROM introduced with #939. But I don't think it is a problem as the default `BOOT_ORDER` is `0xf41`:

> The default boot order is 0xf41 which means continuously try SD then USB mass storage.

See https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md.

When you connect a serial console you likely will see the output I posted in #796. I think to fix this we would have to work on the U-Boot USB stack...

vodoc, 2020-12-11:
It is very interesting, but after unsuccessful upgrade 5.5 -> 5.7 and without BOOT_ORDER line, fresh installation of 5.5 does not want to boot. Only after i added this line with 0xf41value, it began to boot and install.
I do not know how to connect serial console. Sorry.

locorotondo-gitter, 2020-12-12:
> Hm, I guess the fact that a unpowered USB3 hub is enough shows that is not a power issue but something else.
> 
> I do have the same device (see #796) but I honestly only checked with powered USB3 hub. If unpowered USB3 hub works, then I wonder what kind of difference attaching directly makes.

OK, long shot here and probably bonkers to even think it; could it be a timing issue of some sorts? USB Hubs (powered or not) do introduce a little delay in the signal (no matter that it is *really* small a delay). I know; bonkers to even think it.

Below are the dmesg and lsusb -vvv *with* the usb hub attached in between. The hub is an Anker 4 port USB3 hub. No differences (as far as I can tell) in the lsusb. the dmesg of course shows the hub.

[ 9799.633454] usb 2-4: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd 
[ 9799.673894] usb 2-4: New USB device found, idVendor=0bda, idProduct=0411, bcdDevice= 1.21 
[ 9799.673897] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0 
[ 9799.673899] usb 2-4: Product: 4-Port USB 3.0 Hub 
[ 9799.673899] usb 2-4: Manufacturer: Generic 
[ 9799.685784] hub 2-4:1.0: USB hub found 
[ 9799.686975] hub 2-4:1.0: 4 ports detected 
[ 9799.787309] usb 1-4: new high-speed USB device number 5 using xhci_hcd 
[ 9799.975519] usb 1-4: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.21 
[ 9799.975521] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0 
[ 9799.975523] usb 1-4: Product: 4-Port USB 2.0 Hub 
[ 9799.975523] usb 1-4: Manufacturer: Generic 
[ 9799.986050] hub 1-4:1.0: USB hub found 
[ 9799.990012] hub 1-4:1.0: 4 ports detected 
[ 9800.036200] usb 2-4.1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd 
[ 9800.056776] usb 2-4.1: New USB device found, idVendor=152d, idProduct=0583, bcdDevice= 2.08 
[ 9800.056779] usb 2-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3 
[ 9800.056780] usb 2-4.1: Product: USB to PCIE Bridge 
[ 9800.056781] usb 2-4.1: Manufacturer: JMicron 
[ 9800.056782] usb 2-4.1: SerialNumber: 0123456789ABCDEF 
[ 9800.105160] scsi host7: uas 
[ 9800.105844] scsi 7:0:0:0: Direct-Access     JMicron  Generic          0208 PQ: 0 ANSI: 6 
[ 9800.106448] sd 7:0:0:0: Attached scsi generic sg5 type 0 
[ 9802.088676] sd 7:0:0:0: [sdd] 488397168 512-byte logical blocks: (250 GB/233 GiB) 
[ 9802.088678] sd 7:0:0:0: [sdd] 4096-byte physical blocks 
[ 9802.088826] sd 7:0:0:0: [sdd] Write Protect is off 
[ 9802.088828] sd 7:0:0:0: [sdd] Mode Sense: 5f 00 00 08 
[ 9802.089005] sd 7:0:0:0: [sdd] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA 
[ 9802.089557] sd 7:0:0:0: [sdd] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes) 
[ 9802.113832]  sdd: sdd1 sdd2 sdd3 sdd4 sdd5 sdd6 sdd7 sdd8 
[ 9802.115310] sd 7:0:0:0: [sdd] Attached SCSI disk

Bus 002 Device 003: ID 152d:0583 JMicron Technology Corp. / JMicron USA Technology Corp. JMS583Gen 2 to PCIe Gen3x2 Bridge
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp.
  idProduct          0x0583 JMS583Gen 2 to PCIe Gen3x2 Bridge
  bcdDevice            2.08
  iManufacturer           1 JMicron
  iProduct                2 USB to PCIE Bridge
  iSerial                 3 0123456789ABCDEF
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
[... 65 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
        MaxStreams             32
        Data-in pipe (0x03)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
        MaxStreams             32
        Data-out pipe (0x04)

Holland1, 2020-12-14:
Try the following, it worked for me

Power up the pi, without SSD (and without any storage device). Then after a few seconds plug in the SSD drive. 

So, does this workaround show that there is a timing issue?

vodoc, 2020-12-14:
> Try the following, it worked for me
> 
> Power up the pi, without SSD (and without any storage device). Then after a few seconds plug in the SSD drive.
> 
> So, does this workaround show that there is a timing issue?

Thanks, it works for me too.

locorotondo-gitter, 2020-12-20:
> Try the following, it worked for me
> 
> Power up the pi, without SSD (and without any storage device). Then after a few seconds plug in the SSD drive.
> 
> So, does this workaround show that there is a timing issue?

Hi, for me this does not work; but maybe my "few seconds" is different from yours ;-)

closed 2021-05-12 by stale[bot]